zigbeefordomoticz/Domoticz-Zigbee #300: errors in logs

dpocock, 2018-12-16:
**Describe the bug**
I noticed these errors in the logs, I'm running the beta-3.3 branch, should I try another branch or is there a configuration issue?

(zigate) Pushing 'onHeartbeatCallback' on to queue
(zigate) Processing 'onHeartbeatCallback' message
(zigate) Calling message handler 'onHeartbeat'.
Error: (zigate) 'onHeartb#033[1;31meat' failed 'AttributeError':''NoneType' object has no attribute 'sendData''.#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 294 in /usr/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 234 in /usr/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 99 in /usr/lib/domoticz/plugins/Domoticz-Zigate/z_output.py, function sendZigateCmd#033[0;0m
(zigate) Pushing 'onHeartbeatCallback' on to queue
(zigate) Processing 'onHeartbeatCallback' message
(zigate) Calling message handler 'onHeartbeat'.
Error: (zigate) 'onHeartb#033[1;31meat' failed 'AttributeError':''NoneType' object has no attribute 'checkTOwaitFor''.#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 294 in /usr/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 243 in /usr/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 194 in /usr/lib/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py, function processListOfDevices#033[0;0m

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Plugin (please complete the following information):**
 - OS: [e.g. Linux Fedora 28]
 - Plugin version
 - Plugin branch

**Additional context**
Add any other context about the problem here.

pipiche38, 2018-12-16:
Yes, please beta-3.3 is obsolete, please use ’stable'

dpocock, 2018-12-16:
thanks, I checked out stable and did a git pull to make sure I have the latest code, restarted domoticz.  I see the same errors in the log, same line numbers.

pipiche38, 2018-12-17:
Could you please do the following:

(1) Stop Domoticz
(2) go to Domoticz-Zigate
(3) rm __pycache__/*         // Removing Python cache
(4) Start Domoticz

dpocock, 2018-12-17:
on my system it is ```__pycache__/*```

I followed those steps, checked the logs again and I still see the same errors.

```
(zigate) Pushing 'onHeartbeatCallback' on to queue
(zigate) Processing 'onHeartbeatCallback' message
(zigate) Calling message handler 'onHeartbeat'.
Error: (zigate) 'onHeartb#033[1;31meat' failed 'AttributeError':''NoneType' object has no attribute 'checkTOwaitFor''.#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 294 in /usr/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 243 in /usr/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 194 in /usr/lib/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py, function processListOfDevices#033[0;0m
(zigate) Pushing 'onHeartbeatCallback' on to queue
(zigate) Processing 'onHeartbeatCallback' message
(zigate) Calling message handler 'onHeartbeat'.
Error: (zigate) 'onHeartb#033[1;31meat' failed 'AttributeError':''NoneType' object has no attribute 'checkTOwaitFor''.#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 294 in /usr/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 243 in /usr/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 194 in /usr/lib/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py, function processListOfDevices#033[0;0m
```

dpocock, 2018-12-17:
Here it is

```
Status: (zigate) load PluginConf
(zigate) PluginConf.txt = {'channel':'0','allowRemoveZigateDevice':'0','allowStoreDiscoveryFrames':'0','logFORMAT':'0'}
Status: (zigate) ZigateMode: ZigBee sendDelay: 0 reTransmit: 1 zTimeOut: 2
Status: (zigate) logFORMAT: 0 logLQI: 0 allowRemoveZigateDevice: 0
Status: (zigate) allowStoreDiscoveryFrames: 0 forceCreationDevice: 0 networkScan: 0 channel: ['0']
Status: (zigate) load ListOfDevice
(zigate) LoadDeviceList - DeviceList filename : /usr/lib/domoticz/plugins/Domoticz-Zigate/DeviceList-2.txt
Error: (zigate) 'onStart'#033[1;31m failed 'OSError'.#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 266 in /usr/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function onStart#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 133 in /usr/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function onStart#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 39 in /usr/lib/domoticz/plugins/Domoticz-Zigate/z_database.py, function LoadDeviceList#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 24 in /usr/lib/domoticz/plugins/Domoticz-Zigate/z_database.py, function _copyfile#033[0;0m
(zigate) Processing 'PollIntervalDirective' message
(zigate) Heartbeat interval set to: 5.
(zigate) Pushing 'onHeartbeatCallback' on to queue
(zigate) Processing 'onHeartbeatCallback' message
(zigate) Calling message handler 'onHeartbeat'.
Error: (zigate) 'onHeartb#033[1;31meat' failed 'AttributeError':''NoneType' object has no attribute 'checkTOwaitFor''.#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 294 in /usr/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 243 in /usr/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat#033[0;0m
Error: (zigate) ----> Lin#033[1;31me 194 in /usr/lib/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py, function processListOfDevices#033[0;0m
```

Commenting out z_database.py line 39 eliminates the error and now the plugin works again.

I put a ```try``` statement there and logged this extra detail:

```
(zigate) LoadDeviceList: OSError: [Errno 30] Read-only file system: '/usr/lib/domoticz/plugins/Domoticz-Zigate/DeviceList-2.txt_2018-12-17-11:18:31'
```

I notice that several of these files were created yesterday but now it won't create them any more.  The filesystem is not mounted read-only.  If I use ```su -s /bin/bash domoticz``` then I can ```touch``` the file, so I believe the ```domoticz``` user has correct permissions for the directory.

dpocock, 2018-12-17:
The issue is caused by the service file:

https://gitlab.com/dpocock/domoticz-debian-raspbian/blob/debian/backports/stretch/debian/domoticz.service

It worked yesterday because I was running the process from the command line.  When I start it with systemctl then it isn't allowed to write there.  As a workaround, I can comment those lines until it is possible to put these files in ```/var```.  I'm wondering if this will also cause problems for files and directories accessed through symlinks from ```/usr/something -> /var/something```

pipiche38, 2018-12-18:
Will be address in #299 for the next release

pipiche38, 2018-12-20:
As part of #299 implementation
https://github.com/sasu-drooz/Domoticz-Zigate/wiki/PluginConf

pipiche38, 2018-12-21:
@dpocock if you want to try this is on the dev-grpmgt branche

pipiche38, 2018-12-29:
Let's track that with #299

closed 2018-12-29 by pipiche38