zigbeefordomoticz/Domoticz-Zigbee #302: Domoticz crash with Zigate plugin restarting with DeviceList non empty

Dams-38, 2018-12-21:
Hello,

As soon as my Zigate plugin restart it crashes with this log

2018-12-21 09:40:12.484  Status: (Zig) load PluginConf
2018-12-21 09:40:12.487  (Zig) PluginConf.txt = {'channel':'0','allowRemoveZigateDevice':'0','allowStoreDiscoveryFrames':'0','logFORMAT':       '0'}
2018-12-21 09:40:12.490  Status: (Zig) ZigateMode: ZigBee sendDelay: 0 reTransmit: 1 zTimeOut: 2
2018-12-21 09:40:12.491  Status: (Zig) logFORMAT: 0 logLQI: 0 allowRemoveZigateDevice: 0
2018-12-21 09:40:12.492  Status: (Zig) allowStoreDiscoveryFrames: 0 forceCreationDevice: 0 networkScan: 0 channel: ['0']
2018-12-21 09:40:12.524  Status: (Zig) load ListOfDevice
2018-12-21 09:40:12.525  (Zig) LoadDeviceList - DeviceList filename : /root/domoticz/plugins/Domoticz-Zigate/DeviceList-28.txt
2018-12-21 09:40:12.531  (Zig) Open : /root/domoticz/plugins/Domoticz-Zigate/DeviceList-28.txt
2018-12-21 09:40:12.538  (Zig) LoadDeviceList - edc7 => dlVal {'ZDeviceID': '5f03', 'Model': 'lumi.sensor_86sw1', 'NbEp': '3', 'IEEE': '0       0158d0001290559', 'Version': '3', 'Type': 'Button', 'App Version': '4', 'ProfileID': '0104', 'Heartbeat': '13', 'RIA': '0', 'Battery': {}       , 'DeviceType': 'RFD', 'RSSI': 198, 'ReadAttributes': {'Ep': {'01': {'0001': {}}}}, 'ReceiveOnIdle': 'Off', 'Stamp': {'Time': '2018-12-21        09:01:44', 'MsgType': '8102'}, 'MacCapa': '80', 'Ep': {'03': {'0004': {}, '000c': {}, '0005': {}, '0003': {}}, '01': {'0012': {}, '0019'       : {}, '0005': {}, 'ClusterType': {'737': 'Button'}, '0006': {}, '0000': {}, '0004': {}, '0003': {}, 'ffff': {}}, '02': {'0012': {}, '0003       ': {}, '0004': {}, '0005': {}}}, 'Status': 'inDB', 'LogicalType': 'End Device', 'SQN': 'af', 'Manufacturer': '1037', 'PowerSource': 'Batt       ery'}
2018-12-21 09:40:12.589  Error: Domoticz received fatal signal 11 !...
2018-12-21 09:40:12.596  Error:   /root/domoticz/domoticz() [0x262154]
2018-12-21 09:40:12.596  Error:   /root/domoticz/domoticz(_Z14signal_handleri+0x4c) [0x26220c]
2018-12-21 09:40:12.596  Error:   /lib/arm-linux-gnueabihf/libc.so.6(+0x26ae0) [0xb6c8aae0]

My deviceList is

edc7 : {'Model': 'lumi.sensor_86sw1', 'DeviceType': 'RFD', 'IEEE': '00158d0001290559', 'Battery': {}, 'Manufacturer': '1037', 'NbEp': '3', 'Stat
us': 'inDB', 'RSSI': 198, 'Heartbeat': '13', 'SQN': 'af', 'Type': 'Button', 'RIA': '0', 'ReceiveOnIdle': 'Off', 'Ep': {'01': {'0006': {}, 'ffff'
: {}, '0005': {}, '0000': {}, '0019': {}, 'ClusterType': {'737': 'Button'}, '0004': {}, '0012': {}, '0003': {}}, '02': {'0004': {}, '0012': {},
'0005': {}, '0003': {}}, '03': {'000c': {}, '0005': {}, '0003': {}, '0004': {}}}, 'PowerSource': 'Battery', 'MacCapa': '80', 'ReadAttributes': {
'Ep': {'01': {'0001': {}}}}, 'ProfileID': '0104', 'ZDeviceID': '5f03', 'Stamp': {'Time': '2018-12-21 09:01:44', 'MsgType': '8102'}, 'App Version
': '4', 'LogicalType': 'End Device', 'Version': '3'}

21a6 : {'Model': 'lumi.sensor_magnet.aq2', 'Version': '3', 'Ep': {'01': {'0006': {}, 'ffff': {}, 'Type': 'Door', 'ClusterType': {'739': 'Door'},
 '0000': {}, '0003': {}}}, 'RIA': '0', 'MacCapa': '80', 'ProfileID': {}, 'NbEp': '1', 'Status': 'inDB', 'RSSI': 168, 'Type': '', 'Heartbeat': '1
', 'ZDeviceID': {}, 'Stamp': {'Time': '2018-12-21 09:02:40', 'MsgType': '8102'}, 'SQN': '01', 'IEEE': '00158d0002c388e4', 'Battery': {}, 'ReadAttributes': {'Ep': {'01': {'0001': {}}}}, 'App Version': '3'}

b10d : {'Model': {}, 'Version': '3', 'Ep': {'f2': {'0021': {}, 'ClusterType': {}}, '0b': {'0006': {}, '0005': {}, '0000': {}, '0019': {}, 'Clust
erType': {'738': 'LvlControl'}, '0004': {}, '0008': {}, '1000': {}, '0003': {}}}, 'RIA': '0', 'MacCapa': '8e', 'ProfileID': 'c05e', 'NbEp': '2',
 'Status': 'inDB', 'RSSI': {}, 'Type': 'Switch/LvlControl/', 'Heartbeat': '6', 'ZDeviceID': '0100', 'SQN': {}, 'IEEE': '0017880104c81df1', 'Battery': {}, 'ReadAttributes': {'Ep': {'01': {'0001': {}, '0000': {}}}}}

Hardware cubieboard under debian 

Domoticz  version
2018-12-21 09:46:19.589 Status: Domoticz V4.9700 (c)2012-2018 GizMoCuz

Python:
2018-12-21 09:46:20.069 Status: PluginSystem: Started, Python version '3.5.0'.

Zigate plugin:
2018-12-21 09:45:21.829  Status: (Zig) onStart called - Zigate plugin V 4.0.4

pipiche38, 2018-12-21:
Please could you post this issue on the Domoticz Forum ?

Dams-38, 2018-12-21:
Ok sorry for bad place posting

closed 2018-12-21 by Dams-38