zigbeefordomoticz/Domoticz-Zigbee #304: domoticz crash after around 20H

vsaugey, 2018-12-23:
Hello,

 The zigate plugin crash domoticz after around 20H.
Information on the version :
2018-12-20 22:41:03.866  Status: Domoticz V4.9700 (c)2012-2018 GizMoCuz
2018-12-20 22:41:03.866  Status: Build Hash: a3a4590-modified, Date: 2018-06-23 16:24:51
...
2018-12-20 22:41:04.221  Status: PluginSystem: Started, Python version '3.5.1'.
...
2018-12-20 22:41:12.027  Status: (Zigate) Initialized version 4.0.4, author 'zaraki673  pipiche38'

The final lines in my domoticz.log are :
2018-12-21 18:33:21.435  (Zigate) Deallocating connection object 'ZiGate' (/dev/ttyUSB1:).
2018-12-21 18:33:21.758  (Zigate) Connection released by Python, reference count is 0.

pipiche38, 2018-12-23:
Would be interesting to have more log from the stop time. Was Domoticz stoping as well ?

vsaugey, 2018-12-23:
Hello, 

Thank you for your reply. 
Yesterday, I restart my domoticz without backup my log file, sorry. 
By experience, it should crash soon (probably during on the evening). 

When the error happens, everything crash, domoticz instance is stopped. 
Before, I have activated the zigate product, I seem not had this kind of behavior. 

With the default value in "verbose and debuging" set to "none", I see nothing in the log. 
Last time, I set it to "Domoticz Framework - Basic", plugin wrote the last two line, I sent you. 
Do you want I put a different value to better trace the problem (verbose or all) ? 

Regards, 

Vincent 

----- Mail original -----

pipiche38, 2018-12-23:
Vincent,

Debug mode set to None, should give you the minimum which is in most cases sufficient. 
Is the Zigate connected directly to a USB port of the system, or are you going via an USB HUB ?

In your case, it seems that there is a Domoticz crash and we should also get some stack traces.

Keep me posted. I'm also reachable on Keybase.io see my username on guthub. It will be more convenient to chat

pipiche38, 2019-01-14:
Close. Please use the Forum channels next time 

    English channel : https://www.domoticz.com/forum/viewforum.php?f=68
    French Channel : https://easydomoticz.com/forum/viewforum.php?f=28&sid=a8633a9d8cb2acccb32872543b50fd47

closed 2019-01-14 by pipiche38