zigbeefordomoticz/Domoticz-Zigbee #306: problem with adding new device

sonics007, 2018-12-30:
linux xubuntu 18.04, zigate  no wifi
i istalled domoticz and zigate plugin.try to add xiaomi door senzor.pressed button for few secs. and then this
` 2018-12-30 15:12:34.093 Sunrise: 07:30:00 SunSet: 15:53:00
2018-12-30 15:12:34.093 Day length: 08:23:00 Sun at south: 11:05:00
2018-12-30 15:12:34.093 Civil twilight start: 06:54:00 Civil twilight end: 16:30:00
2018-12-30 15:12:34.093 Nautical twilight start: 06:14:00 Nautical twilight end: 17:09:00
2018-12-30 15:12:34.093 Astronomical twilight start: 05:36:00 Astronomical twilight end: 17:47:00
2018-12-30 15:12:34.124 Active notification Subsystems: (0/13)
2018-12-30 15:12:34.127 Starting shared server on: :::6144
2018-12-30 15:12:34.127 Accepting shared server connections via MyDomotiz (see settings menu).
2018-12-30 15:12:34.030 Status: Domoticz V4.10301 (c)2012-2018 GizMoCuz
2018-12-30 15:12:34.030 Status: Build Hash: b1521946, Date: 2018-12-20 19:46:47
2018-12-30 15:12:34.049 Status: Startup Path: /home/sonics007/domoticz/
2018-12-30 15:12:34.093 Status: EventSystem: reset all events...
2018-12-30 15:12:34.122 Status: PluginSystem: Started, Python version '3.6.7'.
2018-12-30 15:12:34.124 Status: WebServer(HTTP) started on address: :: with port 8080
2018-12-30 15:12:34.126 Status: Proxymanager started.
2018-12-30 15:12:34.127 Status: TCPServer: shared server started...
2018-12-30 15:12:34.127 Status: RxQueue: queue worker started...
2018-12-30 15:12:34.126 Error: WebServer(SSL) startup failed on address 0.0.0.0 with port: 443: bind: Address already in use
2018-12-30 15:12:34.126 Error: WebServer(SSL) check privileges for opening ports below 1024
2018-12-30 15:12:35.304 Status: PROXY: Authenticate result: success.
2018-12-30 15:12:36.689 (zigate) ListOfDevices :
2018-12-30 15:12:36.689 (zigate) Establish Zigate connection
2018-12-30 15:12:36.691 (zigate) Connected successfully
2018-12-30 15:12:36.127 Status: (zigate) Started.
2018-12-30 15:12:36.128 Status: EventSystem: reset all events...
2018-12-30 15:12:36.128 Status: EventSystem: reset all device statuses...
2018-12-30 15:12:36.153 Status: Python EventSystem: Initalizing event module.
2018-12-30 15:12:36.153 Status: EventSystem: Started
2018-12-30 15:12:36.153 Status: EventSystem: Queue thread started...
2018-12-30 15:12:36.593 Status: PluginSystem: Entering work loop.
2018-12-30 15:12:36.627 Status: OpenZWave: using config in: /home/sonics007/domoticz/Config/
2018-12-30 15:12:36.628 Status: OpenZWave: Starting...
2018-12-30 15:12:36.628 Status: OpenZWave: Version: 1.4-3297-g98c3bde5-dirty
2018-12-30 15:12:36.687 Status: (zigate) Entering work loop.
2018-12-30 15:12:36.687 Status: (zigate) Initialized version 4.0.0, author 'zaraki673 pipiche38'
2018-12-30 15:12:36.688 Status: (zigate) onStart called - Zigate plugin V 4.0.0
2018-12-30 15:12:36.688 Status: (zigate) load PluginConf
2018-12-30 15:12:36.688 Status: (zigate) ZigateMode: ZigBee sendDelay: 0 reTransmit: 1 zTimeOut: 2
2018-12-30 15:12:36.688 Status: (zigate) logFORMAT: 0 logLQI: 0 allowRemoveZigateDevice: 0
2018-12-30 15:12:36.688 Status: (zigate) allowStoreDiscoveryFrames: 0 forceCreationDevice: 0 networkScan: 0 channel: ['0']
2018-12-30 15:12:36.689 Status: (zigate) load ListOfDevice
2018-12-30 15:12:36.689 Status: (zigate) Entries loaded from /home/sonics007/domoticz/plugins/Domoticz-Zigate/DeviceList-4.txt : 0
2018-12-30 15:12:36.689 Status: (zigate) Connection Name: Zigate, Transport: Serial, Address: /dev/ttyUSB0
2018-12-30 15:12:36.691 Status: (zigate) onConnect called
2018-12-30 15:12:36.691 Status: (zigate) ZigateConf - Setting UTC Time to : 1546179156
2018-12-30 15:12:36.691 Status: (zigate) Start network
2018-12-30 15:12:36.691 Status: (zigate) Zigate enter in discover mode for ever
2018-12-30 15:12:36.796 Status: OpenZWave: Driver Ready
2018-12-30 15:12:36.813 Status: OpenZWave: All Nodes queried
2018-12-30 15:12:43.102 (zigate) receiveDataCmd - unexpected message Type: 0102 Expecting: 8010
2018-12-30 15:12:43.102 (zigate) ZigateRead - Unknow Message Type for : 010102000efa032e3601000600000010000101fb03
2018-12-30 15:12:53.420 Error: (zigate) onMessage : Frame CRC is bad, computed = 122 received = 58
2018-12-30 15:12:55.524 (zigate) receiveDataCmd - unexpected message Type: 00cd Expecting: 8009
2018-12-30 15:12:55.524 (zigate) ZigateRead - Unknow Message Type for : 0100cd000cdde40000150d0002d226ed00fb03
2018-12-30 15:12:55.825 (zigate) receiveDataCmd - unexpected message Type: 00cd Expecting: 8024
2018-12-30 15:12:55.825 (zigate) ZigateRead - Unknow Message Type for : 0100cd000c30e40000150d0002d226ed001603
2018-12-30 15:12:55.825 (zigate) receiveDataCmd - unexpected message Type: 0701 Expecting: 8024
2018-12-30 15:12:55.825 (zigate) ZigateRead - Unknow Message Type for : 01070100031300001603
2018-12-30 15:12:55.826 (zigate) receiveDataCmd - unexpected message Type: 0102 Expecting: 8024
2018-12-30 15:12:55.826 (zigate) ZigateRead - Unknow Message Type for : 0101020023ef00e400010000000500c20016ecf5ede92ef3e5eef3eff2dfede1e7eee5f42ee1f1321903
2018-12-30 15:12:55.826 Error: (zigate) onMessage : Frame CRC is bad, computed = 210 received = 146
2018-12-30 15:12:56.078 (zigate) receiveDataCmd - unexpected message Type: 0102 Expecting: 8024
2018-12-30 15:12:56.078 (zigate) ZigateRead - Unknow Message Type for : 010102002ac901e400010000ff0100c2001d0121c90c03281b042128010521db00062401000000000a210000e41001fe03 `

please whats wrong im newbe

pipiche38, 2018-12-30:
Very hard to understand what the problem is.

What is the plugin version?
What exactly are the steps you are doing ? 
Did you set the Permit to join ?

Last , thanks to use the forum instead of reporting an issue

sonics007, 2018-12-30:
ok write it at forum pls check it there
https://www.domoticz.com/forum/viewtopic.php?f=68&t=26304
thx

closed 2018-12-30 by sonics007