zigbeefordomoticz/Domoticz-Zigbee #307: Xiaomi vibration device - can't change device value type

KingKrush67, 2018-12-31:
**Describe the bug**
The aqara 'vibration' device type can't be modified to 'Off' manually.

2018-12-31 06:31:55.673 (Zigate) mgtCommand : found Device : 42 in Ep 01 Vibration
2018-12-31 06:31:55.673 (Zigate) mgtCommand - List of TypeName : ['Vibration']
2018-12-31 06:31:55.673 (Zigate) mgtCommand - Look you are trying to action a non commandable device Device Lave linge has available Type ['Vibration']

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the device
2. Click on 'Off'
3. See error in the logs

**Expected behavior**
The value should be changed.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser : Chrome
 - Version : 71.0.3578.98 

**Plugin (please complete the following information):**
 - OS: Raspbian
 - Plugin version 4.0.0-199-gdc3b527

KingKrush67, 2018-12-31:
The vibration device seems to work in a strange way.

After a 'Tilt', the state changes to 'Tilt' then 'Off'.
After a 'Free Fall', the state changes to 'Free Fall' then 'Off'.

But after a 'Vibrate', the state changes to  'Vibrate', but then never changes back to 'Off' !

I don't know if it helps...

pipiche38, 2018-12-31:
What branch version ?
Could you provide the associated log ?

The switch can only be updated by the connected object.

KingKrush67, 2018-12-31:
Branch : dev-grpmgt

Log :
2018-12-31 10:46:08.649 (Zigate) DeviceExist - Found in ListOfDevices with status = inDB
2018-12-31 10:46:08.649 (Zigate) Decode8102 : Attribute Report from 3466 SQN = 8e ClusterID = 0101 AttrID = 0055 Attribute Data = 0001
2018-12-31 10:46:08.649 (Zigate) Device : 3466 MacCapa : 80 updating SQN to 8e
2018-12-31 10:46:08.649 (Zigate) updSQN - Device : 3466 updating SQN to 8e
2018-12-31 10:46:08.650 (Zigate) ReadCluster - MsgData lenght is: 28 out of 24+
2018-12-31 10:46:08.650 (Zigate) DeviceExist - Found in ListOfDevices with status = inDB
2018-12-31 10:46:08.650 (Zigate) ReadCluster 0101 - Dev: 3466, EP:01 AttrID: 0055, AttrType: 21, Attribute: 0001
2018-12-31 10:46:08.650 (Zigate) ReadCluster 0101 - Aqara Vibration - Attribute: 0001
2018-12-31 10:46:08.650 (Zigate) MajDomoDevice - Device ID : 00158d0002adc30b - Device EP : 01 - Type : 0101 - Value : 20 - Hue :
2018-12-31 10:46:08.650 (Zigate) MajDomoDevice - Type = Vibration
2018-12-31 10:46:08.650 (Zigate) MajDomoDevice - NWKID = 3466 IEEE = 00158d0002adc30b Unit = 42
2018-12-31 10:46:08.650 (Zigate) MajDomoDevice - {'0101': '00', 'ClusterType': {'42': 'Vibration'}, '0003': {}, '0019': {}, '0005': {}, 'Type': 'Vibration', '0000': {}}
2018-12-31 10:46:08.650 (Zigate) MajDomoDevice - We have 2 EPs with ClusterType
2018-12-31 10:46:08.650 (Zigate) MajDomoDevice - search ClusterType in : {'0101': '00', 'ClusterType': {'42': 'Vibration'}, '0003': {}, '0019': {}, '0005': {}, 'Type': 'Vibration', '0000': {}} for : 42
2018-12-31 10:46:08.651 (Zigate) MajDomoDevice - search ClusterType in : {'42': 'Vibration'} for : 42
2018-12-31 10:46:08.651 (Zigate) MajDomoDevice - NWKID: 3466 SwitchType: 18, DeviceType: Vibration, ClusterType: Vibration, old_nVal: 0 , old_sVal: 00
2018-12-31 10:46:08.651 (Zigate) UpdateDevice_v2 for : 6 Battery Level = 90 Signal Level = 186
2018-12-31 10:46:08.651 (Zigate) UpdateDevice_v2 for : 6 RSSI = 9
2018-12-31 10:46:08.651 (Zigate) Update v2 Values 2:'20:' (Lave linge)
2018-12-31 10:46:08.662 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-31 10:46:08.654 Status: Notification: VIBRATION ON (Vibrate)
2018-12-31 10:46:10.121 Notification sent (pushover) => Success
2018-12-31 10:46:12.068 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-31 10:46:12.072 (Zigate) HB count = 3
2018-12-31 10:46:17.082 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-31 10:46:17.082 (Zigate) Request Basic via Read Attribute request: ce8e EPout = 01
2018-12-31 10:46:17.082 (Zigate) ReadAttributeReq - addr =ce8e Cluster = 0000 Attributes = [0, 5, 1, 2, 3, 4, 7, 16]
2018-12-31 10:46:17.082 (Zigate) attributes: [0, 5, 1, 2, 3, 4, 7, 16] len =8
2018-12-31 10:46:17.083 (Zigate) ReadAttributeReq: 0005 for ce8e/00
2018-12-31 10:46:17.083 (Zigate) sendData - Cmd: 0042 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-31 10:46:17.083 (Zigate) _sendData 0042
2018-12-31 10:46:17.083 (Zigate) ZigateEncode - Encodind data: 0042
2018-12-31 10:46:17.083 (Zigate) Encode in: 0042 / out:021042
2018-12-31 10:46:17.083 (Zigate) ZigateEncode - Encodind data: 0002
2018-12-31 10:46:17.083 (Zigate) Encode in: 0002 / out:02100212
2018-12-31 10:46:17.083 (Zigate) ZigateEncode - Encodind data: 00
2018-12-31 10:46:17.084 (Zigate) Encode in: 00 / out:0210
2018-12-31 10:46:17.084 (Zigate) ZigateEncode - Encodind data: ce8e
2018-12-31 10:46:17.084 (Zigate) Encode in: ce8e / out:ce8e
2018-12-31 10:46:17.084 (Zigate) Request Basic via Read Attribute request: 3466 EPout = 01
2018-12-31 10:46:17.084 (Zigate) ReadAttributeReq - addr =3466 Cluster = 0000 Attributes = [0, 5, 1, 2, 3, 4, 7, 16]
2018-12-31 10:46:17.084 (Zigate) attributes: [0, 5, 1, 2, 3, 4, 7, 16] len =8
2018-12-31 10:46:17.084 (Zigate) sendData - Cmd: 0042 waitQ: 1 dataQ: 1 normalQ: 0
2018-12-31 10:46:17.084 (Zigate) sendData - waitQ: 0042
2018-12-31 10:46:17.085 (Zigate) sendData - waitD: 8042
2018-12-31 10:46:17.085 (Zigate) addCmdToSend: cmd: 0042 data: 3466 reTransmit: 0
2018-12-31 10:46:17.088 (Zigate) HB count = 4
2018-12-31 10:46:17.189 (Zigate) receiveData - new Data coming
2018-12-31 10:46:17.189 (Zigate) receiveData - MsgType: 8000
2018-12-31 10:46:17.189 (Zigate) receivedData - MsgType: 8000 PacketType: 0042
2018-12-31 10:46:17.189 (Zigate) receiveStatusCmd - waitQ: 1 dataQ: 1 normalQ: 1
2018-12-31 10:46:17.189 (Zigate) ZigateRead - decoded data : 0180000005ad006a00420003 lenght : 24
2018-12-31 10:46:17.190 (Zigate) ZigateRead - MsgType: 8000, MsgLength: 0005, MsgCRC: ad, Data: 006a0042; RSSI: 00
2018-12-31 10:46:17.190 (Zigate) ZigateRead - MsgType 8000 - reception status : 0180000005ad006a00420003
2018-12-31 10:46:17.190 (Zigate) Decode8000_v2 - MsgData lenght is : 8 out of 8
2018-12-31 10:46:17.190 (Zigate) Decode8000_v2 - status: Success SEQ: 6a Packet Type: 0042
2018-12-31 10:46:22.048 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 1 normalQ: 1
2018-12-31 10:46:22.049 (Zigate) checkTOwaitForStatus - 8042s enter at: 1546249577 delta: 5
2018-12-31 10:46:22.049 (Zigate) waitForData - Timeout 5 on 8042 Command waiting for 0042
2018-12-31 10:46:22.049 (Zigate) sendData - Cmd: 0042 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-31 10:46:22.049 (Zigate) _sendData 0042
2018-12-31 10:46:22.049 (Zigate) ZigateEncode - Encodind data: 0042
2018-12-31 10:46:22.049 (Zigate) Encode in: 0042 / out:021042
2018-12-31 10:46:22.049 (Zigate) ZigateEncode - Encodind data: 0002
2018-12-31 10:46:22.049 (Zigate) Encode in: 0002 / out:02100212
[... cut ...]
2018-12-31 10:46:32.078 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-31 10:46:32.082 (Zigate) HB count = 7
2018-12-31 10:46:37.091 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-31 10:46:37.095 (Zigate) HB count = 8
2018-12-31 10:46:42.054 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-31 10:46:42.058 (Zigate) HB count = 9
2018-12-31 10:46:47.068 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-31 10:46:47.072 (Zigate) HB count = 10

pipiche38, 2019-01-01:
So the issue is related to the fact that we are not receiving the 'Liberation Mode' , which seems to happen after Tilt and Free-Fall!

https://faire-ca-soi-meme.fr/domotique/2018/09/03/test-xiaomi-aqara-vibration-sensor/#Les_clusters

If someone has a Xiaomi Gateway it would be great to get the behaviour on that, to understand if this is an automatic event , or if the system would expect a kind of Manual interaction to reset to Off .

@Louis22, @KingKrush67

LouiS22, 2019-01-01:
Unfortunately I don't have a Gateway :(

KingKrush67, 2019-01-01:
Same here, no gateway...

pipiche38, 2019-01-16:
4b11d299e2e35a72cb90dadcf74264c8455d4643

deennoo, 2019-01-17:
Ok, 

This fix works for me

closed 2019-01-20 by pipiche38