zigbeefordomoticz/Domoticz-Zigbee #308: Reliability?

0rsa, 2019-01-04:
I'm disappointed by reliability of the plugin for now.
I started using it for 2 months and I lost devices communication updating from 3.x to 4.x
I did a new configuration and after a few weeks, my domoticz crashed.
I restarted it and I lost again devices communication (Xiaomi Smart Door And Window Sensor).
I deleted my existing devices to do again a new configuration and I'm just not able now to link my sensors to domoticz. There is no error at all in domoticz logs but sensors are never detected by the system.
Of course, my zigate key permit mode is on and I allowed adding new devices to domoticz for 5 minutes before trying.

What's your feedback? Do you think it's reliable or do I switch to xiaomi gateway to have a robust system?

pipiche38, 2019-01-04:
What firmware version are you using ? They are issues with 3.0e 

Please get in touch with Zigate supplier as your problem is on that area.

pipiche38, 2019-01-04:
Please check the following issues : (which I think are matching your issue)
https://github.com/fairecasoimeme/ZiGate/issues/91
https://github.com/fairecasoimeme/ZiGate/issues/89

0rsa, 2019-01-04:
I have the original firmware, I'll check this evening to update and I will let you know if it works better.
Thank you very much.

pipiche38, 2019-01-04:
In the logs you should see the firmware version when starting the plugin

pipiche38, 2019-01-04:
As regards to your last problem could you please provide the Log.
Make sure that you are on the latest stable branche N

0rsa, 2019-01-04:
Je vois qu'on est entre français alors je vais me permettre de switch sur la langue de Molière, ça sera sans doute plus simple pour se comprendre.
Du coup, j'ai parcouru les threads en rapport, ça n'est pas pour me rassurer sur la fiabilité de l'ensemble mais je comprends, étant moi-même dev dans un autre domaine d'activité, et je suis prêt à essayer d'apporter le max d'infos afin d'aider à la résolution.
Petite parenthèse, je fais des tests actuellement avec la clé Zigate et un seul détecteur Xiaomi d'ouverture de porte (j'en ai 19 autres dans un carton), la vraie installation qui doit être pérenne et robuste sera installée dans une nouvelle maison le mois prochain. Par là, je veux dire que je suis en mesure de péter ma config domoticz sans problème car il ne s'agit que de tests actuellement donc ça n'est pas dérangeant.

Par rapport à ton dernier message, quels logs souhaites-tu exactement ? Les logs au moment du dernier crash où j'ai perdu la communication avec le module ?
Car actuellement, le log montre que le plugin Zigate démarre bien et charge 0 device car j'avais supprimé le seul détecteur dans Domoticz pour refaire un appairage, chose que je n'arrive plus à faire depuis hier soir après 4-5 tentatives différentes infructueuses (notamment en essayant : reset de la clé, débranchement / rebanchement physique de la clé, redémarrage de domoticz, le log ne montre aucune détection du capteur sans erreur particulière, je passe bien évidemment le capteur en mode association en plus de l'activer pour la clé et dans domoticz).

Est-ce que tu me conseilles de partir sur le 3.0f qui n'est pas encore finalisée ? J'ai lu qu'un membre avait des pertes de détecteur au runtime avec la 3.0d.

Merci ;)

pipiche38, 2019-01-04:
Peux-tu me contacter sur Keybase.io ou le forum Francais. Ceci afin d'eviter autre chose que de l'anglais sur Github.

Merci

pipiche38, 2019-01-05:
You mentioned in your last message you cannot pair any device anymore. 
- If I understood you have remove or clean the database.
- if you don't have any device in Domoticz, could you check that you have also clean the Zigate plugin Database . Is the file DeviceList-xx.txt is empty ?
- Then  if you can provide the domoticz.log file , we could then investigate what is going on and wrong.

Last, please use the forum to post : 
- French Domoticz forum: https://easydomoticz.com/forum/viewforum.php?f=28&sid=f884117fe33a2ce9d0848e51c9a9cdc7
- Engish Domoticz forum: https://www.domoticz.com/forum/viewforum.php?f=68
- As well you can also contact me on keybase.io

Thanks for your understanding.

pipiche38, 2019-01-14:
Close. Use the forum channels for those matters.

closed 2019-01-14 by pipiche38