zigbeefordomoticz/Domoticz-Zigbee #315: Aqara vibration sensor error

blooesky, 2019-01-11:
Hi,

I tried to pairing Aqara vibration sensor and i have this message in domoticz log:

2019-01-11 20:01:13.149 Status: (Zigate) [-] NEW OBJECT: 8e35 Simple Descriptor EP 01
2019-01-11 20:01:13.150 Status: (Zigate) [-] NEW OBJECT: 8e35 ProfileID 0104
2019-01-11 20:01:13.150 Status: (Zigate) [-] NEW OBJECT: 8e35 ZDeviceID 000a
2019-01-11 20:01:13.151 Status: (Zigate) [-] NEW OBJECT: 8e35 Cluster IN Count: 04
2019-01-11 20:01:13.151 Status: (Zigate) [-] NEW OBJECT: 8e35 Cluster In 1: 0000
2019-01-11 20:01:13.152 Status: (Zigate) [-] NEW OBJECT: 8e35 Cluster In 2: 0003
2019-01-11 20:01:13.152 Status: (Zigate) [-] NEW OBJECT: 8e35 Cluster In 3: 0019
2019-01-11 20:01:13.152 Status: (Zigate) [-] NEW OBJECT: 8e35 Cluster In 4: 0101
2019-01-11 20:01:13.153 Status: (Zigate) [-] NEW OBJECT: 8e35 Cluster OUT Count: 06
2019-01-11 20:01:13.153 Status: (Zigate) [-] NEW OBJECT: 8e35 Cluster Out 1: 0000
2019-01-11 20:01:13.154 Status: (Zigate) [-] NEW OBJECT: 8e35 Cluster Out 2: 0004
2019-01-11 20:01:13.154 Status: (Zigate) [-] NEW OBJECT: 8e35 Cluster Out 3: 0003
2019-01-11 20:01:13.154 Status: (Zigate) [-] NEW OBJECT: 8e35 Cluster Out 4: 0005
2019-01-11 20:01:13.155 Status: (Zigate) [-] NEW OBJECT: 8e35 Cluster Out 5: 0019
2019-01-11 20:01:13.155 Status: (Zigate) [-] NEW OBJECT: 8e35 Cluster Out 6: 0101
2019-01-11 20:01:13.315 Status: (Zigate) [0] NEW OBJECT: 8e35 Ox8043 received Infos
2019-01-11 20:01:13.316 Status: (Zigate) [0] NEW OBJECT: 8e35 Trying to create Domoticz device(s) 
019-01-11 20:02:52.352 (Zigate) ReadCluster 0101 - Dev: 8e35 Ep : 01 Attribute Type:21 Attribute : 0055
2019-01-11 20:02:52.352 (Zigate) ReadCluster 0101 - Aqara Vibration - Attribute: 0001
2019-01-11 20:02:52.351 Error: (Zigate) ReadCluster 0101 not fully implemented, please contact us on https://github.com/sasu-drooz/Domoticz-Zigate 

**Expected behavior**
The value should be changed.

**Plugin (please complete the following information):**
OS: Raspbian

pipiche38, 2019-01-11:
Hello,

Could you please use the forum for those matters.
English channel : https://www.domoticz.com/forum/viewforum.php?f=68 <https://www.domoticz.com/forum/viewforum.php?f=68>
French Channel : https://easydomoticz.com/forum/viewforum.php?f=28&sid=a8633a9d8cb2acccb32872543b50fd47 <https://easydomoticz.com/forum/viewforum.php?f=28&sid=a8633a9d8cb2acccb32872543b50fd47>

Please mention also the Plugin Version. 

Best regards
Patrick

pipiche38, 2019-01-14:
Close. Use of Forum channels for such question

closed 2019-01-14 by pipiche38