zigbeefordomoticz/Domoticz-Zigbee #323: Osram Led strip reset failed

Cortexlegeni, 2019-01-14:
Hello,
I recently baught 3 Lightify Osram strip from Amazon, and they are very well recognised in Domoticz, thanks. I had to update the firmware and plugins from older version to version 4 so I did a full reset of the device list.
I managed to reset my light strip pluging and unpluging five time and then having it recognised again.
But since then, sometime, It turns itself on without me doing anything. I thought it was coming from the strip itself so I disabled the plugin for one day and it didn't append again. But 20 minuts ago, I turned it back on and then here again, my led is already on.
Here is the log, tell me if you nedd anything more.

`2019-01-13 23:07:45.592 Status: (Zigate) Stopping threads.
2019-01-13 23:07:45.592 Status: (Zigate) Stopped.
2019-01-14 00:00:00.061 Sunrise: 08:41:00 SunSet: 17:28:00
2019-01-14 00:00:00.061 Day length: 08:47:00 Sun at south: 13:05:00
2019-01-14 00:00:00.061 Civil twilight start: 08:06:00 Civil twilight end: 18:03:00
2019-01-14 00:00:00.061 Nautical twilight start: 07:27:00 Nautical twilight end: 18:42:00
2019-01-14 00:00:00.061 Astronomical twilight start: 06:50:00 Astronomical twilight end: 19:19:00
2019-01-14 00:00:00.061 Status: EventSystem: reset all events...
2019-01-14 21:25:24.646 Status: (Zigate) Started.
2019-01-14 21:25:24.647 Status: Incoming connection from: 192.168.1.36
2019-01-14 21:25:25.098 (Zigate) updateNotificationWidget - didn't find the Widget: Zigate-02-
2019-01-14 21:25:25.099 (Zigate) PluginConf.txt = {'channel':'11,15,16,21,22','self.eraseZigatePDM':'0','vibrationAqarasensitivity':'medium','resetMotiondelay':'30','enablegroupmanagement':'0','allowRemoveZigateDevice':'0','allowStoreDiscoveryFrames':'0','logLQI':'0','logFORMAT':'0'}
2019-01-14 21:25:25.100 (Zigate) Device Management:
2019-01-14 21:25:25.100 (Zigate) -allowStoreDiscoveryFrames : 0
2019-01-14 21:25:25.100 (Zigate) -allowForceCreationDomoDevice: 0
2019-01-14 21:25:25.100 (Zigate) -forceConfigureReporting: 0
2019-01-14 21:25:25.100 (Zigate) -forceReadAttributes: 0
2019-01-14 21:25:25.101 (Zigate) -resetMotiondelay: 30
2019-01-14 21:25:25.101 (Zigate) -vibrationAqarasensitivity: medium
2019-01-14 21:25:25.101 (Zigate) Zigate Configuration
2019-01-14 21:25:25.101 (Zigate) -channel: ['11', '15', '16', '21', '22']
2019-01-14 21:25:25.101 (Zigate) -allowRemoveZigateDevice: 0
2019-01-14 21:25:25.101 (Zigate) -eraseZigatePDM: 0
2019-01-14 21:25:25.101 (Zigate) Plugin Transport
2019-01-14 21:25:25.101 (Zigate) -zmode: ZigBee
2019-01-14 21:25:25.101 (Zigate) -reTransmit: 1
2019-01-14 21:25:25.101 (Zigate) -zTimeOut: 2
2019-01-14 21:25:25.101 (Zigate) -CrcCheck: 1
2019-01-14 21:25:25.101 (Zigate) -sendDelay: 0
2019-01-14 21:25:25.101 (Zigate) Plugin Directories
2019-01-14 21:25:25.101 (Zigate) -pluginHome: /home/pi/domoticz/plugins/Domoticz-Zigate/
2019-01-14 21:25:25.101 (Zigate) -pluginData: /home/pi/domoticz/plugins/Domoticz-Zigate/./
2019-01-14 21:25:25.101 (Zigate) -pluginZData: /home/pi/domoticz/plugins/Domoticz-Zigate/./Zdatas/
2019-01-14 21:25:25.101 (Zigate) -pluginReports: /home/pi/domoticz/plugins/Domoticz-Zigate/../../www/templates/zigate/reports/
2019-01-14 21:25:25.101 (Zigate) -pluginConfig: /home/pi/domoticz/plugins/Domoticz-Zigate/./
2019-01-14 21:25:25.101 (Zigate) -pluginWWW: /home/pi/domoticz/plugins/Domoticz-Zigate/../../www/templates/
2019-01-14 21:25:25.102 (Zigate) -homedirectory: /home/pi/domoticz/plugins/Domoticz-Zigate/
2019-01-14 21:25:25.102 (Zigate) -filename: /home/pi/domoticz/plugins/Domoticz-Zigate/PluginConf.txt
2019-01-14 21:25:25.102 (Zigate) Groups Management
2019-01-14 21:25:25.102 (Zigate) -enablegroupmanagement: 0
2019-01-14 21:25:25.102 (Zigate) -discoverZigateGroups: 1
2019-01-14 21:25:25.102 (Zigate) -enableConfigGroups: 1
2019-01-14 21:25:25.102 (Zigate) Reportings and Statistics
2019-01-14 21:25:25.102 (Zigate) -logLQI: 0
2019-01-14 21:25:25.102 (Zigate) -networkScan: 0
2019-01-14 21:25:25.126 (Zigate) CheckDeviceList - DeviceID (IEEE) = 00158d0001ab3c00 for NetworkID = b43b
2019-01-14 21:25:25.130 (Zigate) CheckDeviceList - DeviceID (IEEE) = 00158d0002272d96 for NetworkID = 8352
2019-01-14 21:25:25.134 (Zigate) CheckDeviceList - DeviceID (IEEE) = 00158d0002272f3d for NetworkID = ccd5
2019-01-14 21:25:25.142 (Zigate) CheckDeviceList - DeviceID (IEEE) = 7cb03eaa00a99fd7 for NetworkID = 1502
2019-01-14 21:25:25.145 (Zigate) CheckDeviceList - DeviceID (IEEE) = 00158d0001b8f82c for NetworkID = 7378
2019-01-14 21:25:25.149 (Zigate) CheckDeviceList - DeviceID (IEEE) = 00158d0002273b7a for NetworkID = a751
2019-01-14 21:25:25.150 (Zigate) ListOfDevices :
2019-01-14 21:25:25.150 (Zigate) ('b43b', {'App Version': '3', 'Model': 'lumi.weather', 'SQN': '00', 'Status': 'inDB', 'Stamp': {'Time': '2019-01-13 21:56:33', 'MsgType': '8102'}, 'ZDeviceID': '5f01', 'DeviceType': 'RFD', 'MacCapa': '80', 'Heartbeat': '8640', 'ReceiveOnIdle': 'Off', 'ProfileID': '0104', 'Ep': {'01': {'Type': 'Temp/Humi/Baro', '0405': 49.0, '0004': {}, 'ClusterType': {'20': 'Temp', '21': 'Humi', '19': 'Temp+Hum', '18': 'Temp+Hum+Baro', '22': 'Baro'}, '0000': {}, 'ffff': {}, '0003': {}, '0403': 997.9, '0402': 20.6}}, 'ReadAttributes': {'TimeStamps': {'01-0000': 1547415750}, 'Ep': {'01': {'0000': {'0000': '00', 'ff01': '86'}}}}, 'RIA': '0', 'PowerSource': 'Battery', 'Version': '3', 'IEEE': '00158d0001ab3c00', 'RSSI': 124, 'NbEp': '1', 'Type': '', 'LogicalType': 'End Device', 'Manufacturer': '1037', 'Battery': 91})
2019-01-14 21:25:25.150 (Zigate) ('7378', {'App Version': '3', 'Version': '3', 'Model': 'lumi.sensor_switch.aq2', 'SQN': '35', 'Status': 'inDB', 'Stamp': {'Time': '2019-01-13 22:55:35', 'MsgType': '8102'}, 'ZDeviceID': '5f01', 'Type': 'SwitchAQ2', 'RSSI': 147, 'NbEp': '1', 'Heartbeat': '8874', 'ProfileID': '0104', 'IEEE': '00158d0001b8f82c', 'Battery': 95, 'Ep': {'01': {'0000': {}, 'ffff': {}, 'ClusterType': {'16': 'SwitchAQ2'}, '0004': {}, '0006': '02'}}, 'ReadAttributes': {'TimeStamps': {'01-0000': 1547415369}, 'Ep': {'01': {'0000': {'0000': '00', 'ff01': '86'}}}}, 'RIA': '0', 'MacCapa': '80'})
2019-01-14 21:25:25.151 (Zigate) ('ccd5', {'App Version': '3', 'Version': '3', 'Model': 'lumi.weather', 'SQN': '00', 'Status': 'inDB', 'Stamp': {'Time': '2019-01-13 21:19:13', 'MsgType': '8102'}, 'ZDeviceID': '5f01', 'Type': '', 'RSSI': 75, 'NbEp': '1', 'Heartbeat': '8624', 'ProfileID': '0104', 'IEEE': '00158d0002272f3d', 'Battery': 92, 'Ep': {'01': {'Type': 'Temp/Humi/Baro', '0405': 50.8, '0004': {}, 'ClusterType': {'29': 'Temp+Hum', '31': 'Humi', '28': 'Temp+Hum+Baro', '32': 'Baro', '30': 'Temp'}, '0000': {}, 'ffff': {}, '0003': {}, '0403': 998.2, '0402': 19.3}}, 'ReadAttributes': {'TimeStamps': {'01-0000': 1547415830}, 'Ep': {'01': {'0000': {'0000': '00', 'ff01': '86'}}}}, 'RIA': '0', 'MacCapa': '80'})
2019-01-14 21:25:25.151 (Zigate) ('1502', {'App Version': '1', 'Model': 'LIGHTIFY Outdoor Flex RGBW', 'SQN': {}, 'Status': 'inDB', 'Stamp': {'Time': '2019-01-13 23:07:43', 'MsgType': '8102'}, 'ZDeviceID': '0210', 'DeviceType': 'FFD', 'MacCapa': '8e', 'Heartbeat': '2', 'HW Version': '1', 'ReceiveOnIdle': 'On', 'ProfileID': 'c05e', 'Stack Version': '2', 'Ep': {'03': {'0019': {}, '0004': {}, '1000': {}, 'ClusterType': {'17': 'ColorControl'}, 'fc0f': {}, '0006': '00', '0005': {}, '0000': {}, '0008': 'fe', '0300': {}, '0003': {}}}, 'ReadAttributes': {'TimeStamps': {'03-0300': 1547417133, '03-0000': 1547416652, '03-0006': 1547417133, '03-0008': 1547417133}, 'Ep': {'03': {'0000': {'0005': '00', 'ff01': '86', '0004': '00', '0000': '00', '0001': '00', '0002': '00', '0007': '00', '0010': '86', '0003': '00'}, '0008': {'0000': '00'}, '0300': {'0005': '86', '0004': '00', '0006': '86', '0001': '00', '0000': '00', '0008': '00', '0007': '00', '0003': '00'}, '0006': {'0000': '00'}}}}, 'RIA': '2', 'PowerSource': 'Main', 'Version': '3', 'IEEE': '7cb03eaa00a99fd7', 'RSSI': 81, 'ConfigureReporting': {'TimeStamps': {'03-0300': 1547417218, '03-0004': 0, '03-0005': 0, '03-0006': 1547417158, '03-1000': 0, '03-0000': 0, '03-fc0f': 0, '03-0019': 0, '03-0003': 0, '03-0008': 1547417218}, 'Ep': {'03': {'0019': {}, '0004': {}, '1000': '', 'fc0f': {}, '0006': '00', '0005': {}, '0000': {}, '0008': '00', '0300': {}, '0003': {}}}}, 'NbEp': '1', 'Type': 'Switch/ColorControl/LvlControl/', 'LogicalType': 'Router', 'Manufacturer': 'bbaa', 'Battery': {}, 'ColorInfos': {'ColorMode': 2, 'Saturation': 0, 'Y': '24701', 'ColorTemperatureMireds': '313', 'X': '24939', 'Hue': 0}})
[... cut ...]
2019-01-14 21:27:30.134 (Zigate) configureReporting for [1502] - cluster: 0300 on Attribute: ['0004', '0001', '0000', '0008', '0007', '0003'] >0215020103030000000000060021000303840E100FFF010021000303840E100FFF01<
2019-01-14 21:27:47.287 (Zigate) Update v2 Values 1:'On:' (Led_Cheminee)
2019-01-14 21:27:51.277 (Zigate) Update v2 Values 1:'100:' (Led_Cheminee)
2019-01-14 21:29:14.489 (Zigate) Out of sequence for Device: 7378 SQN move from 35 to 50 gap of : 27
2019-01-14 21:29:14.998 (Zigate) Out of sequence for Device: 7378 SQN move fro`

pipiche38, 2019-01-14:
Looks like you are using the Beta branch.

I don't really see the Plugin switching on the device without any action . 

From the log given the Led corresponding device has been updated to On and 100% at 21:27 on Domoticz widget.

What could happen is that the Led is reacting to queries ( Read Attributes / Configure Reporting )coming from the Plugin -> Zigate

Cortexlegeni, 2019-01-14:
Hello again, Yes, I'm on the beta branch, I can switch to the master branch if needed.
the thing is that I'm doing nothing in from domoticz, but I have a block script with this device in, I'm gonna try to remove it and see for the night.

Cortexlegeni, 2019-01-14:
And I'm asking myself what are these lines for : 

> 2019-01-14 23:28:31.901 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-14 23:28:33.508 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-14 23:28:35.115 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-14 23:28:36.686 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7

Cortexlegeni, 2019-01-15:
I left it without scene for the night and it's on this morning, here is the log:

> 2019-01-14 21:54:56.987 Status: (Zigate) [0] NEW OBJECT: ccd5 Device Annouce IEEE: 00158d0002272f3d
2019-01-14 21:54:56.989 Status: (Zigate) [0] NEW OBJECT: ccd5 Device Annouce IEEE: 00158d0002272f3d
2019-01-14 21:55:30.467 Status: (Zigate) Temp_Leandre - Last Update from Battery device was: 2019-01-13 21:56:33 --> 00:58:57 ago (More than 2 hours) 86337
2019-01-14 22:25:30.901 Status: (Zigate) Decode8009 : Zigate addresses ieee: 00158d0001b696bd , short addr: 0000
2019-01-14 22:25:30.901 Status: (Zigate) Decode8009 - Network state UP, PANID: 0f91 extPANID: 0xc9e8d9885c090c20 Channel: 11
2019-01-14 22:28:31.253 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-14 22:28:32.861 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-14 22:28:34.468 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-14 22:28:36.090 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-14 23:24:03.942 Status: Incoming connection from: 192.168.1.36
2019-01-14 23:25:31.579 Status: (Zigate) Decode8009 : Zigate addresses ieee: 00158d0001b696bd , short addr: 0000
2019-01-14 23:25:31.579 Status: (Zigate) Decode8009 - Network state UP, PANID: 0f91 extPANID: 0xc9e8d9885c090c20 Channel: 11
2019-01-14 23:28:31.901 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-14 23:28:33.508 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-14 23:28:35.115 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-14 23:28:36.686 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-14 23:32:27.423 Status: EventSystem: reset all events...
2019-01-14 23:32:49.343 Status: EventSystem: reset all events...
2019-01-14 23:33:03.297 Status: EventSystem: reset all events...
2019-01-14 23:33:11.290 Status: EventSystem: reset all events...
2019-01-15 00:00:00.152 Status: EventSystem: reset all events...
2019-01-15 00:25:32.253 Status: (Zigate) Decode8009 : Zigate addresses ieee: 00158d0001b696bd , short addr: 0000
2019-01-15 00:25:32.253 Status: (Zigate) Decode8009 - Network state UP, PANID: 0f91 extPANID: 0xc9e8d9885c090c20 Channel: 11
2019-01-15 00:28:32.588 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-15 00:28:34.194 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-15 00:28:35.802 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-15 01:25:32.913 Status: (Zigate) Decode8009 : Zigate addresses ieee: 00158d0001b696bd , short addr: 0000
2019-01-15 01:25:32.913 Status: (Zigate) Decode8009 - Network state UP, PANID: 0f91 extPANID: 0xc9e8d9885c090c20 Channel: 11
2019-01-15 01:29:33.249 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-15 01:29:34.856 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-15 01:29:36.464 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-15 01:29:38.085 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-15 02:25:33.592 Status: (Zigate) Decode8009 : Zigate addresses ieee: 00158d0001b696bd , short addr: 0000
2019-01-15 02:25:33.592 Status: (Zigate) Decode8009 - Network state UP, PANID: 0f91 extPANID: 0xc9e8d9885c090c20 Channel: 11
2019-01-15 02:29:33.942 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-15 02:29:35.549 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-15 02:29:37.156 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-15 02:29:38.778 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-15 03:25:34.263 Status: (Zigate) Decode8009 : Zigate addresses ieee: 00158d0001b696bd , short addr: 0000
2019-01-15 03:25:34.263 Status: (Zigate) Decode8009 - Network state UP, PANID: 0f91 extPANID: 0xc9e8d9885c090c20 Channel: 11
2019-01-15 03:29:34.582 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-15 03:29:36.190 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-15 03:29:37.797 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-15 03:29:39.419 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-15 04:23:10.119 (Zigate) Update v2 Values 0:'19.2;50.7;1:' (Zigate - Temp+Hum-00158d0002272f3d-01)
2019-01-15 04:23:10.125 (Zigate) Update v2 Values 0:'19.2;50.7;1;1006.5;3:' (Temp_Agathe)
2019-01-15 04:23:10.134 (Zigate) Update v2 Values 0:'1006.5;3:' (Zigate - Baro-00158d0002272f3d-01)
2019-01-15 04:23:10.641 (Zigate) Out of sequence for Device: ccd5 SQN move from cf to f1 gap of : 34
2019-01-15 04:25:34.911 Status: (Zigate) Decode8009 : Zigate addresses ieee: 00158d0001b696bd , short addr: 0000
2019-01-15 04:25:34.911 Status: (Zigate) Decode8009 - Network state UP, PANID: 0f91 extPANID: 0xc9e8d9885c090c20 Channel: 11
2019-01-15 04:28:15.715 (Zigate) Out of sequence for Device: 8352 SQN move from 00 to 9b gap of : 155
2019-01-15 04:28:15.717 (Zigate) ReadCluster - 0000/ff01 Saddr: 8352 Temperature : 19.2
2019-01-15 04:28:15.718 (Zigate) Update v2 Values 0:'19.2;49.2;1;1006.5;3:' (Temp_Capucine)
[... cut ...]
2019-01-15 05:42:21.771 (Zigate) Out of sequence for Device: 7378 SQN move from 1d to 5b gap of : 62
2019-01-15 05:42:21.772 (Zigate) Update v2 Values 2:'10:' (Btn_Led_Cheminee)
2019-01-15 05:42:24.847 (Zigate) Update v2 Values 2:'10:' (Btn_Led_Cheminee)
2019-01-15 06:01:56.771 Status: Incoming connection from: 192.168.1.36
2019-01-15 06:02:02.830 (Zigate) Update v2 Values 0:'Off:' (Led_Cheminee)
2019-01-15 06:02:02.780 Status: User: Admin initiated a switch command (17/Led_Cheminee/Off)

I turned it off manualy

pipiche38, 2019-01-15:
> 2019-01-14 23:28:31.901 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
> 2019-01-14 23:28:33.508 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: [...]
As regards to that, it looks like you have a device 7cb03eaa00a99fd7 which is making device announcement to ( LIGHTIFY Outdoor Flex RGBW ) . Usually  it is what happen when doing a Reset or a Main power off/on.

Here attached is a patch which do not proceed with ConfigureReporting. 
You need to unzip and the replace the existing one by that one.

this file is located under Domoticz-Zigate/Modules ( in beta version )

[z_heartbeat.py.zip](https://github.com/sasu-drooz/Domoticz-Zigate/files/2758414/z_heartbeat.py.zip)

Let me know the end result, if the switch on is not happening, we will then focus on the activities done around that area.

Cortexlegeni, 2019-01-15:
I'm at work and will do it this evening, expect new at about 9PM (France time)

Cortexlegeni, 2019-01-16:
File modified at 22h00 and no problem for the moment.

Cortexlegeni, 2019-01-16:
No problem for editing files or anything, I'm a computer science teacher so I think I will manage.
I just don't have much time with 3 kids else I would very gladly help you on the project.
But if you need anything, ask, I'll do as much as I can.

pipiche38, 2019-01-16:
@Cortexlegeni , While looking after this area, I found some potential issues. 
Don't know if they could be the root cause of your problem (as it looks like no-body complains), but the code I pushed on beta is including some update+enhancement. 

So if you can 'git pull' and of course overwrite the z_heartbeat.py patched version it would be great

Cortexlegeni, 2019-01-16:
problem seems to still be there :

> 2019-01-16 19:50:11.065 Status: (Zigate) load ListOfDevice
2019-01-16 19:50:11.107 Status: (Zigate) Entries loaded from /home/pi/domoticz/plugins/Domoticz-Zigate/./DeviceList-4.txt : 6
2019-01-16 19:50:11.125 Status: (Zigate) Check for Web update finished!
2019-01-16 19:50:11.125 Status: (Zigate) Connection Name: Zigate, Transport: Serial, Address: /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
2019-01-16 19:50:11.130 Status: (Zigate) ZigateConf - Setting UTC Time to : 600983411
2019-01-16 19:50:11.130 Status: (Zigate) Start network
2019-01-16 19:50:11.130 Status: (Zigate) Zigate enter in discover mode for 254 Secs
2019-01-16 19:50:11.232 Status: (Zigate) Major Version Num: 0002
2019-01-16 19:50:11.232 Status: (Zigate) Installer Version Number: 030e
2019-01-16 19:50:11.336 Status: (Zigate) Decode8015 : Number of devices recently active in Zigate = 2
2019-01-16 19:50:11.336 Status: (Zigate) Decode8015 : [00] DevID = 00 Network addr = 7378 IEEE = 00158d0001b8f82c RSSI = 173 Power = 00 HB = 25048 found in ListOfDevices
2019-01-16 19:50:11.336 Status: (Zigate) Decode8015 : [01] DevID = 01 Network addr = 1502 IEEE = 7cb03eaa00a99fd7 RSSI = 082 Power = 01 HB = 505 found in ListOfDevices
2019-01-16 19:50:21.328 (Zigate) Start Network status : Success
2019-01-16 19:50:21.532 (Zigate) Ping - received
2019-01-16 19:50:21.225 Status: (Zigate) Decode8009 : Zigate addresses ieee: 00158d0001b696bd , short addr: 0000
2019-01-16 19:50:21.225 Status: (Zigate) Decode8009 - Network state UP, PANID: 0f91 extPANID: 0xc9e8d9885c090c20 Channel: 11
2019-01-16 19:50:21.532 Status: (Zigate) Permit Join is On
2019-01-16 19:50:54.938 (Zigate) processListOfDevices - Device: ccd5 is in Status = 'Left' for 24804HB
2019-01-16 19:51:20.191 (Zigate) Update v2 Values 1:'On:' (Led_Cheminee)
2019-01-16 19:51:20.363 (Zigate) Update v2 Values 1:'100:' (Led_Cheminee)
2019-01-16 19:51:20.153 Status: User: Admin initiated a switch command (17/Led_Cheminee/On)
2019-01-16 19:51:24.241 (Zigate) Update v2 Values 0:'Off:' (Led_Cheminee)
2019-01-16 19:51:24.964 (Zigate) Processing configureReporting for 1502 cluuster: 0006
2019-01-16 19:51:24.964 (Zigate) bindDevice - ieee: 7cb03eaa00a99fd7, ep: 03, cluster: 0006, dest_ieee: 00158d0001b696bd, desk_ep: 01
2019-01-16 19:51:24.966 (Zigate) Processing configureReporting for 1502 cluuster: 0008
2019-01-16 19:51:24.966 (Zigate) bindDevice - ieee: 7cb03eaa00a99fd7, ep: 03, cluster: 0008, dest_ieee: 00158d0001b696bd, desk_ep: 01
2019-01-16 19:51:24.967 (Zigate) Processing configureReporting for 1502 cluuster: 0300
2019-01-16 19:51:24.968 (Zigate) configureReporting - skip configureReporting for now ... system too busy (False/3) for 1502
2019-01-16 19:51:24.980 (Zigate) Ping - We have receive a message in between 2 Heartbeat
2019-01-16 19:51:24.216 Status: User: Admin initiated a switch command (17/Led_Cheminee/Off)
2019-01-16 19:52:24.953 (Zigate) Processing configureReporting for 1502 cluuster: 0300
2019-01-16 19:52:24.953 (Zigate) bindDevice - ieee: 7cb03eaa00a99fd7, ep: 03, cluster: 0300, dest_ieee: 00158d0001b696bd, desk_ep: 01
2019-01-16 19:52:24.967 (Zigate) Ping - We have receive a message less than 1' ago
2019-01-16 19:52:25.375 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-16 19:52:26.983 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-16 19:52:28.592 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-16 19:52:30.162 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-16 19:52:34.980 (Zigate) processListOfDevices - Device: b43b is in Status = 'Left' for 24840HB
2019-01-16 19:53:25.004 (Zigate) Ping - We have receive a message less than 1' ago
2019-01-16 19:53:54.999 (Zigate) processListOfDevices - Device: ccd5 is in Status = 'Left' for 24840HB
2019-01-16 19:54:25.021 (Zigate) Ping - We didn't receive any messages since 60s
2019-01-16 19:54:25.021 (Zigate) Ping - Send a Ping
2019-01-16 19:55:25.038 (Zigate) Ping - We have receive a message less than 1' ago
2019-01-16 19:55:35.027 (Zigate) processListOfDevices - Device: b43b is in Status = 'Left' for 24876HB
2019-01-16 19:56:25.053 (Zigate) Ping - We didn't receive any messages since 60s
2019-01-16 19:56:25.053 (Zigate) Ping - Send a Ping
2019-01-16 19:56:25.156 (Zigate) Ping - received
2019-01-16 19:56:55.006 (Zigate) processListOfDevices - Device: ccd5 is in Status = 'Left' for 24876HB
2019-01-16 19:57:25.029 (Zigate) Ping - We have receive a message less than 1' ago
2019-01-16 19:57:31.164 (Zigate) Update v2 Values 1:'On:' (Led_Cheminee)
2019-01-16 19:57:35.160 (Zigate) Update v2 Values 1:'100:' (Led_Cheminee)

pipiche38, 2019-01-16:
Here after is the interesting part and which confirm the suspicion.

So it is either the Bind or the ConfigureReporting which is creating the issue. And it looks like it is making the Led to 'reboot/reset' as it create a Device Annouce

```
2019-01-16 19:51:24.216 Status: User: Admin initiated a switch command (17/Led_Cheminee/Off)
2019-01-16 19:52:24.953 (Zigate) Processing configureReporting for 1502 cluuster: 0300
2019-01-16 19:52:24.953 (Zigate) bindDevice - ieee: 7cb03eaa00a99fd7, ep: 03, cluster: 0300, dest_ieee: 00158d0001b696bd, desk_ep: 01
2019-01-16 19:52:24.967 (Zigate) Ping - We have receive a message less than 1' ago
2019-01-16 19:52:25.375 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-16 19:52:26.983 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-16 19:52:28.592 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
2019-01-16 19:52:30.162 Status: (Zigate) [0] NEW OBJECT: 1502 Device Annouce IEEE: 7cb03eaa00a99fd7
```

pipiche38, 2019-01-17:
There are 2 possibilties:
(1) the cluster binding is creating an issue
(2) the configureReporting is creating an issue (and could be related to the fact that we might query some non-existing attributes)

as this should be an iterative work, if you can help this would be really covenient.

If you have the possibility to debug it would be really helpful. Here are some indications

Assuming you are on the latest beta version,

(1) Edit the file PluginConf.txt and add the following line
`'forceConfigureReporting':'1',`
Be careful not to break the syntax of the file. So just add this line in the middle of the file. (coma is needed at the end of the line, except if this is the last line before th closing bracket) .
This will force to restart ConfigureReporting at start

(2) Edit the file Modules/z_heartbeat.py and add the following 4 lines in the 'processListOfDevices()' method ?
Should be line 304 (jlust after self.ZigateComm.checkTOwaitFor() )
```
    intHB = int( self.ListOfDevices[NWKID]['Heartbeat'])
    if (intHB % ( 120 // HEARTBEAT) == 0:
            if '7cb03eaa00a99fd7' in self.IEEE2NWK:
             processConfigureReporting( self, self.IEEE2NWK['7cb03eaa00a99fd7'] )
```
This will make a configurereporting call every 2 minutes . You might want to adjust the 120
1. Line 567 there is a call to establish a cluster binding. You can also test if putting it in comment, what is the impact. It looks like for some devices it is not needed, while for some other it is a must

1. The issue is that there is a protection in ConfigureReporting which avoid to do a call to the device so often ( line 556 of Modules/z_output.py) Might be good to remove that test for now.

1. Line 464 of Modules/z_output.py  there are the list of attributes we are querying. It might be good to remove all of them and start with only attribute 0000.

pipiche38, 2019-02-07:
@Cortexlegeni have you been able to investigate ?

pipiche38, 2019-02-07:
If you take the latest beta version, I have remove the bind in case of LIGHTIFY Outdoor Flex RGBW.
Let me know what is the end result

pipiche38, 2019-02-07:
824e197a744120491f890fbc3bcabdaa367c9b14

Cortexlegeni, 2019-02-07:
Hello, I am a little busy in this moment because of my children being ill sa I don't have muche time.
I will find some time next week or the week later.
Sorry

pipiche38, 2019-02-25:
@Cortexlegeni , if you want to try. I have updated pre-4.1 and I believe your issue is now fix.
Please remove the Led strip from Domoticz and then redo the full pairing process

Seems that the issue was related to a Bind which make ALL osram devices reset

pipiche38, 2019-03-04:
@Cortexlegeni , I suggest to close this issue for now. When you have time please test it with latest pre-4.1 and if the problem still exist, re-open the issue

Cortexlegeni, 2019-06-13:
Hello, I'm back with the same problem. I tried the new plugin with a fresh firmware update and full reset of the zigate and when I tried to reset the LED strip  I have this :

> 2019-06-12 22:15:37.406 Error: (Zigate) onMessage : Frame size is bad, computed = 12 received = 21
2019-06-12 22:15:37.406 Error: (Zigate) onMessage : Frame CRC is bad, computed = 105 received = 151
2019-06-12 22:15:55.460 Error: (Zigate) onMessage : Frame size is bad, computed = 12 received = 32
2019-06-12 22:15:55.460 Error: (Zigate) onMessage : Frame CRC is bad, computed = 106 received = 148
2019-06-13 21:01:23.090 Status: User: Admin initiated a switch command (5/Prise-Osram-01/On)
2019-06-13 21:01:25.763 Status: User: Admin initiated a switch command (5/Prise-Osram-01/Off)
2019-06-13 21:01:28.401 Status: User: Admin initiated a switch command (5/Prise-Osram-01/On)
2019-06-13 21:01:31.021 Status: User: Admin initiated a switch command (5/Prise-Osram-01/Off)
2019-06-13 21:01:33.155 Status: User: Admin initiated a switch command (5/Prise-Osram-01/On)
2019-06-13 21:01:53.366 Status: (Zigate) Device Annoucement ShortAddr: d996, IEEE: 7cb03eaa00a99fd7
2019-06-13 21:01:57.041 Status: (Zigate) [-] NEW OBJECT: d996 Request Simple Descriptor for Ep: 03
2019-06-13 21:01:57.203 Status: (Zigate) [-] NEW OBJECT: d996 Simple Descriptor EP 03
2019-06-13 21:01:57.203 Status: (Zigate) [-] NEW OBJECT: d996 ProfileID c05e
2019-06-13 21:01:57.203 Status: (Zigate) [-] NEW OBJECT: d996 ZDeviceID 0210
2019-06-13 21:01:57.204 Status: (Zigate) [-] NEW OBJECT: d996 Cluster IN Count: 09
2019-06-13 21:01:57.204 Status: (Zigate) [-] NEW OBJECT: d996 Cluster In 1: 1000 (Touchlink)
2019-06-13 21:01:57.204 Status: (Zigate) [-] NEW OBJECT: d996 Cluster In 2: 0000 (Basic)
2019-06-13 21:01:57.204 Status: (Zigate) [-] NEW OBJECT: d996 Cluster In 3: 0003 (Identify)
2019-06-13 21:01:57.204 Status: (Zigate) [-] NEW OBJECT: d996 Cluster In 4: 0004 (Groups)
2019-06-13 21:01:57.204 Status: (Zigate) [-] NEW OBJECT: d996 Cluster In 5: 0005 (Scenes)
2019-06-13 21:01:57.204 Status: (Zigate) [-] NEW OBJECT: d996 Cluster In 6: 0006 (On/Off)
2019-06-13 21:01:57.205 Status: (Zigate) [-] NEW OBJECT: d996 Cluster In 7: 0008 (Level Control)
2019-06-13 21:01:57.205 Status: (Zigate) [-] NEW OBJECT: d996 Cluster In 8: 0300 (Colour Control)
2019-06-13 21:01:57.205 Status: (Zigate) [-] NEW OBJECT: d996 Cluster In 9: fc0f
2019-06-13 21:01:57.205 Status: (Zigate) [-] NEW OBJECT: d996 Cluster OUT Count: 01
2019-06-13 21:01:57.205 Status: (Zigate) [-] NEW OBJECT: d996 Cluster Out 1: 0019 (Over-the-Air Upgrade)
2019-06-13 21:02:01.916 Status: (Zigate) [0] NEW OBJECT: d996 Request Model Name
2019-06-13 21:02:01.919 Status: (Zigate) [0] NEW OBJECT: d996 Request Attribute for Cluster 0x0300 to get ColorMode
2019-06-13 21:02:06.908 Status: (Zigate) [1] NEW OBJECT: d996 Model Name: LIGHTIFY Outdoor Flex RGBW
2019-06-13 21:02:06.908 Status: (Zigate) [1] NEW OBJECT: d996 Model Name: LIGHTIFY Outdoor Flex RGBW
2019-06-13 21:02:06.908 Status: (Zigate) [1] NEW OBJECT: d996 Model Name: LIGHTIFY Outdoor Flex RGBW
2019-06-13 21:02:11.923 Status: (Zigate) [2] NEW OBJECT: d996 Model Name: LIGHTIFY Outdoor Flex RGBW
2019-06-13 21:02:11.923 Status: (Zigate) [2] NEW OBJECT: d996 Model Name: LIGHTIFY Outdoor Flex RGBW
2019-06-13 21:02:11.923 Status: (Zigate) [2] NEW OBJECT: d996 Model Name: LIGHTIFY Outdoor Flex RGBW
2019-06-13 21:02:15.370 Status: User: Admin initiated a switch command (5/Prise-Osram-01/Off)
2019-06-13 21:02:17.462 Status: User: Admin initiated a switch command (5/Prise-Osram-01/On)
2019-06-13 21:02:18.631 Status: (Zigate) Device Annoucement ShortAddr: d996, IEEE: 7cb03eaa00a99fd7
2019-06-13 21:02:19.644 Status: User: Admin initiated a switch command (5/Prise-Osram-01/Off)
2019-06-13 21:02:22.557 Status: User: Admin initiated a switch command (5/Prise-Osram-01/On)
2019-06-13 21:02:23.697 (Zigate) Decode004d - rebind clusters for d996
2019-06-13 21:02:23.697 Status: (Zigate) Device Annoucement ShortAddr: d996, IEEE: 7cb03eaa00a99fd7
2019-06-13 21:02:24.883 (Zigate) UpdateDevice - ( Prise-Osram-01) 0:Off
2019-06-13 21:02:24.865 Status: User: Admin initiated a switch command (5/Prise-Osram-01/Off)
2019-06-13 21:02:25.071 (Zigate) ReadCluster 0000 - d996/03 Unknown attribute: 0012 Value: 01
2019-06-13 21:02:26.949 (Zigate) UpdateDevice - ( Prise-Osram-01) 1:On
2019-06-13 21:02:26.916 Status: User: Admin initiated a switch command (5/Prise-Osram-01/On)
2019-06-13 21:02:28.071 (Zigate) Decode004d - rebind clusters for d996
2019-06-13 21:02:28.070 Status: (Zigate) Device Annoucement ShortAddr: d996, IEEE: 7cb03eaa00a99fd7
2019-06-13 21:02:29.272 (Zigate) ReadCluster 0000 - d996/03 Unknown attribute: 0012 Value: 01
2019-06-13 21:02:29.529 (Zigate) UpdateDevice - ( Prise-Osram-01) 0:Off
2019-06-13 21:02:29.511 Status: User: Admin initiated a switch command (5/Prise-Osram-01/Off)
2019-06-13 21:02:31.702 (Zigate) UpdateDevice - ( Prise-Osram-01) 1:On
2019-06-13 21:02:31.677 Status: User: Admin initiated a switch command (5/Prise-Osram-01/On)
2019-06-13 21:02:32.779 (Zigate) Decode004d - rebind clusters for d996
2019-06-13 21:02:32.779 Status: (Zigate) Device Annoucement ShortAddr: d996, IEEE: 7cb03eaa00a99fd7
2019-06-13 21:02:33.957 (Zigate) Decode004d - rebind clusters for d996
2019-06-13 21:02:33.957 Status: (Zigate) Device Annoucement ShortAddr: d996, IEEE: 7cb03eaa00a99fd7
2019-06-13 21:02:34.315 (Zigate) UpdateDevice - ( Prise-Osram-01) 0:Off
[... cut ...]
2019-06-13 21:08:10.315 Status: setcolbrightnessvalue: ID: 6, bri: 98, color: '{m: 2, RGB: 000000, CWWW: be41, CT: 65}'
2019-06-13 21:08:12.175 (Zigate) UpdateDevice - (LED-Osram-Cheminee) 1:fa
2019-06-13 21:08:12.157 Status: setcolbrightnessvalue: ID: 6, bri: 98, color: '{m: 2, RGB: 000000, CWWW: cc33, CT: 51}'
2019-06-13 21:08:15.756 (Zigate) UpdateDevice - (LED-Osram-Cheminee) 1:fa
2019-06-13 21:08:15.738 Status: setcolbrightnessvalue: ID: 6, bri: 98, color: '{m: 3, RGB: 41ff29, CWWW: 0000, CT: 0}'

>

pipiche38, 2019-06-13:
You need to disable allowReBindingClusters.

if you are on 'stable' branch you need to update Conf/PluginConf.txt and add/update the line
'allowReBindingClusters':'1' ----> 'allowReBindingClusters':'0',

https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/PluginConf.txt.md

pipiche38, 2019-06-20:
Did you disable allowReBindingClusters ? is that solving your issue ?

Cortexlegeni, 2019-06-24:
Hello, I tried and it solved the problem for some time, but I still have every 3/4 days a problem, the strim switch on and don't answer after that.
Here is the log:

> 2019-06-20 21:00:43.930 Status: Domoticz V4.10897 (c)2012-2019 GizMoCuz
2019-06-20 21:00:43.930 Status: Build Hash: 9f2dcd957, Date: 2019-06-12 09:15:45
2019-06-20 21:00:43.931 Status: Startup Path: /home/pi/domoticz/
2019-06-20 21:00:44.603 Sunrise: 05:55:00 SunSet: 21:56:00
2019-06-20 21:00:44.604 Day length: 16:00:00 Sun at south: 13:56:00
2019-06-20 21:00:44.604 Civil twilight start: 05:14:00 Civil twilight end: 22:37:00
2019-06-20 21:00:44.604 Nautical twilight start: 04:18:00 Nautical twilight end: 23:33:00
2019-06-20 21:00:44.604 Astronomical twilight start: 02:44:00 Astronomical twilight end: 01:07:00
2019-06-20 21:00:44.985 Active notification Subsystems: (0/13)
2019-06-20 21:00:44.964 Status: PluginSystem: Started, Python version '3.5.3'.
2019-06-20 21:00:44.998 Status: WebServer(HTTP) started on address: :: with port 8080
2019-06-20 21:00:45.027 Starting shared server on: :::6144
2019-06-20 21:00:45.018 Status: WebServer(SSL) started on address: :: with port 443
2019-06-20 21:00:45.027 Status: TCPServer: shared server started...
2019-06-20 21:00:45.098 Status: RxQueue: queue worker started...
2019-06-20 21:00:47.099 Status: (Zigate) Started.
2019-06-20 21:00:47.129 Status: PluginSystem: Entering work loop.
2019-06-20 21:00:49.619 Status: EventSystem: reset all events...
2019-06-20 21:00:54.652 Status: EventSystem: reset all device statuses...
2019-06-20 21:00:55.649 (Zigate) Debug: 0
2019-06-20 21:00:55.836 (Zigate) Configure Permit To Join
2019-06-20 21:00:55.836 (Zigate) Request discovery mode for 254 seconds
2019-06-20 21:00:55.635 Status: (Zigate) Entering work loop.
2019-06-20 21:00:55.636 Status: (Zigate) Initialized version 4.3.1, author 'zaraki673 pipiche38'
2019-06-20 21:00:55.649 Status: (Zigate) Zigate plugin 4.3.1 started
2019-06-20 21:00:55.649 Status: (Zigate) Python Version - 3.5.3 (default, Jan 19 2017, 14:11:04)
2019-06-20 21:00:55.649 [GCC 6.3.0 20170124]
2019-06-20 21:00:55.649 Status: (Zigate) Switching Heartbeat to 5 s interval
2019-06-20 21:00:55.649 Status: (Zigate) DomoticzVersion: 4.10897
2019-06-20 21:00:55.649 Status: (Zigate) DomoticzHash: 9f2dcd957
2019-06-20 21:00:55.649 Status: (Zigate) DomoticzBuildTime: 2019-06-12 09:15:45
2019-06-20 21:00:55.692 Status: (Zigate) Opening DomoticzDB in raw
2019-06-20 21:00:55.711 Status: (Zigate) load PluginConf
2019-06-20 21:00:55.713 Status: (Zigate) PluginConf: /home/pi/domoticz/plugins/Domoticz-Zigate/Conf/PluginConf.txt
2019-06-20 21:00:55.715 Status: (Zigate) -vibrationAqarasensitivity: medium
2019-06-20 21:00:55.715 Status: (Zigate) -enablegroupmanagement: 0
2019-06-20 21:00:55.715 Status: (Zigate) -resetMotiondelay: 30
2019-06-20 21:00:55.715 Status: (Zigate) -allowReBindingClusters: 0
2019-06-20 21:00:55.715 Status: (Zigate) -logFORMAT: 0
2019-06-20 21:00:55.715 Status: (Zigate) -logLQI: 0
2019-06-20 21:00:55.716 Status: (Zigate) -channel: ['11', '15', '19', '20', '25', '26']
2019-06-20 21:00:55.723 Status: (Zigate) Plugin Database: DeviceList-2.txt
2019-06-20 21:00:55.755 Status: (Zigate) DeviceConf loaded
2019-06-20 21:00:55.755 Status: (Zigate) load ListOfDevice
2019-06-20 21:00:55.825 Status: (Zigate) Entries loaded from /home/pi/domoticz/plugins/Domoticz-Zigate/Data/DeviceList-2.txt : 4
2019-06-20 21:00:55.830 Status: (Zigate) Connection Name: Zigate, Transport: Serial, Address: /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
2019-06-20 21:00:55.962 Status: Python EventSystem: Initalizing event module.
2019-06-20 21:00:55.963 Status: EventSystem: Started
2019-06-20 21:00:55.963 Status: EventSystem: Queue thread started...
2019-06-20 21:00:56.065 Status: (Zigate) Major Version Num: 0003
2019-06-20 21:00:56.065 Status: (Zigate) Installer Version Number: 030f
2019-06-20 21:00:56.269 Status: (Zigate) Zigate in Permit Join: On
2019-06-20 21:00:56.379 Status: (Zigate) Zigate addresses ieee: 00158d0001b696bd , short addr: 0000
2019-06-20 21:00:56.379 Status: (Zigate) Network state UP, PANID: 7cbf extPANID: 0xe4a47b9aed96be8a Channel: 11
2019-06-20 21:00:56.488 Status: (Zigate) Number of devices recently active in Zigate = 3
2019-06-20 21:01:05.661 Status: (Zigate) Plugin with Zigate firmware 030f correctly initialized
2019-06-20 21:01:42.706 Status: User: Admin initiated a switch command (6/LED-Osram-Cheminee/Off)
2019-06-20 21:01:47.423 (Zigate) UpdateDevice - (LED-Osram-Cheminee) 1:On
2019-06-20 21:01:47.386 Status: User: Admin initiated a switch command (6/LED-Osram-Cheminee/On)
2019-06-20 21:01:49.348 (Zigate) UpdateDevice - (LED-Osram-Cheminee) 1:24
2019-06-20 21:01:49.308 Status: User: Admin initiated a switch command (6/LED-Osram-Cheminee/Set Level)
2019-06-20 21:01:51.220 (Zigate) UpdateDevice - (LED-Osram-Cheminee) 1:83
2019-06-20 21:01:51.192 Status: User: Admin initiated a switch command (6/LED-Osram-Cheminee/Set Level)
2019-06-20 21:01:53.149 (Zigate) UpdateDevice - ( btn-13) 0:Off
2019-06-20 21:01:53.173 EventSystem: Event triggered: Test-LED_1
2019-06-20 21:01:53.214 (Zigate) UpdateDevice - (LED-Osram-Cheminee) 1:On
2019-06-20 21:01:53.140 Status: User: Admin initiated a switch command (3/btn-13/Off)
[... 61 lines cut ...]
2019-06-24 00:00:01.096 Astronomical twilight start: 02:45:00 Astronomical twilight end: 01:07:00
2019-06-24 00:00:01.096 Status: EventSystem: reset all events...
[... cut ...]
2019-06-24 09:17:28.002 Status: (Zigate) Device Annoucement ShortAddr: d996, IEEE: 7cb03eaa00a99fd7
2019-06-24 09:17:33.172 Status: (Zigate) Device Annoucement ShortAddr: d996, IEEE: 7cb03eaa00a99fd7
2019-06-24 09:17:35.432 Status: (Zigate) Device Annoucement ShortAddr: d996, IEEE: 7cb03eaa00a99fd7
2019-06-24 09:17:38.647 Status: (Zigate) Device Annoucement ShortAddr: d996, IEEE: 7cb03eaa00a99fd7
2019-06-24 17:18:31.564 Status: Incoming connection from: 192.168.1.36
2019-06-24 17:18:39.963 Status: User: Admin initiated a switch command (6/LED-Osram-Cheminee/Off)

>

Cortexlegeni, 2019-06-24:
Here is the log for the reset of the device:

> 2019-06-24 17:23:29.126 (Zigate) UpdateDevice - ( Prise-Osram-01) 0:Off
2019-06-24 17:23:29.105 Status: User: Admin initiated a switch command (5/Prise-Osram-01/Off)
2019-06-24 17:23:30.899 (Zigate) UpdateDevice - ( Prise-Osram-01) 1:On
2019-06-24 17:23:30.862 Status: User: Admin initiated a switch command (5/Prise-Osram-01/On)
2019-06-24 17:23:33.076 (Zigate) UpdateDevice - ( Prise-Osram-01) 0:Off
2019-06-24 17:23:33.041 Status: User: Admin initiated a switch command (5/Prise-Osram-01/Off)
2019-06-24 17:23:34.947 (Zigate) UpdateDevice - ( Prise-Osram-01) 1:On
2019-06-24 17:23:34.925 Status: User: Admin initiated a switch command (5/Prise-Osram-01/On)
2019-06-24 17:23:36.719 (Zigate) UpdateDevice - ( Prise-Osram-01) 0:Off
2019-06-24 17:23:36.695 Status: User: Admin initiated a switch command (5/Prise-Osram-01/Off)
2019-06-24 17:23:38.594 (Zigate) UpdateDevice - ( Prise-Osram-01) 1:On
2019-06-24 17:23:38.550 Status: User: Admin initiated a switch command (5/Prise-Osram-01/On)
2019-06-24 17:23:40.766 (Zigate) UpdateDevice - ( Prise-Osram-01) 0:Off
2019-06-24 17:23:40.718 Status: User: Admin initiated a switch command (5/Prise-Osram-01/Off)
2019-06-24 17:23:42.692 (Zigate) UpdateDevice - ( Prise-Osram-01) 1:On
2019-06-24 17:23:42.653 Status: User: Admin initiated a switch command (5/Prise-Osram-01/On)
2019-06-24 17:23:45.067 (Zigate) UpdateDevice - ( Prise-Osram-01) 0:Off
2019-06-24 17:23:45.039 Status: User: Admin initiated a switch command (5/Prise-Osram-01/Off)
2019-06-24 17:23:46.841 (Zigate) UpdateDevice - ( Prise-Osram-01) 1:On
2019-06-24 17:23:46.831 Status: User: Admin initiated a switch command (5/Prise-Osram-01/On)
2019-06-24 17:23:48.767 (Zigate) UpdateDevice - ( Prise-Osram-01) 0:Off
2019-06-24 17:23:48.757 Status: User: Admin initiated a switch command (5/Prise-Osram-01/Off)
2019-06-24 17:23:50.538 (Zigate) UpdateDevice - ( Prise-Osram-01) 1:On
2019-06-24 17:23:50.523 Status: User: Admin initiated a switch command (5/Prise-Osram-01/On)
2019-06-24 17:23:52.360 (Zigate) UpdateDevice - ( Prise-Osram-01) 0:Off
2019-06-24 17:23:52.318 Status: User: Admin initiated a switch command (5/Prise-Osram-01/Off)
2019-06-24 17:23:53.987 Status: User: Admin initiated a switch command (5/Prise-Osram-01/On)
2019-06-24 17:23:54.034 (Zigate) UpdateDevice - ( Prise-Osram-01) 1:On
2019-06-24 17:24:00.811 Status: (Zigate) Stop directive received.
2019-06-24 17:24:00.812 Status: (Zigate) Exiting work loop.
2019-06-24 17:24:00.926 Status: (Zigate) onStop called
2019-06-24 17:24:01.991 Status: (Zigate) Statistics on message
2019-06-24 17:24:01.991 Status: (Zigate) Sent:
2019-06-24 17:24:01.991 Status: (Zigate) TX commands : 1124
2019-06-24 17:24:01.991 Status: (Zigate) Max Load (Queue) : 4
2019-06-24 17:24:01.991 Status: (Zigate) TX failed : 4 (0.04%)
2019-06-24 17:24:01.991 Status: (Zigate) TX timeout : 0 (0.0%)
2019-06-24 17:24:01.992 Status: (Zigate) TX data timeout : 10 (0.89%)
2019-06-24 17:24:01.992 Status: (Zigate) TX reTransmit : 0 (0.0%)
2019-06-24 17:24:01.992 Status: (Zigate) Received:
2019-06-24 17:24:01.992 Status: (Zigate) RX frame : 3506
2019-06-24 17:24:01.992 Status: (Zigate) RX crc errors : 0 (0.0%)
2019-06-24 17:24:01.992 Status: (Zigate) RX lentgh errors : 0 (0.0%)
2019-06-24 17:24:01.992 Status: (Zigate) RX clusters : 1256
2019-06-24 17:24:01.992 Status: (Zigate) RX clusters KO : 0
2019-06-24 17:24:01.993 Status: (Zigate) Operating time : 20 Hours 23 Mins 6 Secs
2019-06-24 17:24:02.614 Status: (Zigate) Stopping threads.
2019-06-24 17:24:02.614 Status: (Zigate) Stopped.
2019-06-24 17:24:05.732 Status: (Zigate) Started.
2019-06-24 17:24:06.180 (Zigate) Debug: 0
2019-06-24 17:24:06.290 (Zigate) Configure Permit To Join
2019-06-24 17:24:06.290 (Zigate) Request discovery mode for 254 seconds
2019-06-24 17:24:06.172 Status: (Zigate) Entering work loop.
2019-06-24 17:24:06.173 Status: (Zigate) Initialized version 4.3.1, author 'zaraki673 pipiche38'
2019-06-24 17:24:06.180 Status: (Zigate) Zigate plugin 4.3.1 started
2019-06-24 17:24:06.180 Status: (Zigate) Python Version - 3.5.3 (default, Jan 19 2017, 14:11:04)
2019-06-24 17:24:06.180 [GCC 6.3.0 20170124]
2019-06-24 17:24:06.180 Status: (Zigate) Switching Heartbeat to 5 s interval
2019-06-24 17:24:06.181 Status: (Zigate) DomoticzVersion: 4.10897
2019-06-24 17:24:06.181 Status: (Zigate) DomoticzHash: 9f2dcd957
2019-06-24 17:24:06.181 Status: (Zigate) DomoticzBuildTime: 2019-06-12 09:15:45
2019-06-24 17:24:06.194 Status: (Zigate) Opening DomoticzDB in raw
2019-06-24 17:24:06.195 Status: (Zigate) load PluginConf
2019-06-24 17:24:06.196 Status: (Zigate) PluginConf: /home/pi/domoticz/plugins/Domoticz-Zigate/Conf/PluginConf.txt
2019-06-24 17:24:06.197 Status: (Zigate) -vibrationAqarasensitivity: medium
2019-06-24 17:24:06.197 Status: (Zigate) -enablegroupmanagement: 0
2019-06-24 17:24:06.197 Status: (Zigate) -resetMotiondelay: 30
2019-06-24 17:24:06.197 Status: (Zigate) -allowReBindingClusters: 0
[... 10 lines cut ...]
2019-06-24 17:24:06.595 Status: (Zigate) Zigate in Permit Join: On
2019-06-24 17:24:06.702 Status: (Zigate) Zigate addresses ieee: 00158d0001b696bd , short addr: 0000
2019-06-24 17:24:06.702 Status: (Zigate) Network state UP, PANID: 7cbf extPANID: 0xe4a47b9aed96be8a Channel: 11
2019-06-24 17:24:06.808 Status: (Zigate) Number of devices recently active in Zigate = 3
2019-06-24 17:24:12.522 Status: (Zigate) Device Annoucement ShortAddr: d996, IEEE: 7cb03eaa00a99fd7
[... cut ...]
2019-06-24 17:24:49.639 (Zigate) UpdateDevice - (LED-Osram-Cheminee) 1:d1
2019-06-24 17:24:49.624 Status: setcolbrightnessvalue: ID: 6, bri: 82, color: '{m: 3, RGB: ff061f, CWWW: 0000, CT: 0}'
2019-06-24 17:24:54.825 (Zigate) UpdateDevice - (LED-Osram-Cheminee) 1:d1
2019-06-24 17:24:54.787 Status: setcolbrightnessvalue: ID: 6, bri: 82, color: '{m: 3, RGB: 23ff1f, CWWW: 0000, CT: 0}'
2019-06-24 17:24:59.939 (Zigate) UpdateDevice - (LED-Osram-Cheminee) 0:Off
2019-06-24 17:24:59.898 Status: User: Admin initiated a switch command (6/LED-Osram-Cheminee/Off)

pipiche38, 2019-06-24:
> Hello, I tried and it solved the problem for some time, but I still have every 3/4 days a problem, the strim switch on and don't answer after that.

What do you mean the Strip switch ON ? Does it switch on alone, or do you Switch On and then you cannot do anything more ?

If you have the possibility to switch to the beta branch, it would be great if you could experiment this configuration. There is a risk that we fall back to the initial issue.

Edit the file Conf/PluginConf-xx.json
```
"doUnbindBind": 0,
 "allowReBindingClusters": 1,
```

PS/ Est-ce celui-ci ?
https://www.amazon.fr/Osram-Lightify-Connect%C3%A9-R%C3%A9glable-Plastique/dp/B00JDJC1VK/ref=pd_lpo_sbs_201_t_2?_encoding=UTF8&psc=1&refRID=N3BXFXSK44MM75MXH4MD

Cortexlegeni, 2019-06-24:
No, the strip switched on by himself.
I can switch on to the beta branch with no problem, I'm on a test domoticz, so no problem.
C'est cette version : https://www.amazon.fr/Osram-Connecté-Extérieur-Couleurs-Zigbee/dp/B07481Z6RC/ref=sr_1_3?__mk_fr_FR=ÅMÅŽÕÑ&keywords=osram+flex&qid=1561403191&s=lighting&sr=1-3

pipiche38, 2019-06-25:
So please try to pass to the Beta version and update the parameters accordingly.
Even, you can try the User Interface ( https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/WebUserInterface.md )

The fact that the strip switch on on its own looks very weird. Need to understand what is happening and see what is happening around that time

pipiche38, 2019-06-25:
Would be great if you could attached the file Data/DeviceList-xx.txt

Cortexlegeni, 2019-06-25:
I did the update and parameters change yesterday evening and for the moment still no problem. I'm not at home right now so I will send you the file later today.

pipiche38, 2019-06-25:
> 
> 
> I did the update and parameters change yesterday evening and for the moment still no problem. I'm not at home right now so I will send you the file later today.

Excellent, let see how it behave, but I have done additional changes, which might also have a positive impact. So in case it switch on again on its own, please simple do a git pull (still on beta) and lets monitor.
I suspect that the plugin is requesting an information to the strip which is not existing and then the strip simply reset.

Cortexlegeni, 2019-06-28:
Here I did have the same problem this evening, turning on and loosing connection:

> 2019-06-24 21:12:37.855 Status: Domoticz V4.10897 (c)2012-2019 GizMoCuz
2019-06-24 21:12:37.855 Status: Build Hash: 9f2dcd957, Date: 2019-06-12 09:15:45
2019-06-24 21:12:37.856 Status: Startup Path: /home/pi/domoticz/
2019-06-24 21:12:38.394 Status: PluginSystem: Started, Python version '3.5.3'.
2019-06-24 21:12:45.361 Status: WebServer(HTTP) started on address: :: with port 8080
2019-06-24 21:12:46.395 Status: WebServer(SSL) started on address: :: with port 443
2019-06-24 21:12:46.405 Status: TCPServer: shared server started...
2019-06-24 21:12:46.464 Status: RxQueue: queue worker started...
2019-06-24 21:12:48.464 Status: (Zigate) Started.
2019-06-24 21:12:48.465 Status: EventSystem: reset all events...
2019-06-24 21:12:48.472 Status: EventSystem: reset all device statuses...
2019-06-24 21:12:48.536 Status: PluginSystem: Entering work loop.
2019-06-24 21:12:48.683 Status: Python EventSystem: Initalizing event module.
2019-06-24 21:12:48.703 Status: EventSystem: Started
2019-06-24 21:12:48.703 Status: EventSystem: Queue thread started...
2019-06-24 21:12:54.656 Status: (Zigate) Entering work loop.
2019-06-24 21:12:54.658 Status: (Zigate) Initialized version beta-4.4, author 'zaraki673 pipiche38'
2019-06-24 21:12:54.673 Status: (Zigate) Zigate plugin beta-4.4 started
2019-06-24 21:12:54.673 Status: (Zigate) Python Version - 3.5.3 (default, Jan 19 2017, 14:11:04)
2019-06-24 21:12:54.673 [GCC 6.3.0 20170124]
2019-06-24 21:12:54.674 Status: (Zigate) Switching Heartbeat to 5 s interval
2019-06-24 21:12:54.674 Status: (Zigate) DomoticzVersion: 4.10897
2019-06-24 21:12:54.674 Status: (Zigate) DomoticzHash: 9f2dcd957
2019-06-24 21:12:54.674 Status: (Zigate) DomoticzBuildTime: 2019-06-12 09:15:45
2019-06-24 21:12:54.674 Status: (Zigate) Opening DomoticzDB in raw
2019-06-24 21:12:54.695 Status: (Zigate) load PluginConf
2019-06-24 21:12:54.706 Status: (Zigate) pluginOTAFirmware set to /home/pi/domoticz/plugins/Domoticz-Zigate/OTAFirmware/
2019-06-24 21:12:54.706 Status: (Zigate) filename set to /home/pi/domoticz/plugins/Domoticz-Zigate/Conf/PluginConf-02.json
2019-06-24 21:12:54.706 Status: (Zigate) pluginConfig set to /home/pi/domoticz/plugins/Domoticz-Zigate/Conf/
2019-06-24 21:12:54.706 Status: (Zigate) pluginData set to /home/pi/domoticz/plugins/Domoticz-Zigate/Data/
2019-06-24 21:12:54.707 Status: (Zigate) homedirectory set to /home/pi/domoticz/plugins/Domoticz-Zigate/
2019-06-24 21:12:54.707 Status: (Zigate) pluginWWW set to /home/pi/domoticz/plugins/Domoticz-Zigate/www/
2019-06-24 21:12:54.707 Status: (Zigate) pluginZData set to /home/pi/domoticz/plugins/Domoticz-Zigate/Zdatas/
2019-06-24 21:12:54.707 Status: (Zigate) pluginReports set to /home/pi/domoticz/plugins/Domoticz-Zigate/Reports/
2019-06-24 21:12:54.707 Status: (Zigate) pluginHome set to /home/pi/domoticz/plugins/Domoticz-Zigate/
2019-06-24 21:12:54.714 Status: (Zigate) Plugin Database: DeviceList-2.txt
2019-06-24 21:12:54.757 Status: (Zigate) DeviceConf loaded
2019-06-24 21:12:54.757 Status: (Zigate) load ListOfDevice
2019-06-24 21:12:55.323 Status: (Zigate) Entries loaded from /home/pi/domoticz/plugins/Domoticz-Zigate/Data/DeviceList-2.txt : 4
2019-06-24 21:12:55.328 Status: (Zigate) Connection Name: Zigate, Transport: Serial, Address: /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
2019-06-24 21:12:55.335 Status: (Zigate) Request Accepting new Hardware for 254 seconds
2019-06-24 21:12:55.336 Status: (Zigate) Trigger a Topology Scan
2019-06-24 21:12:55.336 Status: (Zigate) Trigger a Energy Level Scan
2019-06-24 21:12:55.439 Status: (Zigate) Major Version Num: 0003
2019-06-24 21:12:55.439 Status: (Zigate) Installer Version Number: 030f
2019-06-24 21:12:55.643 Status: (Zigate) Accepting new Hardware: On
2019-06-24 21:12:55.752 Status: (Zigate) Zigate addresses ieee: 00158d0001b696bd , short addr: 0000
2019-06-24 21:12:55.753 Status: (Zigate) Network state UP, PANID: 7cbf extPANID: 0xe4a47b9aed96be8a Channel: 11
2019-06-24 21:12:55.861 Status: (Zigate) Number of devices recently active in Zigate = 3
2019-06-24 21:13:22.716 Status: (Zigate) Plugin with Zigate firmware 030f correctly initialized
2019-06-24 21:15:41.385 Status: Incoming connection from: 192.168.1.36
2019-06-24 21:15:50.304 Status: setcolbrightnessvalue: ID: 6, bri: 82, color: '{m: 2, RGB: 000000, CWWW: 56a9, CT: 169}'
2019-06-24 21:15:52.578 Status: setcolbrightnessvalue: ID: 6, bri: 82, color: '{m: 2, RGB: 000000, CWWW: 7d82, CT: 130}'
2019-06-24 21:15:53.716 Status: setcolbrightnessvalue: ID: 6, bri: 82, color: '{m: 2, RGB: 000000, CWWW: a45b, CT: 91}'
2019-06-24 21:15:54.884 Status: setcolbrightnessvalue: ID: 6, bri: 82, color: '{m: 2, RGB: 000000, CWWW: c639, CT: 57}'
2019-06-24 21:15:57.283 Status: setcolbrightnessvalue: ID: 6, bri: 82, color: '{m: 2, RGB: 000000, CWWW: 08f7, CT: 247}'
2019-06-24 21:15:57.841 Error: (Zigate) Command: 0x00C0:Move to colour temperature failed on
2019-06-24 21:15:57.841 Error: (Zigate) - Device: LED-Osram-Cheminee NwkID: d996 IEEE: 7cb03eaa00a99fd7
2019-06-24 21:15:57.842 Error: (Zigate) - Code: e9 Status: No acknowledgement received when expected
2019-06-24 21:15:57.842 Error: (Zigate) Command: 0x0081:Move to Level w/o on/off failed on
2019-06-24 21:15:57.842 Error: (Zigate) - Device: LED-Osram-Cheminee NwkID: d996 IEEE: 7cb03eaa00a99fd7
[... cut ...]
2019-06-28 23:17:26.596 (Zigate) UpdateDevice - (LED-Osram-Cheminee) 0:Off
2019-06-28 23:17:26.711 (Zigate) timedOutDevice unit LED-Osram-Cheminee nwkid: d996
2019-06-28 23:17:26.723 Error: (Zigate) Command: 0x0092:On/Off failed on
2019-06-28 23:17:26.723 Error: (Zigate) - Device: LED-Osram-Cheminee NwkID: d996 IEEE: 7cb03eaa00a99fd7
2019-06-28 23:17:26.724 Error: (Zigate) - Code: d4 Status: Unicast frame does not have a route available but it is buffered for automatic resend
2019-06-28 23:18:04.799 Status: Incoming connection from: 192.168.1.36

Cortexlegeni, 2019-06-28:
here is the devicelist:

> 9154 : {'ConfigureReporting': {'TimeStamps': {'02-0000': 1560373485, '02-0001': 1560373485}, 'Ep': {'02': {'0001': {}, '0000': {}}}}, 'IEEE$
0000 : {'Ep': {'01': {'0008': {}, '0004': {}, '0006': {}}}, 'PowerSource': 'Main', 'version': '3', 'IEEE': '00158d0001b696bd'}
efcb : {'ConfigureReporting': {'TimeStamps': {'01-0006': 1560370577, '01-0000': 1560370577}, 'Ep': {'01': {'0000': {}, '0006': {}}}}, 'IEEE$
3a03 : {'ConfigureReporting': {'TimeStamps': {'03-0000': 1560452461, '03-0006': 1561737907}, 'Ep': {'03': {'0000': '8c', '0006': '00'}}}, '$
d996 : {'ConfigureReporting': {'TimeStamps': {'03-0008': 1561738868, '03-0000': 1561738808, '03-0006': 1561738808, '03-0300': 1561738808}, $

pipiche38, 2019-06-28:
From what I see in the log , the device is not on the Network anymore or is not reachable !
So I suggest to redo the pairing.
Instead of copy-paste the logs and other infos, please attached , as then it is much easier to process the file.

Cortexlegeni, 2019-06-29:
[DeviceList-2.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/3341993/DeviceList-2.txt)

Cortexlegeni, 2019-06-29:
New log for reinclusion :
[log.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/3342003/log.txt)

pipiche38, 2019-06-29:
Merci. Here is the Device Configuration: (pretty complet list of Attributes for Cluster 0x0300

```
d996 : {
	'ConfigureReporting': {'TimeStamps': {
					'03-0008': 1561814512, 
					'03-0000': 1561814452, 
					'03-0006': 1561814452, 
					'03-0300': 1561814452}, 
				'Ep': {'03': {'0300': {}, '0008': {}, '0000': {}, '0006': {}}}}, 
	'IEEE': '7cb03eaa00a99fd7', 
	'Manufacturer': 'bbaa', 
	'MacCapa': '8e', 
	'SWBUILD_1': '20140331CNWT****', 
	'Type': 'Switch/LvlControl/ColorControl', 
	'Status': 'inDB', 
	'LogicalType': 'Router', 
	'ProfileID': 'c05e', 
	'Bind': {'0008': {'Stamp': 1561814512, 'Phase': 'requested', 'Status': ''}}, 
	'RIA': '3', 
	'Manufacturer Name': 'OSRAM', 
	'Health': 'Not Reachable', 
	'Battery': {}, 
	'HW Version': '1', 
	'Stack Version': '2', 
	'Stamp': {'MsgType': '8702', 'LastSeen': 1561738788, 'Time': '2019-06-29 15:21:57'}, 
	'DeviceType': 'FFD', 
	'Maufacturer Name': '', 
	'APS Failure': [(1561757045.1695101, 'd4'), (1561814457.6582692, 'd4'), (1561814467.6521318, 'd4'), (1561814477.692547, 'd4'), (1561814517.66846, 'd4')], 
	'ZCL Version': '1', 
	'SQN': {}, 
	'Ep': {'03': {'ClusterType': {'6': 'ColorControlRGBWW'}, 'fc0f': {}, '0000': {}, '0003': {}, '0300': {}, '0019': {}, '0006': '00', '0005': {}, '1000': {}, '0004': {}, '0008': 'd1'}}, 
	'ZDeviceName': '', 
	'PowerSource': 
	'Main', 
	'Attributes List': 
		{'Ep': {'03': {
				'0300': {
						'0002': '21', 
						'0001': '20', 
						'0012': '21', 
						'0021': '21', 
						'001a': '21', 
						'0004': '21', 
						'0008': '30', 
						'0026': '20', 
						'0011': '21', 
						'0019': '21', 
						'0000': '20', 
						'0007': '21', 
						'001b': '20', 
						'0015': '21', 
						'0010': '20', 
						'0017': '20', 
						'0020': '21', 
						'0016': '21', 
						'002a': '20', 
						'0029': '21', 
						'0025': '21', 
						'0003': '21', 
						'0022': '20', 
						'0024': '21', 
						'0028': '21', 
						'0013': '20'}, 
				'0000': {'0002': '20', '0000': '20', '0003': '20', '0012': '10', '0006': '42', '0001': '20', '4000': '42', '0007': '30', '0005': '42', '0004': '42'}, 
				'0003': {'0000': '21'}, 
				'0006': {'4002': '21', '0000': '10', '4000': '10', '4001': '21'}, 
				'0005': {'0001': '20', '0002': '21', '0000': '20', '0003': '10', '0004': '18'}, 
				'0004': {'0000': '18'}, 
				'0008': {'0001': '21', '0012': '21', '0011': '20', '0000': '20', '0013': '21', '0014': '20', '0010': '21'}}}}, 
	'Heartbeat': '12356', 
	'ZDeviceID': '0210', 
	'Model': 'LIGHTIFY Outdoor Flex RGBW', 
	'NbEp': '1', 
	'Last Cmds': [(1561757045.054545, '0092'), (1561814457.5529323, '0120'), (1561814467.5463607, '0120'), (1561814477.5869288, '0120'), (1561814517.5630887, '0120')], 
	'Version': '3', 
	'ColorInfos': {'Saturation': 0, 'Hue': 0, 'Y': '5031', 'X': '24939', 'ColorMode': 2}, 
	'ReceiveOnIdle': 'On', 
	'SWBUILD_3': 'V1.04.93', 
	'RSSI': 87, 
	'ReadAttributes': {'TimeStamps': {'03-0000': 1561738788}, 'Ep': {'03': {'0000': {'0002': '00', '0000': '00', '0003': '00', '0012': '00', '0006': '00', '0001': '00', '4000': '00', '0007': '00', '0005': '00', '0004': '00'}}}}, 
	'App Version': '1'}
```

pipiche38, 2019-06-29:
From the Log, it looks like the device is sending un-expected Attributes for Custer 0x0000

```
2019-06-29 16:57:06.109 (Zigate) ReadCluster 0000 - d996/03 Unknown attribute: 0012 Value: 01
2019-06-29 16:57:06.870 (Zigate) ReadCluster 0000 - d996/03 Unknown attribute: 0012 Value: 01

```

Cortexlegeni, 2019-07-04:
Hello, What should I do with the thing you said ?
Should I change the Device configuration manualy ?
I have other strip of the same model but not in use yet, should I try them too ?

pipiche38, 2019-07-04:
Right now the configuration you have is as follow  (which are the default on beta-4.4)
"doUnbindBind": 0,
"allowReBindingClusters": 1,

(1) Do you still have the issue of spontaneous switch-on ? (which was the initial problem ) 
(2) Is the 2nd problem occurred again ( turning on and loosing connection ) ? Which is a different problem and more related to a lost of communication between Zigate and the strip

Cortexlegeni, 2019-07-04:
The configuration is as in point (1)
and yes, yesterday I had the lost connection again.
[log error 2.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/3359127/log.error.2.txt)

pipiche38, 2019-07-04:
The link between Strip and Zigate seems to have a bad quality estimated LQI is 46 is one way and 54 in the other way. While the LQI may vary up to 255 !

So for me the issue here is much more on the mesh level and the link quality. 
As regards to the question to add other Strip, if they benefit to the mesh , this could be a good think

Last, I would tend to say that the initial problem related to the spontaneous switch seems to be closed.

See the report extracted from the Log. If you use the Web interface you can also display in a graphical way.

```
 Node Node Relation Type Deepth LQI Rx-Idle
 d996 3a03 Sibling Router 1 84 Rx-On
 d996 0000 Parent Coordinator 1 46 Rx-On
 3a03 d996 Sibling Router 1 86 Rx-On
 3a03 0000 Parent Coordinator 1 82 Rx-On
 0000 9154 Child End Device 1 140 Rx-Off
 0000 3a03 Sibling Router 0 93 Rx-On
 efcb Child End Device 1 167 Rx-Off
 0000 d996 Sibling Router 0 54 Rx-On
```

Cortexlegeni, 2019-07-05:
For the link quality, I d'ont understand the problem, the distance between he zigate and the led strip is 3m of air. There is an osram plug wher the strip is plugged. (due to impossibility to plug it or unplugg without using a ladder)

pipiche38, 2019-07-23:
I suggest to close this issue as my understanding is the initial problem related to the reset is now fixed.

As regards to the Link Quality , this is somehow independent of the plugin itself.

There are some discussion on various forums that Osram plug have a buggy firmware which make the routing buggy.

closed 2019-07-23 by pipiche38

Cortexlegeni, 2020-06-03:
Hello, I'm back with a problem of the same type, today when the power went off due to a storm, the led strim turned on and wont respond to Zigate command again.