zigbeefordomoticz/Domoticz-Zigbee #325: Wrong Energy counter on Xiaomi smart sockets in Domoticz Utility

blooesky, 2019-01-16:
I tested today Energy counter from my Xiaomi smart sockets...and i saw the counter is not correct in Utility page...but in Utility device **log** is correct energy counter. Please check!
[image]

**Plugin (please complete the following information):**
 - Raspberry Pi with Raspbian
 - Domoticz-Zigate 4.0.7

pipiche38, 2019-01-16:
As you wrote the data are correct in the DEVICE Log which means that the information send from the plugin to Domoticz are correct.

I'm having similar behaviour with a Salus Plug

Will see how to report that issue to Domoticz

pipiche38, 2019-01-17:
@blooesky , could you kindly do a git pull from Beta branch and check ?

pipiche38, 2019-01-21:
Works for me on beta with Xiaomi and Salus Plug

pipiche38, 2019-01-26:
Here is a plugin to reproduce what is done by the plugin

closed 2019-02-07 by pipiche38