zigbeefordomoticz/Domoticz-Zigbee #326: re-factor the Creation of Light widget

pipiche38, 2019-01-17:
In order to provide a best Light widget on Domoticz, the widget creation must be based on : 

- Today it is only based one cluster 
- ProfileID/ZDeviceID
- Cluster 0300 Attribute

pipiche38, 2019-02-11:
@deennoo 

3 Subtype for ColorControl Widget:

    ColorControlRGB = 0x02 # RGB color palette / Dimable
    ColorControlWC = 0x08  # White color palette / Dimable
    ColorControlFull= 0x07  # 3 Color palettes widget (default)

**Question is which Bulb needs which Widget**
lumi.light.aqcn02 -> ColorControlWC 
TRADFRI bulb E27 CWS opal 600lm -->
'TRADFRI bulb E27 WS opal 950lm  --> 
'TRADFRI bulb E27 WS clear 950lm -->
Philipps Hue 'LCT001                      -->

ben33880, 2019-02-11:
pour les LCT001 et LST002, actuellement elles sont reconnues comme des RGBWW dimmable.
Ce qui correspond il me semble à ce que j'avais avec le pont HUE.

pipiche38, 2019-02-11:
'ColorControlRGB':
                    Subtype_ = 0x02 # RGB color palette / Dimable
'ColorControlRGBWW':
                    Subtype_ = 0x04  # RGB + WW / Dimable
 'ColorControlFull':
                    Subtype_ = 0x07  # 3 Color palettes widget
 'ColorControlWW':
                    Subtype_ = 0x08  # White color palette / Dimable

Can be use directly in DeviceConf.txt

pipiche38, 2019-02-11:
https://github.com/sasu-drooz/Domoticz-Zigate/wiki/DeviceConf

closed 2019-02-11 by pipiche38