zigbeefordomoticz/Domoticz-Zigbee #327: Error: (Zigate) processNotinDBDevices after update to beta (domoticz and plugin)

blooesky, 2019-01-24:
Hi, 
Yesterday i have updated domoticz to beta version and plugin to beta version and now i have a problem with xiaomi wireless switch because is not working...i have this error in log file...and is displayed every second:

2019-01-24 21:28:51.162 Status: (Zigate) [0] NEW OBJECT: 2e64 Model Name: lumi.sensor_switch
2019-01-24 21:28:51.163 Status: (Zigate) [0] NEW OBJECT: 2e64 Model Name: lumi.sensor_switch
2019-01-24 21:28:51.163 Status: (Zigate) [0] NEW OBJECT: 2e64 Model Name: lumi.sensor_switch
2019-01-24 21:28:51.163 Status: (Zigate) [0] NEW OBJECT: 2e64 Trying to create Domoticz device(s)
2019-01-24 21:28:51.165 Error: (Zigate) processNotinDBDevices - Devices already exist. Zigate - SwitchAQ2-00158d0001223dd2-01 with {'Version': '3', 'ProfileID': {}, 'Battery': {}, 'ZDeviceID': {}, 'Stamp': {'MsgType': '8102', 'Time': '2019-01-24 21:21:00'}, 'Ep': {'01': {'0006': '01', '0000': {}}}, 'RIA': '0', 'Model': 'lumi.sensor_switch', 'MacCapa': '80', 'Type': 'SwitchAQ2', 'RSSI': 102, 'App Version': '10', 'IEEE': '00158d0001223dd2', 'Status': '0045', 'Heartbeat': '311', 'SQN': '19'}
2019-01-24 21:28:51.165 Error: (Zigate) processNotinDBDevices - Please cross check the consistency of the Domoticz and Plugin database.
2019-01-24 21:28:56.188 Status: (Zigate) [0] NEW OBJECT: 2e64 Model Name: lumi.sensor_switch
2019-01-24 21:28:56.189 Status: (Zigate) [0] NEW OBJECT: 2e64 Model Name: lumi.sensor_switch
2019-01-24 21:28:56.189 Status: (Zigate) [0] NEW OBJECT: 2e64 Model Name: lumi.sensor_switch
2019-01-24 21:28:56.189 Status: (Zigate) [0] NEW OBJECT: 2e64 Trying to create Domoticz device(s)
2019-01-24 21:28:56.191 Error: (Zigate) processNotinDBDevices - Devices already exist. Zigate - SwitchAQ2-00158d0001223dd2-01 with {'Version': '3', 'ProfileID': {}, 'Battery': {}, 'ZDeviceID': {}, 'Stamp': {'MsgType': '8102', 'Time': '2019-01-24 21:21:00'}, 'Ep': {'01': {'0006': '01', '0000': {}}}, 'RIA': '0', 'Model': 'lumi.sensor_switch', 'MacCapa': '80', 'Type': 'SwitchAQ2', 'RSSI': 102, 'App Version': '10', 'IEEE': '00158d0001223dd2', 'Status': '0045', 'Heartbeat': '312', 'SQN': '19'}
2019-01-24 21:28:56.191 Error: (Zigate) processNotinDBDevices - Please cross check the consistency of the Domoticz and Plugin database.
2019-01-24 21:29:01.189 Status: (Zigate) [0] NEW OBJECT: 2e64 Model Name: lumi.sensor_switch
2019-01-24 21:29:01.190 Status: (Zigate) [0] NEW OBJECT: 2e64 Model Name: lumi.sensor_switch
2019-01-24 21:29:01.190 Status: (Zigate) [0] NEW OBJECT: 2e64 Model Name: lumi.sensor_switch
2019-01-24 21:29:01.190 Status: (Zigate) [0] NEW OBJECT: 2e64 Trying to create Domoticz device(s)
2019-01-24 21:29:01.192 Error: (Zigate) processNotinDBDevices - Devices already exist. Zigate - SwitchAQ2-00158d0001223dd2-01 with {'Version': '3', 'ProfileID': {}, 'Battery': {}, 'ZDeviceID': {}, 'Stamp': {'MsgType': '8102', 'Time': '2019-01-24 21:21:00'}, 'Ep': {'01': {'0006': '01', '0000': {}}}, 'RIA': '0', 'Model': 'lumi.sensor_switch', 'MacCapa': '80', 'Type': 'SwitchAQ2', 'RSSI': 102, 'App Version': '10', 'IEEE': '00158d0001223dd2', 'Status': '0045', 'Heartbeat': '313', 'SQN': '19'}
2019-01-24 21:29:01.192 Error: (Zigate) processNotinDBDevices - Please cross check the consistency of the Domoticz and Plugin database.

i use:
 - Raspberry pi with Raspbian
 - Domoticz 4.1 beta
- Plugin beta

How can i fix this?  Thx

zaraki673, 2019-01-24:
Can you try to remove your switch from Domoticz and add it again?

blooesky, 2019-01-24:
Awesome!!! Now it's working and NOW i haven't events delay so far i had delay 2-3seconds at events But now it is working without delay and it's perfect!

Thank you very much Zaraki!

zaraki673, 2019-01-25:
Thanks pipiche's work 😁😉

closed 2019-02-03 by pipiche38

tof92130, 2020-04-21:
Hi, I face same problem can we remove a Device and keep the Data attached to that device ?

pipiche38, 2020-04-21:
> Hi, I face same problem can we remove a Device and keep the Data attached to that device ?

could you kindly describe what you mean ? 
(1) what is the relation with the Issue in subject ?
(2) If you remove a Device, why keeping the data ?

Please use the forum French or English as described at the end of the Wiki
https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Home.md

tof92130, 2020-04-21:
My log shows an error (jointed). If I suppress the two devices that present the problem, the error should disappear but I fear to loose the history of the Device

`
2020-04-21 10:07:03.387 Error: (Zigate) processNotinDBDevices - Devices already exist. Zigate Ciné with {'ReceiveOnIdle': '', 'Capability': ['Reduced-Function Device', 'Standard security', 'NwkAddr should be allocated'], 'HW Version': '', 'Announced': 1587452821.5619762, 'ConfigSource': 'DeviceConf', 'Heartbeat': '721', 'RIA': '0', 'App Version': '3', 'MacCapa': '80', 'NbEp': '', 'Ep': {'01': {'0405': {'0000': 52.0}, '0000': {'0001': '3', 'ff01': '0121950b0421a80105211200062401000000006429500865218823662bf48401000a210000', '0005': 'lumi.weather', '0004': 'LUMI'}, '0003': {}, '0403': {'0000': 995, '0010': 995.7, '0014': 'ff'}, 'ffff': {}, 'Type': 'Temp/Humi/Baro', '0402': {'0000': 21.1}}}, 'Stamp': {'LastSeen': 1587452821, 'Time': '2020-04-21 09:08:46', 'MsgType': '8102'}, 'SQN': '', 'Model': 'lumi.weather', 'IEEE': '00158d0002c8cca3', 'ZCL Version': '', 'PowerSource': '', 'Version': '3', 'Manufacturer': '', 'RSSI': 60, 'ReadAttributes': {'Ep': {'06': {'0000': {}}, '09': {'0000': {}}, '03': {'0000': {}}, '02': {'0000': {}}, '01': {'0405': {'0000': '00'}, '0000': {'0001': '00', 'ff01': '00', '0005': '00', '0004': '00'}, '0403': {'0000': '00', '0010': '00', '0014': '00'}, '0402': {'0000': '00'}}}, 'TimeStamps': {'01-0000': 1587452821, '06-0000': 1587452821, '03-0000': 1587452821, '09-0000': 1587452821, '02-0000': 1587452821}}, 'Health': 'Live', 'Type': '', 'Manufacturer Name': 'LUMI', 'ZDeviceName': 'Salle Cinéma', 'ZDeviceID': {}, 'Last Cmds': [(1587452821.5628076, '0100', None), (1587452821.6156576, '0100', None), (1587452821.6664658, '0100', None), (1587452821.7171416, '0100', None), (1587452821.767853, '0100', None)], 'Stack Version': '', 'ProfileID': {}, 'Battery': {}, 'Status': '0045', 'DeviceType': 'RFD', 'Attributes List': {}, 'LogicalType': 'End Device'}
2020-04-21 10:07:03.387 Error: (Zigate) processNotinDBDevices - Please cross check the consistency of the Domoticz and Plugin database.
2020-04-21 10:07:03.391 Error: (Zigate) 'onHeartbeat' failed 'UnicodeEncodeError'.
2020-04-21 10:07:03.391 Error: (Zigate) ----> Line 883 in '/usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2020-04-21 10:07:03.391 Error: (Zigate) ----> Line 726 in '/usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
`

pipiche38, 2020-04-21:
@tof92130 , as mentioned, please use one of the forum to get support on that issue. Thanks