zigbeefordomoticz/Domoticz-Zigbee #328: Error when pushing Aqara wireless switch WXKG03LM

davidllerena, 2019-01-26:
Hi,

When I push on Aqara Wireless Switch (WXKG03LM) I receive these messages in log:

 ```
 2019-01-26 16:52:09.911 (ZiGate) onMessage incoming data : '018102120210021fbf5491021d021102101202105502102102100212021002119f03'
2019-01-26 16:52:09.912 (ZiGate) onMessage Frame : Zero1=0 Zero3=34
2019-01-26 16:52:09.913 (ZiGate) onMessage Frame length : 22 22
2019-01-26 16:52:09.915 (ZiGate) onMessage Frame : ComputedChekcum=191 ReceivedChecksum=191
2019-01-26 16:52:09.916 (ZiGate) ZigateRead - decoded data : 018102000fbf54910d01001200550021000200019f03 lenght : 44
2019-01-26 16:52:09.917 (ZiGate) ZigateRead - MsgType 8102 - Report Individual Attribute response : 018102000fbf54910d01001200550021000200019f03
2019-01-26 16:52:09.919 (ZiGate) Decode8102 - reception data : 0001 ClusterID : 0012 Attribut ID : 0055 Src Addr : 910d Scr Ep: 01 RSSI = 9f
2019-01-26 16:52:09.920 (ZiGate) Decode8012 : RSSI set to 159/9f for 910d
2019-01-26 16:52:09.921 (ZiGate) Decode8102 : Attribute Report from 910d SQN = 54 ClusterID = 0012 AttrID = 0055 Attribute Data = 0001
2019-01-26 16:52:09.923 (ZiGate) ReadCluster - MsgData lenght is : 28 out of 24+
2019-01-26 16:52:09.924 (ZiGate) cube action : Not expected value 1
2019-01-26 16:52:09.926 (ZiGate) MajDomoDevice - Device ID : 910d - Device EP : 01 - Type : 0012 - Value : 1 - Hue :
2019-01-26 16:52:09.927 (ZiGate) MajDomoDevice - Type = XCube
2019-01-26 16:52:09.929 Error: (ZiGate) 'onMessage' failed 'KeyError'.
2019-01-26 16:52:09.930 Error: (ZiGate) ----> Line 268 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onMessage
2019-01-26 16:52:09.931 Error: (ZiGate) ----> Line 213 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onMessage
2019-01-26 16:52:09.933 Error: (ZiGate) ----> Line 243 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_input.py, function ZigateRead
2019-01-26 16:52:09.934 Error: (ZiGate) ----> Line 563 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_input.py, function Decode8102
2019-01-26 16:52:09.936 Error: (ZiGate) ----> Line 927 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_input.py, function ReadCluster
2019-01-26 16:52:09.937 Error: (ZiGate) ----> Line 175 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py, function MajDomoDevice  
```

On every heartbeat these messages appear in log:
```
2019-01-26 16:49:44.354 Error: (ZiGate) 'onHeartbeat' failed 'KeyError'.
2019-01-26 16:49:44.354 Error: (ZiGate) ----> Line 280 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2019-01-26 16:49:44.354 Error: (ZiGate) ----> Line 242 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2019-01-26 16:49:44.354 Error: (ZiGate) ----> Line 427 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py, function ResetDevice
```
The switch has been recognised as 
ZiGate - XCube-910d-01 | Light/Switch | Selector Switch

I'm connecting these switches directly to ZiGate, not using Aqara Hub.

I also have WXKG02LM, but not tested yet in depth. Some errors found also, but only when pushing the second button.

 ```
2019-01-26 16:53:36.902 (ZiGate) onMessage incoming data : '018102120210021fb23d23e502120210120210550210210210021202100211a203'
2019-01-26 16:53:36.904 (ZiGate) onMessage Frame : Zero1=0 Zero3=33
2019-01-26 16:53:36.906 (ZiGate) onMessage Frame length : 22 22
2019-01-26 16:53:36.907 (ZiGate) onMessage Frame : ComputedChekcum=178 ReceivedChecksum=178
2019-01-26 16:53:36.909 (ZiGate) ZigateRead - decoded data : 018102000fb23d23e50200120055002100020001a203 lenght : 44
2019-01-26 16:53:36.911 (ZiGate) ZigateRead - MsgType 8102 - Report Individual Attribute response : 018102000fb23d23e50200120055002100020001a203
2019-01-26 16:53:36.912 (ZiGate) Decode8102 - reception data : 0001 ClusterID : 0012 Attribut ID : 0055 Src Addr : 23e5 Scr Ep: 02 RSSI = a2
2019-01-26 16:53:36.914 (ZiGate) Decode8012 : RSSI set to 162/a2 for 23e5
2019-01-26 16:53:36.916 (ZiGate) Decode8102 : Attribute Report from 23e5 SQN = 3d ClusterID = 0012 AttrID = 0055 Attribute Data = 0001
2019-01-26 16:53:36.917 (ZiGate) ReadCluster - MsgData lenght is : 28 out of 24+
2019-01-26 16:53:36.919 (ZiGate) cube action : Not expected value 1
2019-01-26 16:53:36.921 (ZiGate) MajDomoDevice - Device ID : 23e5 - Device EP : 02 - Type : 0012 - Value : 1 - Hue :
2019-01-26 16:53:36.923 (ZiGate) MajDomoDevice - Type = XCube
2019-01-26 16:53:36.924 (ZiGate) MajDomoDevice - Dtypename = XCube
2019-01-26 16:53:36.926 Error: (ZiGate) 'onMessage' failed 'TypeError':'unorderable types: int() >= str()'.
2019-01-26 16:53:36.928 Error: (ZiGate) ----> Line 268 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onMessage
2019-01-26 16:53:36.929 Error: (ZiGate) ----> Line 213 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onMessage
2019-01-26 16:53:36.931 Error: (ZiGate) ----> Line 243 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_input.py, function ZigateRead
2019-01-26 16:53:36.933 Error: (ZiGate) ----> Line 563 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_input.py, function Decode8102
2019-01-26 16:53:36.934 Error: (ZiGate) ----> Line 927 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_input.py, function ReadCluster
2019-01-26 16:53:36.936 Error: (ZiGate) ----> Line 363 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py, function MajDomoDevice
``` 

Domoticz Version 4.9700
SO: Raspbian Jessie
Zigate Plugin: updated today (26/01/2019)

zaraki673, 2019-01-26:
Are you on master or beta of plug-in version?

pipiche38, 2019-01-26:
> Effectively I was in main branch.
> After moving to beta it works!!
>
> Any idea when these changes would release on main branch?

Main under ‘master’ is discontinued, we will maintain stable (as a replacement of master) beta and time to time we will have devel for ongoing development.

closed 2019-01-29 by pipiche38