zigbeefordomoticz/Domoticz-Zigbee #329: Recover Domoticz angle of rotation of the Cube from Xiaomi.

nicog2, 2019-01-27:
Hello,

Would it be possible to remake under domoticz the value of the angle made. With for example a positive angle if the rotation is clockwise, and negative if anti-clockwise. This value can go up in a second device (ID). In order to control, for example, the volume, a variable lamp ...

thanks in advance

pipiche38, 2019-01-27:
Today's widget provides Clock wise and Anti-Clock wise information as well as Flip 90 or 180 and some other possible actions from the Cube.

Creating a Second widget when pairing a Cube is of course possible. 
This second widget would be a Text widget ( not a switch  ) and will be available on the Utility section (not in the switch section ).

In order to make it more useful, we could eventually push all Cube events to this Text widget and of course when receiving angle pushing the clock+anti-clock angle ( with + or - )

Now, the main question is can an update to a Text Widget trigger LUA/dzEvent ?

nicog2, 2019-01-27:
when i connect the cube to the xiaomi gateway, i have to widget on domoticz, one in switch with the action (rotation, shake,...) and another Text widdget in Utility section with the angle of rotation. 
With this case, when an event from cube in switch, you can read the angle if necessary.
[image]
[image]

pipiche38, 2019-01-29:
If you switch to beta ( 959df90 ), you should be able to see in the Logs the XCube rotation angle. Could you kindly check ?

ben33880, 2019-01-29:
yes i see rotation : 

2019-01-29 19:30:58.171 (Zigate) ReadCluster - ClusterId=000c - Magic Cube angle: -51.40000534057617
2019-01-29 19:30:58.171 (Zigate) Update v2 Values 90:'90:' (Xiaomi Cube Lumiere)
2019-01-29 19:30:58.975 (Zigate) ReadCluster - ClusterId=000c - Magic Cube angle: -9.149998664855957
2019-01-29 19:31:02.633 (Zigate) Update v2 Values 40:'40:' (Xiaomi Cube Lumiere)

ben33880, 2019-01-31:
I do not know if it's related to the changes made for the rotation but the flip_90 only works once. you have to make another move for the flip_90 to work again

pipiche38, 2019-02-01:
That is correct. The Cube is not sending a new message after a flip, and I beleive this is true for all. If you do clock rotation, and then you do it once more ... the 2nd one is not seen.

From the LOg I do not see any message from the device itself.

ben33880, 2019-02-02:
with xiaomi gateway, you can flip many times.

pipiche38, 2019-02-02:
could you reach me on  keybase.io ?

pipiche38, 2019-02-02:
> Nope it's not true for all, on my installation it's same.
> For exemple, if I use the cube to set a speaker volume, I make a clockwise rotation, if the sound is not hight enought, I make a second clockwise rotation more faster. No need to make a "blank" move to enable the rotation again.

@smanar long time ;-)

Indeed you are right. But then the 2nd , 3rd time you just get the rotation level

If I do 1 flip 90 i see it. If I continue and do one more I don’t see it !
However if after a flip 90 , I do a flip 180 I see it.

pipiche38, 2019-02-03:
@ben33880 please use latest beta version from that morning .

ben33880, 2019-02-03:
we can now flip many times !!!!

pipiche38, 2019-02-03:
Will work on the text widget to get the rotation angle .

Smanar, 2019-02-03:
Lol, Hello.
I m still here, I m still following this project.
I think there is a mistake on my side with rotation and flip. But on all case flip or rotation, you need to trigger notification, and perhaps use the same ResetDevice() fonction than the motion sensor.

pipiche38, 2019-02-03:
The issue was on the fact that we try to avoid Updating domoticz when the value doesn’t change …. ;-)

Smanar, 2019-02-03:
Yep, I have just see the corrective > ForceUpdate_ = True

pipiche38, 2019-02-03:
@ben33880 If you can do an update of beta branch. Delete the existing Domoticz widget related to Xcube and repair it you should then see 2 widgets created for the device. The classic one and a text widget (can be found in Utility/Mesures)

For now, Rotation angle are reported to the Text widget .

ben33880, 2019-02-03:
i update, i saw new device text and selector.
but if i use cube, I can not use other zigbee devices anymore (hue, tradfri). the cube no longer communicates either

pipiche38, 2019-02-03:
What do you have in the log

pipiche38, 2019-02-03:
@nicog2 could you confirm that it works as expected ? Please use latest beta branch

pipiche38, 2019-02-04:
Is this still happening?

ben33880, 2019-02-04:
2019-02-04 19:44:02.510 (zig) Update v2 Values 20:'20:' (Cube Lumières)
2019-02-04 19:45:24.319 (zig) Update v2 Values 0:'-65:' (Cube Lumières Angle)
2019-02-04 19:45:24.321 (zig) Update v2 Values 90:'90:' (Cube Lumières)

Cube Lumières Angle is the text widget

pipiche38, 2019-02-04:
Can we close the issue ?

ben33880, 2019-02-05:
it works for me !!!

nicog2, 2019-02-05:
I have not the time for testing. But if it works for ben33880, i think it work for me.

closed 2019-02-05 by nicog2