zigbeefordomoticz/Domoticz-Zigbee #330: Domoticz crashes after Zigate plugin updated to beta

davidllerena, 2019-01-29:
Hi all, 

After updating Zigate plugin to beta, Domoticz has been come very inestable. Every some hours Domoticz crashes and I have to restart the service manually.

```
ene 29 04:14:19 raspi domoticz[19965]: (ZiGate) Update v2 Values 21.9:'21.9:' (Temperatura - Salon)
ene 29 04:14:19 raspi domoticz[19965]: (ZiGate) Update v2 Values 0:'21.9;31.4;2;943.9;4:' (Salon - Temp+Hum+Baro-00158d0001fa75d6-01)
ene 29 04:14:19 raspi domoticz[19965]: (ZiGate) Update v2 Values 0:'21.9;31.4;2:' (Salon - Temp+Hum-00158d0001fa75d6-01)
ene 29 04:14:19 raspi domoticz[19965]: (ZiGate) Update v2 Values 0:'21.9;31.4;2;943;4:' (Salon - Temp+Hum+Baro-00158d0001fa75d6-01)
ene 29 04:14:19 raspi domoticz[19965]: (ZiGate) Update v2 Values 0:'943;4:' (Barómetro - Salon)
ene 29 04:14:19 raspi domoticz[19965]: (ZiGate) Update v2 Values 0:'21.9;31.4;2;943.6;4:' (Salon - Temp+Hum+Baro-00158d0001fa75d6-01)
ene 29 04:14:19 raspi domoticz[19965]: (ZiGate) Update v2 Values 0:'943.6;4:' (Barómetro - Salon)
ene 29 04:14:30 raspi domoticz[19965]: (ZiGate) ReadCluster - 0000/ff01 Saddr: 74a3 ClusterData : 64100003281c9839000000009539c91d2a3e05210e009a201008...0009210102
ene 29 04:14:30 raspi domoticz[19965]: (ZiGate) ReadCluster - 0000/ff01 Saddr: 74a3 sOnOff: 00
ene 29 04:14:32 raspi domoticz[19965]: Domoticz received fatal signal 11 !...
```

Domoticz Version 4.9700
SO: Raspbian Jessie
Zigate plugin: ## 07 January 2019 - Version Beta 4.1.x - Updated from stable at 26/01/2019

davidllerena, 2019-01-29:
more logs:

```
/var/log $ sudo tail -n 20 domoticz.log
2019-01-29 16:19:07.121  (ZiGate) Ping - Send a Ping
2019-01-29 16:19:07.122  Error: (ZiGate) Transport is not connected, write directive to 'ZiGate' ignored.
2019-01-29 16:20:07.165  (ZiGate) Ping - We didn't receive any messages since 60s
2019-01-29 16:20:07.166  (ZiGate) Ping - no Heartbeat with Zigate
2019-01-29 16:20:07.169  (ZiGate) Transport.reConn: Name: 'ZiGate', Transport: 'Serial', Protocol: 'None', Address: '/dev/ttyUSB0', Port: '', Baud: 115200, Bytes: 9499, Connected: False, Last Seen: 2019-01-29 16:17:34
2019-01-29 16:20:07.170  (ZiGate) Lost connection, reConn Transport.reConn: Name: 'ZiGate', Transport: 'Serial', Protocol: 'None', Address: '/dev/ttyUSB0', Port: '', Baud: 115200, Bytes: 9499, Connected: False, Last Seen: 2019-01-29 16:17:34
2019-01-29 16:20:07.170  Status: (ZiGate) Connection Name: Zigate, Transport: Serial, Address: /dev/ttyUSB0
2019-01-29 16:20:07.173  Error: (ZiGate) Transport is not connected, write directive to 'ZiGate' ignored.
2019-01-29 16:20:07.176  (ZiGate) Connected successfully
2019-01-29 16:20:07.177  Status: (ZiGate) Reconnected after failure
2019-01-29 16:20:07.177  Status: (ZiGate) ZigateConf - Setting UTC Time to : 602094007
2019-01-29 16:20:07.178  Status: (ZiGate) Start network
2019-01-29 16:20:07.178  Status: (ZiGate) Zigate enter in discover mode for 120 Secs
2019-01-29 16:20:07.605  Error: Domoticz received fatal signal 11 !...
2019-01-29 16:20:07.613  Error:   /home/xx/domoticz/domoticz() [0x262154]
2019-01-29 16:20:07.614  Error:   /home/xx/domoticz/domoticz(_Z14signal_handleri+0x4c) [0x26220c]
2019-01-29 16:20:07.614  Error:   /lib/arm-linux-gnueabihf/libc.so.6(__default_sa_restorer_v2+0) [0x76bf4180]
2019-01-29 16:20:07.614  Error:   /home/xx/domoticz/domoticz(_ZN7Plugins16CPluginTransport16VerifyConnectionEv+0x28) [0x5d513c]
2019-01-29 16:20:07.614  Error:   /home/xx/domoticz/domoticz(_ZN7Plugins7CPlugin7Do_WorkEv+0x19c) [0x5c4184]
2019-01-29 16:20:07.614  Error:   /home/xx/domoticz/domoticz() [0x847f38]

```

pipiche38, 2019-01-29:
please use the forum to report such issues.

In here Log, your issue seems to be related to a lost of connectivity with the Zigate.

pipiche38, 2019-02-06:
Don't know if you still have the issue ?

If that is the case, could you update the beta version ( git pull ), and I suggest that you disable the ping functionality, by editing the file PluginConf.txt where there is a parameter Ping set to 1. 

Set ping to 0

'Ping':'0',

davidllerena, 2019-02-07:
Yes, I still have the issue.
Will try and update

davidllerena, 2019-02-07:
Hi,

no crashes for today at this moment, but since I have updated and modified PluginConf.txt and restarted domoticz, exactly 58 minutes from restart, I receive this messages on log every 5 seconds:
`Error: (ZiGate) Transport is not connected, write directive to 'ZiGate' ignored.
`

Will maintain in observation some days.

pipiche38, 2019-02-07:
What do you mean, you have such message now, or it was before ?

> I receive this messages on log every 5 seconds:
> `Error: (ZiGate) Transport is not connected, write directive to 'ZiGate' ignored. `

pipiche38, 2019-02-07:
1d2be21

davidllerena, 2019-02-07:
> What do you mean, you have such message now, or it was before ?
> 
> > I receive this messages on log every 5 seconds:
> > `Error: (ZiGate) Transport is not connected, write directive to 'ZiGate' ignored.`

I have checked the log and I can see this message days before, but not as continuous as today, after update at 11:47

```
xx@xx:/var/log $ sudo cat /var/log/domoticz.log | grep "Error: (ZiGate) Transport"| grep "2019-02-07" | wc -l
3652
xx@xx:/var/log $ sudo cat /var/log/domoticz.log | grep "Error: (ZiGate) Transport"| grep "2019-02-06" | wc -l
515
xx@xx:/var/log $ sudo cat /var/log/domoticz.log | grep "Error: (ZiGate) Transport"| grep "2019-02-05" | wc -l
480
xx@xx:/var/log $ sudo cat /var/log/domoticz.log | grep "Error: (ZiGate) Transport"| grep "2019-02-04" | wc -l
266
xx@xx:/var/log $ sudo cat /var/log/domoticz.log | grep "Error: (ZiGate) Transport"| grep "2019-02-03" | wc -l
235
```

At 19:00 I have enabled ping in PluginConf.txt and restarted Domoticz and the number of error messages have decreased a lot.

```
xx@xx:/var/log $ sudo cat /var/log/domoticz.log | grep "Error: (ZiGate) Transport"| grep "2019-02-07 1[2-9]" | wc -l
3368
xx@xx:/var/log $ sudo cat /var/log/domoticz.log | grep "Error: (ZiGate) Transport"| grep "2019-02-07 2[0-2]" | wc -l
53
```

pipiche38, 2019-02-08:
Those error message are still weird and I wonder if you don’t have an issue somewhere else ….

I have not a single one !

davidllerena, 2019-02-08:
Thanks for your help and your patientty.

When messages appear continuously I can't use any Aqara selector switch and control hue bulbs. I can't say that about temp sensors, I can't test if they are communicating with Zigate/Domoticz in that moment.

There is a relation between messages and Zigate/Domoticz communication with devices. 

So:
- If ping enabled, domoticz crash (last night Domoticz has crashed again, I think I have to wait some days to confirm if disabling ping resolve this)
- if ping disabled, no communication between devices and Zigate when messages appear.

davidllerena, 2019-02-08:
if you need log, any file, etc....

pipiche38, 2019-02-08:
@davidllerena I'm afraid that I cannot help much on the communication issue. I would suggest you to troubleshoot around USB port . The communication lost is definitively an issue to be resolved

davidllerena, 2019-02-13:
Don’t know why, but if I activate debug framework in Zigate hardware plugin in Domoticz all work fine. No crashes and no connecting fails in log or devices for days.

pipiche38, 2019-02-13:
Interesting: 
- Could you remind me what version of Python are you using ? (that is printed at the start of the plugin)

davidllerena, 2019-02-13:
Python Version - 3.4.2 (default, Sep 26 2018, 07:50:08)

pipiche38, 2019-02-17:
@davidllerena , I don't think we can do much more on the plugin side, as this is more related to the Framework.

I believe you have already check that one : 
https://www.domoticz.com/wiki/Using_Python_plugins

As for the plugin is python intensive we are highly recommending python3.5 and above

I'm closing this issue

closed 2019-02-17 by pipiche38