zigbeefordomoticz/Domoticz-Zigbee #332: Innr BY 265-2

petitPoulpe, 2019-01-30:
Hello,
I have this product and it is compatible with Zigate plugin :
https://www.amazon.fr/gp/product/B07H5S7JV5/ref=ppx_yo_dt_b_asin_title_o09__o00_s00?ie=UTF8&psc=1

In Domoticz, it creates two devices but just one seems to respond.

pipiche38, 2019-01-30:
Could you kindly let me know what are the 2 widgets created ?
Would appreciate if you could attached the DeviceList-xx.txt file so I can get more information on the device, and maybe avoid the 2nd widget

Patrick

petitPoulpe, 2019-01-30:
6ce4 : {'ProfileID': '0104', 'App Version': '3', 'Type': '', 'Status': 'inDB', 'Version': '3', 'ZDeviceID': '5f01', 'IEEE': '00158d0001b97dee', 'RIA': '1', 'RSSI': 105, 'ReadAttributes': {'TimeStamps': {'01-0000': 1548871429, '01-0006': 1548871293}, 'Ep': {'01': {'0000': {'0005': '00', '0000': '00', '0001': '00', 'ff01': '86'}, '0001': {}, '0006': {'0000': '00'}}}}, 'Stamp': {'Time': '2019-01-30 19:03:49', 'MsgType': '8102'}, 'NbEp': '1', 'Model': 'lumi.sensor_magnet.aq2', 'Ep': {'01': {'0006': '00', 'Type': 'Door', 'ClusterType': {'99': 'Switch'}, '0000': {}, '0003': {}, 'ffff': {}}}, 'MacCapa': '80', 'Battery': 91, 'Heartbeat': '445277', 'SQN': '7c'}
71c0 : {'App Version': '3', 'Type': '', 'Status': 'inDB', 'Version': '3', 'IEEE': '00158d0002772a67', 'RIA': '0', 'Stamp': {'Time': '2019-01-26 16:01:41', 'MsgType': '8102'}, 'LogicalType': 'End Device', 'Ep': {'01': {'0402': 17.8, 'Type': 'Temp/Humi/Baro', 'ClusterType': {'90': 'Temp', '89': 'Temp+Hum', '92': 'Baro', '88': 'Temp+Hum+Baro', '91': 'Humi'}, '0000': {}, '0003': {}, '0405': 55.6, '0403': 995.9, 'ffff': {}}}, 'MacCapa': '80', 'Battery': 91, 'ReceiveOnIdle': 'Off', 'DeviceType': 'RFD', 'SQN': '5b', 'ProfileID': '0104', 'Manufacturer': '1037', 'PowerSource': 'Battery', 'ZDeviceID': '5f01', 'RSSI': 54, 'ReadAttributes': {'TimeStamps': {'01-0000': 1547489049}, 'Ep': {'01': {'0000': {'0000': '00', 'ff01': '86'}, '0001': {}}}}, 'NbEp': '1', 'Model': 'lumi.weather', 'Heartbeat': '445271'}
8e8b : {'ProfileID': '0104', 'App Version': '2', 'Type': 'LvlControl/Switch/', 'Status': 'inDB', 'Version': '3', 'ZDeviceID': '000c', 'IEEE': '0017880104ef3a9e', 'RIA': '0', 'RSSI': 69, 'ReadAttributes': {'TimeStamps': {'02-0001': 1548833900, '02-0000': 1547391196, '01-0006': 1548149034, '01-0000': 1548497359, '01-0008': 1548149034}, 'Ep': {'02': {'0000': {}, '0001': {'0020': '00', '0021': '00', '0010': '86', '0000': '86'}}, '01': {'0000': {'0005': '00', '0001': '00', 'ff01': '86', '0002': '00', '0010': '86', '0000': '00', '0007': '00', '0004': '00', '0003': '00'}, '0001': {}, '0006': {'0000': '86'}, '0008': {'0000': '86'}}}}, 'Stamp': {'Time': '2019-01-27 08:37:33', 'MsgType': '8102'}, 'NbEp': '2', 'Model': 'RWL021', 'Ep': {'02': {'0001': '0;0;28;200', '0019': {}, 'ClusterType': {'104': 'LvlControl'}, '0000': {}, '0003': {}, 'fc00': {}, '000f': {}}, '01': {'0005': {}, '0006': {}, '0004': {}, 'ClusterType': {'103': 'LvlControl'}, '0000': {}, '0003': {}, '0008': {}}}, 'MacCapa': '80', 'GroupMgt': {'01': {'0014': {'Phase': 'TimeOut', 'Phase-Stamp': 1548869703}}}, 'Battery': 200, 'Heartbeat': '63483', 'HW Version': '1', 'Stack Version': '1', 'SQN': '00'}
cc19 : {'App Version': '12', 'Type': 'Switch/', 'Status': 'inDB', 'Version': '3', 'Stack Version': '2', 'IEEE': '7cb03eaa00b21e26', 'RIA': '0', 'Stamp': {'Time': '2019-01-30 19:03:03', 'MsgType': '8102'}, 'LogicalType': 'Router', 'Ep': {'03': {'0005': {}, '0006': '01', '0004': {}, 'ClusterType': {'87': 'Switch'}, '0000': {}, '0003': {}, 'fc0f': {}, '1000': {}, '0b04': {}}}, 'MacCapa': '8e', 'Battery': {}, 'HW Version': '1', 'ReceiveOnIdle': 'On', 'ConfigureReporting': {'TimeStamps': {'03-0000': 0, '03-0006': 1548791001, '03-0005': 0, '03-1000': 0, '03-0b04': 0, '03-fc0f': 0, '03-0003': 0}, 'Ep': {'03': {'0005': {}, '0006': '00', '0004': '', '0000': {}, '0003': {}, 'fc0f': {}, '1000': {}, '0b04': {}}}}, 'DeviceType': 'FFD', 'SQN': {}, 'ProfileID': 'c05e', 'Manufacturer': 'bbaa', 'PowerSource': 'Main', 'ZDeviceID': '0010', 'GroupMgt': {'03': {'0014': {'Phase': 'OK-Membership', 'Phase-Stamp': 0}}}, 'RSSI': 150, 'ReadAttributes': {'TimeStamps': {'03-0000': 1548832510, '03-0006': 1548871298}, 'Ep': {'01': {'0000': {}, '0001': {}}, '03': {'0000': {'0005': '00', '0001': '00', 'ff01': '86', '0002': '00', '0010': '86', '0000': '00', '0007': '00', '0004': '00', '0003': '00'}, '0006': {'0000': '00'}}}}, 'NbEp': '1', 'Model': 'Plug 01', 'Heartbeat': '3016'}
eb85 : {'App Version': '12', 'Type': 'Switch/', 'Status': 'inDB', 'Version': '3', 'Stack Version': '2', 'IEEE': '84182600001009c2', 'RIA': '0', 'Stamp': {'Time': '2019-01-30 19:02:29', 'MsgType': '8102'}, 'LogicalType': 'Router', 'Ep': {'03': {'0005': {}, '0006': '01', '0004': {}, 'ClusterType': {'85': 'Switch'}, '0000': {}, '0003': {}, 'fc0f': {}, '1000': {}, '0b04': {}}}, 'MacCapa': '8e', 'Battery': {}, 'HW Version': '1', 'ReceiveOnIdle': 'On', 'ConfigureReporting': {'TimeStamps': {'03-0000': 0, '03-0006': 1548790881, '03-0005': 0, '03-1000': 0, '03-0b04': 0, '03-fc0f': 0, '03-0003': 0}, 'Ep': {'03': {'0005': {}, '0006': '00', '0004': '', '0000': {}, '0003': {}, 'fc0f': {}, '1000': {}, '0b04': {}}}}, 'DeviceType': 'FFD', 'SQN': {}, 'ProfileID': 'c05e', 'Manufacturer': 'bbaa', 'PowerSource': 'Main', 'ZDeviceID': '0010', 'GroupMgt': {'03': {}}, 'RSSI': 150, 'ReadAttributes': {'TimeStamps': {'03-0000': 1548833165, '03-0006': 1548871348}, 'Ep': {'01': {'0000': {}, '0001': {}}, '03': {'0000': {'0005': '00', '0001': '00', 'ff01': '86', '0002': '00', '0010': '86', '0000': '00', '0007': '00', '0004': '00', '0003': '00'}, '0006': {'0000': '00'}}}}, 'NbEp': '1', 'Model': 'Plug 01', 'Heartbeat': '3018'}
768a : {'ProfileID': '0104', 'App Version': '5', 'Type': 'Aqara', 'Status': 'inDB', 'Version': '3', 'ZDeviceID': '5f03', 'IEEE': '00158d0002564bdc', 'RIA': '0', 'RSSI': 111, 'ReadAttributes': {'TimeStamps': {'01-0000': 1548842507}, 'Ep': {'01': {'0000': {'0005': '00', '0000': '00', '0001': '00', 'ff01': '86'}, '0001': {}}}}, 'Stamp': {'Time': '2019-01-30 18:58:22', 'MsgType': '8102 [... cut ...] 871388}, 'Ep': {'01': {'0000': {'0000': '00', 'ff01': '86'}, '0001': {}, '0006': {'0000': '00'}}}}, 'Stamp': {'Time': '2019-01-30 19:03:10', 'MsgType': '8102'}, 'NbEp': '1', 'Model': 'lumi.sensor_magnet.aq2', 'Ep': {'01': {'0006': '00', 'Type': 'Door', 'ClusterType': {'98': 'Door'}, '0000': {}, '0003': {}, 'ffff': {}}}, 'MacCapa': '80', 'Battery': 91, 'Heartbeat': '445270', 'SQN': '79'}
0000 : {'IEEE': '00158d0001edb760', 'Ep': {'01': {'0006': {}, '0004': {}, '0008': {}}}, 'PowerSource': 'Main'}

petitPoulpe, 2019-01-30:
The two widgets are Dimmer

petitPoulpe, 2019-01-30:
device 8e8b is Philips Hue remote control, this devices not seem to work.
yes  device b008 and ad2c are 'Model': 'BY 265'
All this devices have been paired with beta plugin.

pipiche38, 2019-01-31:
Are you using the standard DeviceConf.txt , or did you create/update it ?

petitPoulpe, 2019-01-31:
i  use the Standard DeviceConf.txt

pipiche38, 2019-01-31:
Bug found. Fix in progress

petitPoulpe, 2019-01-31:
Great !! Thank you

pipiche38, 2019-01-31:
Fix in beta branch f9864c1

petitPoulpe, 2019-01-31:
How can i test it ?

pipiche38, 2019-01-31:
You should update to the latest version of beta, remove the 2 devices and then re-do the pairing

pipiche38, 2019-02-03:
Could you let me know if you have been able to validate the fact that only one device is created now ?

closed 2019-02-05 by pipiche38

petitPoulpe, 2019-02-05:
It works ! Thank You

commit f9864c16 (2019-01-31): Fix #332 - Creating unexpected widget