zigbeefordomoticz/Domoticz-Zigbee #333: hue bulb

ben33880, 2019-01-30:
hi,
hue bulb and lightstrip are recognized as WW Color switch, not as RGBWWZ.
can u do something ??

thanks. 
[DeviceConf.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2814496/DeviceConf.txt)

ben33880, 2019-01-30:
[DeviceList-35.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2814507/DeviceList-35.txt)

pipiche38, 2019-01-30:
Before opening an issue, it would be great to qualify if that is an issue via the forum.

You are droping a bunch of data in the DeviceList-35.txt. Thanks to provide some insights

Logs when creating the device 
Version of the plugin

ben33880, 2019-01-30:
ok sorry, what forum ?
last beta version.

ben33880, 2019-01-31:
ok, by editing the following line in the file DeviceConf.txt, the LCT007 hue works in RGBWW.

on the other hand, how should I do for the other models in the Hue range?
must we create a new line for each generation of the Hue range?

'Ampoule.phillips.hue':{'Ep':{'0b':{'0000':'','0003':'','0004':'','0005':'','0006':'','0008':'','0300':'','fc01':'','1000':'','Type':'ColorControl','ColorMode':'2'}},'Type':'','ProfileID':'c05e','ZDeviceID':'0210'},

pipiche38, 2019-02-01:
You can create in DeviceConf.txt a line by Model as you did. So if you are able to identify the model of the others LCT007, LST002

We know that we have to rebuild the management of widget creation for Lights/bulb #326

pipiche38, 2019-02-02:
Thanks for the path. 

I have pushed an update on beta branch . If you want to try

pipiche38, 2019-02-02:
@ben33880 if you want to chat , you can reach me on keybase.io (we could continue in french)

ben33880, 2019-02-03:
yes it's cool, thank you very much hue bulb LCT007 and strip LST002 works now.

firm 3.0f
beta plugin 4.1.x

ben33880, 2019-02-04:
for LCT001, i must insert new line DeviceConf.txt with this  : 
'LCT001':{'Ep':{'0b':{'0000':'','0003':'','0004':'','0005':'','0006':'','0008':'','0300':'','fc01':'','1000':'','Type':'ColorControl','ColorMode':'2'}},'Type':'','ProfileID':'c05e','ZDeviceID':'0210'},

pipiche38, 2019-02-04:
Integrated into DeviceCOnf.txt beta : b1373b490c17a2916b5edacc3f0e0e3a9cf6ecad

ben33880, 2019-02-05:
thanks.
now Hue LCT007, LCT001, LST002 works nice.
i only have this one, u can close if u want.