zigbeefordomoticz/Domoticz-Zigbee #334: Ikea TRADFRI wall socket plug switch

mcmikev, 2019-01-31:
Hi,

I have 3 ikea tradfri wall plugs and they get recognized as LVcontrol devices (AKA dimmers).
When removing them and re-add them (many many times) then sometimes they get recognized as the right type PLUG on/off. 
I don't know how to reproduce the right adding of the type, but normal include does not work right.
It is no dimmer just and on/off plug socket.

Is this a know thing or what can I do to have the right device type in my domoticz with Zigate plugin?

When the device is discoverd right then It is a plug and only one device is created (switch on/off) but when dimmer is discovered then there are 2 devices both dimmer added..

Help is appreciated :-)

pipiche38, 2019-01-31:
It would be great if you use the domoticz forum (english or french channel) next time.

Could you clarify which version of the plugin you are ( github branch as well)

Could you attached the DeviceList-xx.txt when those plugs are paired and the devices are created (whatever the domoticz widget )

mcmikev, 2019-01-31:
I use the latest beta plugin, with stable same behavior.
github branch beta

[DeviceList-48.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2818840/DeviceList-48.txt)

I have changed the device type in domoticz from dimmer to on/off device.

I have tried a complete new setup without any devices, same result. Sometimes the right plug type otherwise dimmer

mcmikev, 2019-01-31:
So it is not me doing something wrong ?

pipiche38, 2019-01-31:
No, you are not doing anything wrong. SO I think in the mean-time just replace using the Domoticz edit function

pipiche38, 2019-02-01:
Related to #326

pipiche38, 2019-02-03:
If you update to the latest version of beta branch. You should get only one device created.

mcmikev, 2019-02-11:
When I have the time to upgrade to firmware 3.0f I will test (going to re-add everything new).

Let you know!!

mcmikev, 2019-02-11:
Upgraded to 3.0f firmware and did git checkout pre-4.1 plugin and re-added all my devices.
So far so good. Were detected as plug and not lvcontrol anymore. And that in just one try :-)

closed 2019-02-13 by deennoo