zigbeefordomoticz/Domoticz-Zigbee #335: How to handle Philipps Hue remote control

pipiche38, 2019-02-10:
As firmware 3.0f and plugin pre-4.1 provide group management and cluster 0xFC000, how can we get support of the Philips Hue remote

ben33880, 2019-02-10:
[telco philips.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2848896/telco.philips.txt)
les logs lors de l'apparaige

ben33880, 2019-02-10:
[DiscoveryDevice-0017880103e4583d.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2848897/DiscoveryDevice-0017880103e4583d.txt)

le fichier dans Zdatas

deennoo, 2019-02-10:
J'en ai une également, qui n 'a jamais été appairer a un hub Philips, elle est donc "brut"

Elle crée deux widget dans Domoticz en fonction des EP 1 et 2

Il n'y a que le EP2 qui semble etre mis a jour

Voici quelque elements
ON : 
```
2019-02-10 22:37:46.054 (Zigate) Decode8095 - MsgData: 6001000602c46d01 len: 16
2019-02-10 22:37:46.054 (Zigate) Decode8095 - SQN: 60, Addr: c46d, Ep: 01, Cluster: 0006, Cmd: 01, Unknown: 02
```

OFF : 
```
2019-02-10 22:38:42.227 (Zigate) Decode8095 - MsgData: 0001000602c46d40 len: 16
2019-02-10 22:38:42.227 (Zigate) Decode8095 - SQN: 00, Addr: c46d, Ep: 01, Cluster: 0006, Cmd: 40, Unknown: 02
```

Luminosité + (le bouton en dessous de ON) : 
```
2019-02-10 22:39:41.499 (Zigate) Decode8085 - MsgData: 6401000802c46d02 len: 16
2019-02-10 22:39:41.499 (Zigate) Decode8085 - SQN: 64, Addr: c46d, Ep: 01, Cluster: 0008, Cmd: 02, Unknown: 02
2019-02-10 22:39:41.499 (Zigate) Decode8085 - Selector: click_down
2019-02-10 22:39:41.500 (Zigate) Update v2 Values 7:'70:' (Tele Hue 2)
```

Luminosité - (le boutton au dessus de OFF) : 

```
2019-02-10 22:41:00.781 (Zigate) Decode8085 - MsgData: 6701000802c46d02 len: 16
2019-02-10 22:41:00.781 (Zigate) Decode8085 - SQN: 67, Addr: c46d, Ep: 01, Cluster: 0008, Cmd: 02, Unknown: 02
2019-02-10 22:41:00.781 (Zigate) Decode8085 - Selector: click_down
2019-02-10 22:41:00.781 (Zigate) Update v2 Values 7:'70:' (Tele Hue 2)
```

lboue, 2019-02-10:
Bonjour,

Pour ma part:
- firmware 3.0f et plugin pre-4.1 également
- l’appairage du Hue remote control avec la zigate fonctionne
- lorsque j'appuie sur un des boutons, aucun état n'est remonté dans Domoticz 

[image]

```
cat DeviceList-15.txt 
2075 : {'Battery': {}, 'Status': 'inDB', 'Heartbeat': '5950', 'Model': 'RWL021', 'RSSI': 102, 'SQN': '3a', 'ProfileID': '0104', 'IEEE': '0017880104ad2d90', 'RIA': '1', 'Type': 'LvlControl/Switch/Voltage', 'ZDeviceID': '000c', 'MacCapa': '80', 'ReadAttributes': {'TimeStamps': {'01-0008': 1549811714, '02-0000': 1549804535, '02-0001': 1549804445, '01-0006': 1549811714}, 'Ep': {'01': {'0006': {'0000': '86'}, '0008': {'0000': '86'}}, '02': {'0001': {}, '0000': {'0005': '00', '0007': '00', '0010': '86', '000a': '86', '0004': '00', '0000': '00'}}}}, 'NbEp': '2', 'Stamp': {'MsgType': '8102', 'Time': '2019-02-10 16:15:14'}, 'Version': '3', 'Ep': {'01': {'0005': {}, 'ClusterType': {'344': 'LvlControl'}, '0008': {}, '0003': {}, '0004': {}, '0006': {}, '0000': {}}, '02': {'ClusterType': {'345': 'Voltage'}, 'fc00': {}, '000f': {}, '0003': {}, '0001': {}, '0019': {}, '0000': {}}}}
```
Voici le logs à appairage:

```
2019-02-10 14:13:40.372  Status: (Zigate) [0] NEW OBJECT: 2075 Trying to create Domoticz device(s)
2019-02-10 14:13:40.373  (Zigate) processNotinDBDevices - ready for creation: {'Battery': {}, 'Status': '8043', 'Heartbeat': '1', 'Model': {}, 'RSSI': {}, 'SQN': 'be', 'ProfileID': '0104', 'IEEE': '0017880104ad2d90', 'RIA': '1', 'Type': {}, 'ZDeviceID': '000c', 'MacCapa': '80', 'ReadAttributes': {'TimeStamps': {'02-0000': 1549804420}, 'Ep': {'02': {'0000': {}}}}, 'NbEp': '2', 'Stamp': {'MsgType': '  4d', 'Time': '2019-02-10 14:13:36'}, 'Version': '3', 'Ep': {'01': {'0005': {}, '0008': {}, '0003': {}, '0004': {}, '0006': {}, '0000': {}}, '02': {'fc00': {}, '000f': {}, '0003': {}, '0001': {}, '0019': {}, '0000': {}}}}
2019-02-10 14:13:40.373  (Zigate) CreatDomoDevice - Process EP : 01
2019-02-10 14:13:40.373  (Zigate) GetType - Model {} Profile ID : 0104 ZDeviceID : 000c
2019-02-10 14:13:40.373  (Zigate) GetType - Model not found in DeviceConf : {} Let's go for Cluster search
2019-02-10 14:13:40.374  (Zigate) GetType - check Type for Cluster : 0005
2019-02-10 14:13:40.374  (Zigate) GetType - Type will be set to : 
2019-02-10 14:13:40.374  (Zigate) GetType - check Type for Cluster : 0008
2019-02-10 14:13:40.374  (Zigate) GetType - Type will be set to : LvlControl
2019-02-10 14:13:40.374  (Zigate) GetType - check Type for Cluster : 0003
2019-02-10 14:13:40.374  (Zigate) GetType - Type will be set to : LvlControl/
2019-02-10 14:13:40.374  (Zigate) GetType - check Type for Cluster : 0004
2019-02-10 14:13:40.375  (Zigate) GetType - Type will be set to : LvlControl//
2019-02-10 14:13:40.375  (Zigate) GetType - check Type for Cluster : 0006
2019-02-10 14:13:40.375  (Zigate) GetType - Type will be set to : LvlControl///Switch
2019-02-10 14:13:40.375  (Zigate) GetType - check Type for Cluster : 0000
2019-02-10 14:13:40.375  (Zigate) GetType - Type will be set to : LvlControl///Switch/
2019-02-10 14:13:40.376  (Zigate) CreateDomoDevice - Type via GetType: ['LvlControl', 'Switch', ''] Ep : 01
2019-02-10 14:13:40.376  (Zigate) adding Type : LvlControl to Global Type: []
2019-02-10 14:13:40.376  (Zigate) adding Type : Switch to Global Type: ['LvlControl']
2019-02-10 14:13:40.376  (Zigate) CreateDomoDevice - Creating devices based on Type: ['LvlControl', 'Switch', '']
2019-02-10 14:13:40.376  (Zigate) CreateDomoDevice - DevId: 0017880104ad2d90 DevEp: 01 Type: LvlControl
2019-02-10 14:13:40.377  (Zigate) Creating device 'LvlControl-0017880104ad2d90-01'.
2019-02-10 14:13:40.379  (Zigate) CreatDomoDevice - Process EP : 02
2019-02-10 14:13:40.380  (Zigate) GetType - Model {} Profile ID : 0104 ZDeviceID : 000c
2019-02-10 14:13:40.380  (Zigate) GetType - Model not found in DeviceConf : {} Let's go for Cluster search
2019-02-10 14:13:40.380  (Zigate) GetType - check Type for Cluster : fc00
2019-02-10 14:13:40.380  (Zigate) GetType - Type will be set to : 
2019-02-10 14:13:40.380  (Zigate) GetType - check Type for Cluster : 000f
2019-02-10 14:13:40.380  (Zigate) GetType - Type will be set to : 
2019-02-10 14:13:40.381  (Zigate) GetType - check Type for Cluster : 0003
2019-02-10 14:13:40.381  (Zigate) GetType - Type will be set to : 
2019-02-10 14:13:40.381  (Zigate) GetType - check Type for Cluster : 0001
2019-02-10 14:13:40.381  (Zigate) GetType - Type will be set to : Voltage
2019-02-10 14:13:40.381  (Zigate) GetType - check Type for Cluster : 0019
2019-02-10 14:13:40.381  (Zigate) GetType - Type will be set to : Voltage/
2019-02-10 14:13:40.381  (Zigate) GetType - check Type for Cluster : 0000
2019-02-10 14:13:40.382  (Zigate) GetType - Type will be set to : Voltage//
2019-02-10 14:13:40.382  (Zigate) CreateDomoDevice - Type via GetType: ['Voltage', ''] Ep : 02
2019-02-10 14:13:40.382  (Zigate) adding Type : Voltage to Global Type: ['LvlControl', 'Switch']
2019-02-10 14:13:40.382  (Zigate) CreateDomoDevice - Creating devices based on Type: ['Voltage', '']
2019-02-10 14:13:40.382  (Zigate) CreateDomoDevice - DevId: 0017880104ad2d90 DevEp: 02 Type: Voltage
2019-02-10 14:13:40.383  (Zigate) Creating device 'Voltage-0017880104ad2d90-02'.
2019-02-10 14:13:40.385  (Zigate) CreateDomoDevice - DevId: 0017880104ad2d90 DevEp: 02 Type: 
2019-02-10 14:13:40.386  (Zigate) GlobalType: ['LvlControl', 'Switch', 'Voltage']
[... cut ...]
2019-02-10 14:14:05.602  (Zigate) receiveData - MsgType: 8000
2019-02-10 14:14:05.602  (Zigate) receivedData - MsgType: 8000 PacketType: 0100
2019-02-10 14:14:05.602  (Zigate) receiveStatusCmd - waitQ: 1 dataQ: 0 normalQ: 1
2019-02-10 14:14:05.603  (Zigate) sendData         - Cmd: 0100 waitQ: 0 dataQ: 0 normalQ: 0
2019-02-10 14:14:05.603  (Zigate) _sendData 0100
2019-02-10 14:14:05.604  (Zigate) ZigateEncode - Encodind data: 0100
```

Faut-il appairer avec une ampoule au préalable pour ce cela fonctionne ?

Cordialement,

pipiche38, 2019-02-11:
Question:
Dans le cas de la Commande Philipps RWL021, combien il y a t il de boutons ? Est-ce qu'il faut un widget de type Switch selector (comme pour l'Ikea), ou est-ce que un Switch/LvlControl est suffisant ?

pipiche38, 2019-02-11:
@lboue oui il faut que tu associes ta remote à une ampoule. @deennoo tu confirmes ?

ben33880, 2019-02-11:
il y a 4 boutons : on, off, plus, moins

ben33880, 2019-02-11:
OFF : 
2019-02-11 20:59:16.398 (zig) Decode8095 - MsgData: 0001000602827140 len: 16
2019-02-11 20:59:16.398 (zig) Decode8095 - SQN: 00, Addr: 8271, Ep: 01, Cluster: 0006, Cmd: 40, Unknown: 02

ON :
2019-02-11 20:59:48.969 (zig) Decode8095 - MsgData: 6201000602827101 len: 16
2019-02-11 20:59:48.969 (zig) Decode8095 - SQN: 62, Addr: 8271, Ep: 01, Cluster: 0006, Cmd: 01, Unknown: 02

(+) :
2019-02-11 21:00:52.441 (zig) Decode8085 - MsgData: 6501000802827102 len: 16
2019-02-11 21:00:52.441 (zig) Decode8085 - SQN: 65, Addr: 8271, Ep: 01, Cluster: 0008, Cmd: 02, Unknown: 02
2019-02-11 21:00:52.441 (zig) Decode8085 - Selector: click_down

(-) :
2019-02-11 21:00:26.792 (zig) Decode8085 - MsgData: 6401000802827102 len: 16
2019-02-11 21:00:26.792 (zig) Decode8085 - SQN: 64, Addr: 8271, Ep: 01, Cluster: 0008, Cmd: 02, Unknown: 02
2019-02-11 21:00:26.792 (zig) Decode8085 - Selector: click_down

appui long sur le - : 
2019-02-11 21:01:30.010 (zig) Decode8085 - MsgData: 6601000802827102 len: 16
2019-02-11 21:01:30.010 (zig) Decode8085 - SQN: 66, Addr: 8271, Ep: 01, Cluster: 0008, Cmd: 02, Unknown: 02
2019-02-11 21:01:30.010 (zig) Decode8085 - Selector: click_down
2019-02-11 21:01:30.762 (zig) Decode8085 - MsgData: 6701000802827102 len: 16
2019-02-11 21:01:30.762 (zig) Decode8085 - SQN: 67, Addr: 8271, Ep: 01, Cluster: 0008, Cmd: 02, Unknown: 02
2019-02-11 21:01:30.762 (zig) Decode8085 - Selector: click_down
2019-02-11 21:01:31.614 (zig) Decode8085 - MsgData: 6801000802827102 len: 16
2019-02-11 21:01:31.614 (zig) Decode8085 - SQN: 68, Addr: 8271, Ep: 01, Cluster: 0008, Cmd: 02, Unknown: 02
2019-02-11 21:01:31.614 (zig) Decode8085 - Selector: click_down
2019-02-11 21:01:31.814 (zig) Decode8085 - MsgData: 6901000802827103 len: 16
2019-02-11 21:01:31.814 (zig) Decode8085 - SQN: 69, Addr: 8271, Ep: 01, Cluster: 0008, Cmd: 03, Unknown: 02
2019-02-11 21:01:31.814 (zig) Decode8085 - Selector: release_down

appui long sur le + : 
2019-02-11 21:02:48.809 (zig) Decode8085 - MsgData: 6a01000802827102 len: 16
2019-02-11 21:02:48.809 (zig) Decode8085 - SQN: 6a, Addr: 8271, Ep: 01, Cluster: 0008, Cmd: 02, Unknown: 02
2019-02-11 21:02:48.809 (zig) Decode8085 - Selector: click_down
2019-02-11 21:02:49.610 (zig) Decode8085 - MsgData: 6b01000802827102 len: 16
2019-02-11 21:02:49.610 (zig) Decode8085 - SQN: 6b, Addr: 8271, Ep: 01, Cluster: 0008, Cmd: 02, Unknown: 02
2019-02-11 21:02:49.610 (zig) Decode8085 - Selector: click_down
2019-02-11 21:02:50.412 (zig) Decode8085 - MsgData: 6c01000802827102 len: 16
2019-02-11 21:02:50.412 (zig) Decode8085 - SQN: 6c, Addr: 8271, Ep: 01, Cluster: 0008, Cmd: 02, Unknown: 02
2019-02-11 21:02:50.412 (zig) Decode8085 - Selector: click_down
2019-02-11 21:02:50.913 (zig) Decode8085 - MsgData: 6d01000802827103 len: 16
2019-02-11 21:02:50.913 (zig) Decode8085 - SQN: 6d, Addr: 8271, Ep: 01, Cluster: 0008, Cmd: 03, Unknown: 02
2019-02-11 21:02:50.913 (zig) Decode8085 - Selector: release_down

appui long sur OFF : 
2019-02-11 21:03:28.033 (zig) Decode8095 - MsgData: 0001000602827140 len: 16
2019-02-11 21:03:28.033 (zig) Decode8095 - SQN: 00, Addr: 8271, Ep: 01, Cluster: 0006, Cmd: 40, Unknown: 02

appui long sur le ON : 
2019-02-11 21:03:59.341 (zig) Decode8095 - MsgData: 6f01000602827101 len: 16
2019-02-11 21:03:59.341 (zig) Decode8095 - SQN: 6f, Addr: 8271, Ep: 01, Cluster: 0006, Cmd: 01, Unknown: 02

pipiche38, 2019-02-12:
Bizare, pas de distinction entre + et - Meme commandes , meme Unknown ...

pipiche38, 2019-02-12:
@deennoo or @ben33880 do you have the possibility to check the Hue Remote behaviour with ZGUI ?

Connect the Zigate to ZGUI.
- then press HUE Remote + and capture the trace output in ZGUI
- then press HUE Remote - and capture the trace output in ZGUI

pipiche38, 2019-02-12:
https://github.com/fairecasoimeme/ZiGate/issues/147

deennoo, 2019-02-12:
> Question:
> Dans le cas de la Commande Philipps RWL021, combien il y a t il de boutons ? Est-ce qu'il faut un widget de type Switch selector (comme pour l'Ikea), ou est-ce que un Switch/LvlControl est suffisant ?

Un selector avec ON/Off/Dim+ et Dim - fait le boulot

deennoo, 2019-02-12:
> @lboue oui il faut que tu associes ta remote à une ampoule. @deennoo tu confirmes ?

Oui chez moi elle est appairer a une ampoule IKEA, ensuite la demarche est la meme que pour la telecommande Ikea Round

deennoo, 2019-02-12:
> @deennoo or @ben33880 do you have the possibility to check the Hue Remote behaviour with ZGUI ?
> 
> Connect the Zigate to ZGUI.
> 
> * then press HUE Remote + and capture the trace output in ZGUI
> * then press HUE Remote - and capture the trace output in ZGUI

For + (or Dimmer up) : 
Type: 0x8085 (Unrecognized)
21:22:19.462 <- 01 80 85 00 09 9E BF 01 00 08 02 C4 6D 02 8D 03

For - (or Dimmer Down) : 
21:23:01.070 <- 01 80 85 00 09 E1 C0 01 00 08 02 C4 6D 02 8D 03

i have to admit that group created with this remote and a bulb is fully fonctionnal

pipiche38, 2019-02-14:
@ben33880 @deennoo can you do once more a test ?

(1) Update the pre-4.1 branch
(2) remove the existing widget/device from Domoticz and Plugin
(3) redo a full pairing of the HUE remote
(4) Check the Logs when doing the Pairing - I'm interested on the 'Binding messages'

And then let see how the Remote <-> Plugin react when actioning the remote

deennoo, 2019-02-16:
Just redo a pairing with your last change : 

On Button : 

2019-02-16 05:58:44.530 (Zigate) ReadCluster - fc00 - f7f9/02 MsgAttrID: 0001, MsgAttType: 42, MsgAttSize: 0005, : 3000210000
2019-02-16 05:58:44.530 (Zigate) Decode8095 - SQN: 2b, Addr: f7f9, Ep: 01, Cluster: 0006, Cmd: 01, Unknown: 02
2019-02-16 05:58:44.681 (Zigate) ReadCluster - fc00 - f7f9/02 MsgAttrID: 0001, MsgAttType: 42, MsgAttSize: 0005, : 3002210100

Dim + : 

2019-02-16 05:59:41.557 (Zigate) ReadCluster - fc00 - f7f9/02 MsgAttrID: 0002, MsgAttType: 42, MsgAttSize: 0005, : 3000210000
2019-02-16 05:59:41.758 (Zigate) ReadCluster - fc00 - f7f9/02 MsgAttrID: 0002, MsgAttType: 42, MsgAttSize: 0005, : 3002210200

Dim - : 

2019-02-16 06:00:06.155 (Zigate) ReadCluster - fc00 - f7f9/02 MsgAttrID: 0003, MsgAttType: 42, MsgAttSize: 0005, : 3000210000
2019-02-16 06:00:06.305 (Zigate) ReadCluster - fc00 - f7f9/02 MsgAttrID: 0003, MsgAttType: 42, MsgAttSize: 0005, : 3002210100

Off : 

2019-02-16 06:00:40.696 (Zigate) ReadCluster - fc00 - f7f9/02 MsgAttrID: 0004, MsgAttType: 42, MsgAttSize: 0005, : 3000210000
2019-02-16 06:00:40.696 (Zigate) Decode8095 - SQN: 00, Addr: f7f9, Ep: 01, Cluster: 0006, Cmd: 40, Unknown: 02
2019-02-16 06:00:40.997 (Zigate) ReadCluster - fc00 - f7f9/02 MsgAttrID: 0004, MsgAttType: 42, MsgAttSize: 0005, : 3002210300

Plugin create 2 widgets : 
- level
- battery

Action on remote doesn't update any widget

deennoo, 2019-02-18:
Testing with the last Pre-4.1 version : 

Level Controle Widget is Created : 
 
On/Off : Ok
Dim+/Dim- : OK

Perfect integration and well done

pipiche38, 2019-02-18:
The DIM+/- long press needs some adjustement ...

pipiche38, 2019-02-18:
The duration provided when DIM+/- is not in seconds but in tenth of seconds

pipiche38, 2019-02-21:
Closed / Implemented in pre-4.1

closed 2019-02-21 by pipiche38