zigbeefordomoticz/Domoticz-Zigbee #337: Livolo 2 Gang Switch Zigbee

deennoo, 2019-02-12:
Hey guys ! 

I just get a livolo 2 Gang Zigbee switch : https://fr.aliexpress.com/item/Livolo-APP-Touch-Control-Interrupteur-Zigbee-WiFi-Domotique-Intelligent-T-l-commande-Travail-avec-Echo-seulement/32918552579.html?spm=a2g0s.9042311.0.0.40696c3745S41u

Working on channel 26 only

Some dev are done on Zigbee2mqtt : https://github.com/Koenkk/zigbee2mqtt/issues/592

Device discovering frame can be found here : 

https://drive.google.com/open?id=1cfGdCO45b0WEtQzfln2wrZz0uqm6PgEL

pipiche38, 2019-02-12:
I suggest that you start with ZGUI in order to see if the Zigate is able to retreive all necessary information.
From what I see iun the Discovery Frame:
EndPoint: 06

Last the Log of the all Discovery process from Device Announce to TimeOut would be helpfull.

pipiche38, 2019-02-12:
In summary:
- Comunication with the Device is Ok. This mean Zigate talk to Livolo on channel 26 back and forth

From the Disocvery Frame:
"8043": {"06": "1b00ef750e0601040000000200000003010006"}, 
        SQN: 1b
        Status: 00
        NwkID: ef75
        Len: 0e
        EP: 06
        Profile: 0104
        DeviceId: 0000
        BitField: 00
        ClusterIn:      02
              0000
              0003
       ClusterOut: 01
             0006
"ZDeviceID": {"0000": "06"}, 
"ProfileID": {"0104": "06"}}

pipiche38, 2019-02-12:
Bug found, in the Decode8043 - fix to follow

pipiche38, 2019-02-12:
Fix: 0dc9067..239add0
New json format for the discovery ...

deennoo, 2019-02-12:
here is some input from ZWGUI :

deennoo, 2019-02-12:
Thx to @KiwiHC16 and https://github.com/KiwiHC16/Abeille/issues/570

Who give a full detailled information about how to drive Livolo Switch.

pipiche38, 2019-02-13:
With ZGUI can you get the Attribute of the Device ?
- Read Attribute on Device, Ep: 06, Cluster 0x0006 and Attribute 0x0000

With ZGUI did you try to send Level Control (as mentioned in the Abeille subject)
Level: 108, Level: 1, 

Can you also post the latest discovery file (which is now Json format).

deennoo, 2019-02-13:
Yes i manage to Drive it sending 108/1 or 108/2 or 1/1 or 1/2.

This Works fine.

I have to manage how to set Infinite permit join, because when this end, Switch is out.

I have to get some help to trying to configure reporting

pipiche38, 2019-02-13:
108 On
1 Off
Gauche : Timing 1
Droit: Timing 2

pipiche38, 2019-02-14:
Here is what I suggest for the widget side:
- Switch Selector ( like we do for the Xiaomi ). With 
Left On
Left Off
Right On
Right Off

Just need to get the Disocvery file to see how we can identify this . Especially is there a Model name

deennoo, 2019-02-16:
Ok for selector, 

Even better if this can be a separate on/off switch as we all ready do for xiaomi double switch : This switch is plug to a real bulb, put it on a Plan will show a selector instead of a switch, of course we can make a dummy on/off switch for each side but that a trick.

Trying to pair it with the last release off pre 4.1 : 

`2019-02-16 05:10:27.156 Status: (Zigate) [2] NEW OBJECT: 3a30 Model Name: {}
2019-02-16 05:10:27.156 Status: (Zigate) [2] NEW OBJECT: 3a30 Model Name: {}
2019-02-16 05:10:27.156 Status: (Zigate) [2] NEW OBJECT: 3a30 TimeOut in 0045 restarting at 0x004d
2019-02-16 05:10:27.156 Status: (Zigate) [2] NEW OBJECT: 3a30 Request Model Name
2019-02-16 05:10:27.308 Status: (Zigate) [-] NEW OBJECT: 3a30 Request Simple Descriptor for Ep: 06
2019-02-16 05:10:27.912 Status: (Zigate) [-] NEW OBJECT: 3a30 Request Simple Descriptor for Ep: 06
2019-02-16 05:10:28.466 Status: (Zigate) [-] NEW OBJECT: 3a30 Request Simple Descriptor for Ep: 06
2019-02-16 05:10:29.171 (Zigate) Decode8043 - Overwrite ProfileID {} with 0104 from Ep: 06
2019-02-16 05:10:29.171 (Zigate) Decode8043 - Overwrite ZDeviceID {} with 0104 from Ep: 06
2019-02-16 05:10:29.070 Status: (Zigate) [-] NEW OBJECT: 3a30 Request Simple Descriptor for Ep: 06
2019-02-16 05:10:29.171 Status: (Zigate) [-] NEW OBJECT: 3a30 Simple Descriptor EP 06
2019-02-16 05:10:29.171 Status: (Zigate) [-] NEW OBJECT: 3a30 ProfileID 0104
2019-02-16 05:10:29.171 Status: (Zigate) [-] NEW OBJECT: 3a30 ZDeviceID 0000
2019-02-16 05:10:29.171 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster IN Count: 02
2019-02-16 05:10:29.171 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster In 1: 0000
2019-02-16 05:10:29.171 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster In 2: 0003
2019-02-16 05:10:29.172 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster OUT Count: 01
2019-02-16 05:10:29.172 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster Out 1: 0006
2019-02-16 05:10:29.323 Status: (Zigate) [-] NEW OBJECT: 3a30 Simple Descriptor EP 06
2019-02-16 05:10:29.323 Status: (Zigate) [-] NEW OBJECT: 3a30 ProfileID 0104
2019-02-16 05:10:29.323 Status: (Zigate) [-] NEW OBJECT: 3a30 ZDeviceID 0000
2019-02-16 05:10:29.323 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster IN Count: 02
2019-02-16 05:10:29.323 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster In 1: 0000
2019-02-16 05:10:29.323 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster In 2: 0003
2019-02-16 05:10:29.323 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster OUT Count: 01
2019-02-16 05:10:29.323 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster Out 1: 0006
2019-02-16 05:10:29.424 Status: (Zigate) [-] NEW OBJECT: 3a30 Simple Descriptor EP 06
2019-02-16 05:10:29.424 Status: (Zigate) [-] NEW OBJECT: 3a30 ProfileID 0104
2019-02-16 05:10:29.424 Status: (Zigate) [-] NEW OBJECT: 3a30 ZDeviceID 0000
2019-02-16 05:10:29.424 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster IN Count: 02
2019-02-16 05:10:29.424 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster In 1: 0000
2019-02-16 05:10:29.424 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster In 2: 0003
2019-02-16 05:10:29.424 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster OUT Count: 01
2019-02-16 05:10:29.424 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster Out 1: 0006
2019-02-16 05:10:29.575 Status: (Zigate) [-] NEW OBJECT: 3a30 Simple Descriptor EP 06
2019-02-16 05:10:29.575 Status: (Zigate) [-] NEW OBJECT: 3a30 ProfileID 0104
2019-02-16 05:10:29.575 Status: (Zigate) [-] NEW OBJECT: 3a30 ZDeviceID 0000
2019-02-16 05:10:29.575 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster IN Count: 02
2019-02-16 05:10:29.575 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster In 1: 0000
2019-02-16 05:10:29.575 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster In 2: 0003
2019-02-16 05:10:29.575 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster OUT Count: 01
2019-02-16 05:10:29.575 Status: (Zigate) [-] NEW OBJECT: 3a30 Cluster Out 1: 0006
2019-02-16 05:10:32.132 (Zigate) Request a Bind for 3a30/06 on Cluster 0006
2019-02-16 05:10:32.132 Status: (Zigate) [3] NEW OBJECT: 3a30 Model Name: TI0001
2019-02-16 05:10:32.132 Status: (Zigate) [3] NEW OBJECT: 3a30 Model Name: TI0001
2019-02-16 05:10:32.132 Status: (Zigate) [3] NEW OBJECT: 3a30 Model Name: TI0001
2019-02-16 05:10:32.132 Status: (Zigate) [3] NEW OBJECT: 3a30 Model Name: TI0001
2019-02-16 05:10:32.132 Status: (Zigate) [3] NEW OBJECT: 3a30 Trying to create Domoticz device(s)
2019-02-16 05:10:32.132 Error: (Zigate) Domoticz widget creation failed. ID: -1, Name: 'Switch-00124b0018015b39-06', nValue: 0, sValue: ''
`

You can get device discovery here : https://drive.google.com/open?id=1hAMvEm4w8-7c7Icb_58OT98eUnPh9eoX

deennoo, 2019-02-18:
After Testing this morning version : 

This switch doesn't work any more : 

Widget is created as a simple ON/OFF (Not a selector One), trying to activate it doesn't give any output as the previous version on the log and it doesn't work.

pipiche38, 2019-02-18:
Please provide Logs in order to understand what happen ... Without it is really hard to see

deennoo, 2019-02-20:
Tested with last pre 4.1 version.

Plugin create 2 Widgets : 
Zigate - LivoloSWL-00124b0018015b39-06
Zigate - LivoloSWR-00124b0018015b39-06

Those widget are working.

Now i'm facing to a loosing connection issue : 

After Pairing and Adding Livolo stop working a few minutes after, this is very similar to what is provied on this issue on Abeille : 

"There is a difference in the switch behaviour that depends on coordinator. The switch always sends 'Data request' commands 3-4 times per second (!!!). If there is no answer even for less than a second the switch sends 'Orphan Notification' then 'Beacon Requests'. If it is paired with original livolo gateway, it's ok, it sends these requests as long as it's necessary, but if it is custom coordinator the switch stops working after a second, and there is only one way to make it work again, it's touch the sensor for 5 secs or power it off then on. So, my switch works fine as long as the connection is perfect, in other way it stops working after some time."

and this one on Zigbee2Mqtt : https://github.com/Koenkk/zigbee2mqtt/issues/592

pipiche38, 2019-02-21:
Dedicated branch ( git checkout livolo )

pipiche38, 2019-02-27:
I suggest to park this issue, until Zigate is able to handle it.
Close for now and we will re-open when compatible/managed by Zigate

pipiche38, 2019-07-25:
To be work on.

pipiche38, 2019-07-25:
Akila referring that Livolo will be supported on 3.1b ( https://zigate.fr/forum/topic/livolo-compatible-zigbee/#postid-596 )

pipiche38, 2019-08-02:
In order to force the Extebded PanID
Conf/PluginConf-xx.txt
  "extendedPANID": 5149013381575969,

 Status: (Zigate) Network state UP, PANID: 1fb8 extPANID: 0x00124b00198d7521 Channel: 26

Permit to Join must be always on, as after an electric PowerOff, Livolo restart the Pairing process

KiwiHC16, 2019-08-02:
Salut
There is no need to force the PanId with mine.

pipiche38, 2019-08-02:
@KiwiHC16 

> 
> Salut
> There is no need to force the PanId with mine.

Fine for you, anyway I saw quiet a lot of changes done by Akila on the Firmware for 3.1b,

KiwiHC16, 2019-08-02:
Salut,

Oui suite à ma demande.

Mais ces changement ne changent rien sur le fonctionnement de base de la 
zigate et le fait qu'il n'est pas nécessaire de forcer le PanId. Cette 
info (Forcer le PanId) vient de m2q et pour moi c'est une fausse info.

pipiche38, 2019-10-13:
drop!

closed 2019-10-13 by pipiche38