zigbeefordomoticz/Domoticz-Zigbee #343: TRADFRI bulb E27 WS opal 980lm

pipiche38, 2019-02-13:
TRADFRI bulb E27 WS opal 980lm not recognized an no Widget created

pipiche38, 2019-02-13:
0c8a : {'RIA': '2', 'Manufacturer': '117c', 'ReceiveOnIdle': 'On', 'ConfigureReporting': {'TimeStamps': {'01-0019': 0, '01-0300': 1550085687, '01-0006': 1550085627, '01-0000': 0, '01-1000': 0, '01-0004': 0, '01-0b05': 0, '01-0003': 0, '01-0005': 0, '01-0020': 0, '01-0008': 1550085627}, 'Ep': {'01': {'0020': {}, '0b05': {}, '0005': {}, '0300': '85', '0003': '', '0019': {}, '0004': {}, '0006': '00', '1000': {}, '0000': {}, '0008': '00'}}}, 'RSSI': 246, 'Status': 'inDB', 'Model': 'TRADFRI bulb E27 WS opal 980lm', 'Stamp': {'Time': '2019-02-13 21:37:54', 'MsgType': '8702'}, 'ZDeviceID': '0220', 'NbEp': '1', 'Heartbeat': '1051', 'Ep': {'01': {'0b05': {}, '0008': 'fe', '0300': {}, '0003': {}, 'ClusterType': {}, '0000': {}, '0005': {}, '0020': {}, '0019': {}, '0004': {}, '0006': '01', '1000': {}}}, 'ColorInfos': {'ColorMode': 2, 'Y': '26909', 'X': '30138'}, 'LogicalType': 'Router', 'IEEE': 'd0cf5efffe1dd78e', 'PowerSource': 'Main', 'Battery': {}, 'MacCapa': '8e', 'Version': '3', 'ProfileID': 'c05e', 'Type': {}, 'GroupMgt': {'01': {}}, 'SQN': {}, 'ReadAttributes': {'TimeStamps': {'01-0300': 1550089891, '01-0006': 1550090171, '01-0000': 1550085307, '01-0008': 1550090171}, 'Ep': {'01': {'0300': {'0001': '86', '0004': '00', '0008': '00', '0000': '86', '0003': '00'}, '0006': {'0000': '00'}, '0008': {'0000': '00'}, '0000': {'0007': '00', '0010': '86', '0004': '00', '0005': '00', '0000': '00'}}}}, 'DeviceType': 'FFD'}

{
    "004d": "0c8ad0cf5efffe1dd78e8e",
    "8043": {},
    "8045": {},
    "Attribute Discovery": {
        "Ep": {
            "01": {
                "0000": {
                    "0002": "20"
                },
                "0003": {
                    "fffd": "21"
                },
                "0004": {
                    "fffd": "21"
                },
                "0005": {
                    "0002": "21"
                },
                "0006": {
                    "4001": "21"
                },
                "0008": {
                    "000f": "18"
                },
                "0300": {
                    "0003": "21"
                },
                "0b05": {
                    "0101": "23"
                }
            }
        }
    },
    "Ep": {},
    "IEEE": "d0cf5efffe1dd78e",
    "MacCapa": "8e",
    "ProfileID": {},
    "ZDeviceID": {}
}

pipiche38, 2019-02-13:
9449a09..b8564f3

Gemies, 2019-02-14:
This is the log i got after two unsuccesfull pairing attempts!

[Log from not paireble bulb E27 WS opal 980lm.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2866746/Log.from.not.paireble.bulb.E27.WS.opal.980lm.txt)

Gemies, 2019-02-14:
After a git pull tonight it paired with the bulb, this issue will be closed

pipiche38, 2019-02-14:
Thanks @Gemies for reporting and tetsing it

closed 2019-02-14 by pipiche38