zigbeefordomoticz/Domoticz-Zigbee #344: TRADFRI bulb E27 CWS opal 600lm false widget

lukytux, 2019-02-14:
The widget created it's the wrong one. It's différent as in the past and it doesn't work.
Changing intensity doesn't work
Changing Color doesn't work
On/off OK
Her print screen from the group but the widget is the same
[image]

Logs:
2019-02-14 08:47:05.145  Status: (Zigate) Permit Join is On
2019-02-14 08:47:05.145  (Zigate) Ping - received
2019-02-14 08:47:22.742  Status: (Zigate) [0] NEW OBJECT: df3f Device Annouce IEEE: 000d6ffffe2511c7
2019-02-14 08:47:27.353  (Zigate) checkTOwaitForStatus - Request a reTransmit of Command : 0045/df3f (0)
2019-02-14 08:47:27.354  Status: (Zigate) [0] NEW OBJECT: df3f Model Name: {}
2019-02-14 08:47:27.354  Status: (Zigate) [0] NEW OBJECT: df3f Model Name: {}
2019-02-14 08:47:28.560  Status: (Zigate) [0] NEW OBJECT: df3f Device Annouce IEEE: 000d6ffffe2511c7
2019-02-14 08:47:28.612  Status: (Zigate) [0] NEW OBJECT: df3f Device Annouce IEEE: 000d6ffffe2511c7
2019-02-14 08:47:29.065  Status: (Zigate) [-] NEW OBJECT: df3f Request Simple Descriptor for Ep: 01
2019-02-14 08:47:29.171  Status: (Zigate) [-] NEW OBJECT: df3f Simple Descriptor EP 01
2019-02-14 08:47:29.171  (Zigate) Decode8043 - Overwrite ProfileID {} with c05e from Ep: 01 
2019-02-14 08:47:29.171  Status: (Zigate) [-] NEW OBJECT: df3f ProfileID c05e
2019-02-14 08:47:29.171  (Zigate) Decode8043 - Overwrite ZDeviceID {} with c05e from Ep: 01 
2019-02-14 08:47:29.172  Status: (Zigate) [-] NEW OBJECT: df3f ZDeviceID 0200
2019-02-14 08:47:29.172  Status: (Zigate) [-] NEW OBJECT: df3f Cluster IN Count: 09
2019-02-14 08:47:29.172  Status: (Zigate) [-] NEW OBJECT: df3f Cluster In 1: 0000
2019-02-14 08:47:29.172  Status: (Zigate) [-] NEW OBJECT: df3f Cluster In 2: 0003
2019-02-14 08:47:29.172  Status: (Zigate) [-] NEW OBJECT: df3f Cluster In 3: 0004
2019-02-14 08:47:29.172  Status: (Zigate) [-] NEW OBJECT: df3f Cluster In 4: 0005
2019-02-14 08:47:29.172  Status: (Zigate) [-] NEW OBJECT: df3f Cluster In 5: 0006
2019-02-14 08:47:29.172  Status: (Zigate) [-] NEW OBJECT: df3f Cluster In 6: 0008
2019-02-14 08:47:29.172  Status: (Zigate) [-] NEW OBJECT: df3f Cluster In 7: 0300
2019-02-14 08:47:29.172  Status: (Zigate) [-] NEW OBJECT: df3f Cluster In 8: 0b05
2019-02-14 08:47:29.172  Status: (Zigate) [-] NEW OBJECT: df3f Cluster In 9: 1000
2019-02-14 08:47:29.172  Status: (Zigate) [-] NEW OBJECT: df3f Cluster OUT Count: 04
2019-02-14 08:47:29.172  Status: (Zigate) [-] NEW OBJECT: df3f Cluster Out 1: 0005
2019-02-14 08:47:29.172  Status: (Zigate) [-] NEW OBJECT: df3f Cluster Out 2: 0019
2019-02-14 08:47:29.173  Status: (Zigate) [-] NEW OBJECT: df3f Cluster Out 3: 0020
2019-02-14 08:47:29.173  Status: (Zigate) [-] NEW OBJECT: df3f Cluster Out 4: 1000
2019-02-14 08:47:32.380  Status: (Zigate) [0] NEW OBJECT: df3f Model Name: {}
2019-02-14 08:47:32.380  Status: (Zigate) [0] NEW OBJECT: df3f Model Name: {}
2019-02-14 08:47:32.380  Status: (Zigate) [0] NEW OBJECT: df3f Request Model Name
2019-02-14 08:47:32.382  Status: (Zigate) [0] NEW OBJECT: df3f Request Attribute for Cluster 0x0300 to get ColorMode
2019-02-14 08:47:37.343  Status: (Zigate) [1] NEW OBJECT: df3f Model Name: TRADFRI bulb E27 CWS opal 600lm
2019-02-14 08:47:37.343  Status: (Zigate) [1] NEW OBJECT: df3f Model Name: TRADFRI bulb E27 CWS opal 600lm
2019-02-14 08:47:37.343  Status: (Zigate) [1] NEW OBJECT: df3f Model Name: TRADFRI bulb E27 CWS opal 600lm
2019-02-14 08:47:37.344  Status: (Zigate) [1] NEW OBJECT: df3f Trying to create Domoticz device(s)
2019-02-14 08:47:37.347  (Zigate) looking for bind ep: 01
2019-02-14 08:47:37.347  (Zigate) looking for bind ep: 01 cluster: 0005
2019-02-14 08:47:37.348  (Zigate) looking for bind ep: 01 cluster: 0003
2019-02-14 08:47:37.348  (Zigate) looking for bind ep: 01 cluster: 0020
2019-02-14 08:47:37.348  (Zigate) looking for bind ep: 01 cluster: 0008
2019-02-14 08:47:37.348  (Zigate) Finaly Request a Bind for df3f/01 - 0008
2019-02-14 08:47:37.350  (Zigate) looking for bind ep: 01 cluster: 0006
2019-02-14 08:47:37.350  (Zigate) Finaly Request a Bind for df3f/01 - 0006
2019-02-14 08:47:37.350  (Zigate) looking for bind ep: 01 cluster: 0b05
2019-02-14 08:47:37.350  (Zigate) looking for bind ep: 01 cluster: 0019
2019-02-14 08:47:37.350  (Zigate) looking for bind ep: 01 cluster: 1000
2019-02-14 08:47:37.351  (Zigate) looking for bind ep: 01 cluster: 0004
2019-02-14 08:47:37.351  (Zigate) looking for bind ep: 01 cluster: 0300
2019-02-14 08:47:37.351  (Zigate) Finaly Request a Bind for df3f/01 - 0300
2019-02-14 08:47:37.351  (Zigate) looking for bind ep: 01 cluster: 0000
2019-02-14 08:47:37.422  (Zigate) Devices size has changed , let's write ListOfDevices on disk
2019-02-14 08:47:37.423  (Zigate) Write /home/pi/domoticz/plugins/Domoticz-Zigate/./Data/DeviceList-16.txt = {'df3f': {'ConfigSource': 'DeviceConf', 'Stamp': {'MsgType': '8102', 'Time': '2019-02-14 08:47:32'}, 'Version': '3', 'Battery': 
{}, 'ZDeviceID': '0200', 'NbEp': '1', 'Bind': {'0008': {'Phase': 'requested', 'Stamp': 1550130457, 'Status': ''}, '0006': {'Phase': 'requested', 'Stamp': 1550130457, 'Status': ''}, '0300': {'Phase': 'requested', 'Stamp': 1550130457, 'Sta
tus': ''}}, 'RSSI': 87, 'IEEE': '000d6ffffe2511c7', 'Model': 'TRADFRI bulb E27 CWS opal 600lm', 'Heartbeat': '2', 'Ep': {'01': {'0005': {}, '0003': {}, '0020': {}, '0008': {}, '0006': {}, '0b05': {}, 'Type': 'ColorControlRGB', 'ClusterTy
pe': {'87': 'ColorControlRGB'}, '0019': {}, '1000': {}, '0004': {}, '0300': {}, '0000': {}}}, 'MacCapa': '8e', 'RIA': '1', 'ReadAttributes': {'TimeStamps': {'01-0300': 1550130452, '01-0000': 1550130452}, 'Ep': {'01': {'0300': {'0008': '0
0', '0000': '86', '0003': '00', '0004': '00', '0001': '86'}, '0000': {'0005': '00', '0007': '00', '0000': '00', '0010': '86', '0004': '00'}}}}, 'Type': 'ColorControlRGB', 'Status': 'inDB', 'ColorInfos': {'Y': '26870', 'X': '30015', 'Colo
rMode': 2}, 'SQN': {}, 'ProfileID': 'c05e'}, 'ffff': {'IEEE': '00158d00021986a2', 'version': '3', 'Ep': {'01': {'0008': {}, '0006': {}, '0004': {}}}, 'PowerSource': 'Main'}}
2019-02-14 08:47:37.430  (Zigate) Write /home/pi/domoticz/plugins/Domoticz-Zigate/www/zigate/reports/DeviceList-16.json = {'df3f': {'ConfigSource': 'DeviceConf', 'Stamp': {'MsgType': '8102', 'Time': '2019-02-14 08:47:32'}, 'Version': '3'
, 'Battery': {}, 'ZDeviceID': '0200', 'NbEp': '1', 'Bind': {'0008': {'Phase': 'requested', 'Stamp': 1550130457, 'Status': ''}, '0006': {'Phase': 'requested', 'Stamp': 1550130457, 'Status': ''}, '0300': {'Phase': 'requested', 'Stamp': 155
0130457, 'Status': ''}}, 'RSSI': 87, 'IEEE': '000d6ffffe2511c7', 'Model': 'TRADFRI bulb E27 CWS opal 600lm', 'Heartbeat': '2', 'Ep': {'01': {'0005': {}, '0003': {}, '0020': {}, '0008': {}, '0006': {}, '0b05': {}, 'Type': 'ColorControlRGB
', 'ClusterType': {'87': 'ColorControlRGB'}, '0019': {}, '1000': {}, '0004': {}, '0300': {}, '0000': {}}}, 'MacCapa': '8e', 'RIA': '1', 'ReadAttributes': {'TimeStamps': {'01-0300': 1550130452, '01-0000': 1550130452}, 'Ep': {'01': {'0300'
: {'0008': '00', '0000': '86', '0003': '00', '0004': '00', '0001': '86'}, '0000': {'0005': '00', '0007': '00', '0000': '00', '0010': '86', '0004': '00'}}}}, 'Type': 'ColorControlRGB', 'Status': 'inDB', 'ColorInfos': {'Y': '26870', 'X': '
30015', 'ColorMode': 2}, 'SQN': {}, 'ProfileID': 'c05e'}, 'ffff': {'IEEE': '00158d00021986a2', 'version': '3', 'Ep': {'01': {'0008': {}, '0006': {}, '0004': {}}}, 'PowerSource': 'Main'}}

pipiche38, 2019-02-14:
Can you find domoticz.db (should be one level up of the plugins)

sqlite3 domoticz.db
> select * from DeviceStatus where HardwareID=16 and DeviceID = 00158d00021986a2;
> .exit

If you can copy-paste the result of that , this would be really great

lukytux, 2019-02-14:
```
sqlite> select * from DeviceStatus where HardwareID=16 and DeviceID = 00158d00021986a2;
Error: unrecognized token: "00158d00021986a2"
sqlite> select * from DeviceStatus where HardwareID=16;
87|16|000d6ffffe2511c7|1|Lumière escalier_stock|1|241|2|7|0|6|255|1|54|2019-02-14 14:51:40|87|0.0|1.0|0.0|1.0|||54|0|0|||
88|16|d0cf5efffe1e85ee|2|Zigate - telco_escalier|1|244|62|18|0|6|255|3|30|2019-02-14 14:39:46|88|0.0|1.0|0.0|1.0|||10|0|0||LevelActions:fHx8fHw=;LevelNames:T2ZmfFRvZ2dsZU9uT2ZmfExlZnRfY2xpY2t8UmlnaHRfY2xpY2t8VXBfY2xpY2t8RG93bl9jbGljaw==;LevelOffHidden:ZmFsc2U=;SelectorStyle:MQ==|
89|16|Zigate-02-16|3|Zigate - Status 16|1|243|22|0|0|12|255|1|Ready|2019-02-14 14:39:12|89|0.0|1.0|0.0|1.0|||0|0|0|||
90|16|Zigate-03-16|4|Zigate -  Notifications 16|1|243|19|0|0|12|255|0|Groups management startup completed|2019-02-14 14:39:12|90|0.0|1.0|0.0|1.0|||0|0|0|||
91|16|90fd9ffffeebcd71|5|Lumière escalier|1|241|2|7|0|3|255|1|54|2019-02-14 14:51:41|91|0.0|1.0|0.0|1.0|||54|0|0|||
92|16|e8e9|6|Zigate - groupescaliere8e9|1|241|2|7|0|12|255|1|54|2019-02-14 14:49:02|92|0.0|1.0|0.0|1.0|||54|0|0|||
sqlite> 
```

pipiche38, 2019-02-14:
Super.
They are all as expected (241,2,7) and must be RGB

lukytux, 2019-02-14:
Super... Mmh... but the result is not as expected

pipiche38, 2019-02-14:
In Device List could you do a screenshot of the list of devices ?

lukytux, 2019-02-14:
[image]

pipiche38, 2019-02-14:
That is indeed the same conclusion. I did test on my end and all the widgets seems to be correctly created.
in your list they are indeed RGB controlwidget.

Now why you have such widget instead, I have no clue. Maybe some custom mode on Domoticz itself ....

lukytux, 2019-02-14:
Mmh... Nothing custom in my domoticz. 
It's doing a lot of thing from long time... So i really don't want to restart...
I have to look how to solve this problem with the domoticz team i presume.
Thx

lukytux, 2019-02-14:
I have found... The problem are from chrome web browser! No problem with firefox.
So this have nothing todo with zigate

closed 2019-02-14 by lukytux

pipiche38, 2019-02-14:
Ouf !