zigbeefordomoticz/Domoticz-Zigbee #345: No feedback from devices.

deve87, 2019-02-14:
I running Zigate USB stabel 4.0 plugin

Got 4 Philip Hue bulbs and devices don't update status in Domoticz. When I turn on wall switch, light will go to 100%, but Domoticz stays in last info.

I also have a Elko GLED316 Zigbee dimmer. It works changed dim, On/Off from Domoticz. But when I do any changes on the dimmer. Domoticz don't get status, and stays on last value.

pipiche38, 2019-02-14:
COuld you kindly use either the French or English forum for that question ?

Your first place to get support is via the Forums.

    English channel : https://www.domoticz.com/forum/viewforum.php?f=68
    French Channel : https://easydomoticz.com/forum/viewforum.php?f=28&sid=a8633a9d8cb2acccb32872543b50fd47

I'm going to close this issue. We might re-open if there is a real issue/bug

pipiche38, 2019-02-15:
This should be fixed on pre-4.1 ( 1ad7f2d..2b627e2 )

closed 2019-02-15 by pipiche38