zigbeefordomoticz/Domoticz-Zigbee #347: Left and right button support tradfri remote for colour temp

Gemies, 2019-02-14:
Your first place to get support is via the Forums.

    English channel : https://www.domoticz.com/forum/viewforum.php?f=68
    French Channel : https://easydomoticz.com/forum/viewforum.php?f=28&sid=a8633a9d8cb2acccb32872543b50fd47

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Plugin (please complete the following information):**
 - OS: [e.g. Linux Fedora 28]
 - Plugin version
 - Plugin branch

**Additional context**
Add any other context about the problem here.

Gemies, 2019-02-14:
The Zigate can now read all the button actions off the tradfri remote. It would be nice if the paired bulbs would also change colour temp when pressing left and right button of the tradfri remote.
[remote colour temp log.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2866851/remote.colour.temp.log.txt)

pipiche38, 2019-02-14:
@Gemies it would indeed be great. For that you need to program it by retrieving the event, and then triggering changes to the Bulb.
Basically this is how the Tradfri Hub is reacting by using scene which are programmed there

So every one can do what every he wants

pipiche38, 2019-03-04:
Let's reopen this issue. 

When capturing the remote event, we can consider sending a color change  to a device or a group.

pipiche38, 2019-03-06:
First version is delivered on beta branch : 701bf9a

As White Color schema, it works pretty well.
For Color one, this is not fancy and not usfull, I'm thinking to go with a series of preset defined color that left/right could scroll

pipiche38, 2019-03-06:
Final Version : ee03c0e..c02f09b

You need to add the Remote (which have been paired before with a Bulb) to the Group where you have already added also the Zigate.

closed 2019-03-06 by pipiche38