zigbeefordomoticz/Domoticz-Zigbee #359: Make Legrand Window covering working

pipiche38, 2019-02-21:
With firmware 30f where it support cluster 0x0101 and Windowscovering command, please make it work

pipiche38, 2019-02-27:
Pairing OK
On/Off Ok

Level Control and Reporting to be tested

levertacoaliastoto26, 2019-03-02:
Some logs : 
 2019-03-02 16:36:42.449 (Zigate) UpdateDevice - ( Volet roulant) 1:On
2019-03-02 16:36:42.412 Status: User: Admin initiated a switch command (5/Volet roulant/On)
2019-03-02 16:36:44.855 Status: User: Admin initiated a switch command (5/Volet roulant/Set Level)
2019-03-02 16:36:44.877 Error: (Zigate) 'onCommand' failed 'TypeError':'not all arguments converted during string formatting'.
2019-03-02 16:36:44.878 Error: (Zigate) ----> Line 571 in '/usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py', function onCommand
2019-03-02 16:36:44.878 Error: (Zigate) ----> Line 375 in '/usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py', function onCommand
2019-03-02 16:36:44.878 Error: (Zigate) ----> Line 177 in '/usr/local/domoticz/var/plugins/Domoticz-Zigate/Modules/command.py', function mgtCommand
2019-03-02 16:36:55.325 (Zigate) ReadCluster - 0102 - d860/01 - Attribute: 0008, Type: 20, Size: 0001 Data: 32-50
2019-03-02 16:36:55.325 (Zigate) Current position lift

pipiche38, 2019-03-02:
A priori cela devrait etre corrigé dans la version : 4c3349d..812715d (pre-4.1)

- Correction du Level Control
- L'etat est bien remonté. A priroi ton volet est à 50%

pipiche38, 2019-03-04:
As https://github.com/fairecasoimeme/ZiGate/issues/125 lift (level control) not supported by the Shutter.

closed 2019-03-04 by pipiche38