zigbeefordomoticz/Domoticz-Zigbee #383: PluginConf naming is not the same as DeviceList for HardwareId with only one digit

SylvainPer, 2019-03-13:
For example, with 'HardwareID':'2', the files are: 
 DeviceList-2.txt 
 PluginConf-02.txt 

due to 
```
        # Import PluginConf.txt
        self.filename = self.pluginConfig + "PluginConf-%02d.txt" %hardwareid
        if not os.path.isfile(self.filename) :
            self.filename = self.pluginConfig + "PluginConf.txt"
```
in PluginConf.py l84

pipiche38, 2019-03-13:
Thanks for reporting the issue and agree with you that there is a lack of consistency here.
And this is bringing also an other issue which might happen in case the HardwareID of the plugin is above 99 in Domoticz, we would face an other issue. The probability to reach that number is somehow limited, but still exist.

SylvainPer, 2019-03-14:
Not sure that there is a problem for 3 digits, %02d only add 0 to have 2 digits but do nothing if the number is >= 10. It's not a limitation to 2 digits.

pipiche38, 2019-03-14:
1a38084..4096686

closed 2019-03-14 by pipiche38