zigbeefordomoticz/Domoticz-Zigbee #388: lumi.sensor_smoke doesn't work as expected

deennoo, 2019-03-27:
I have a lumi.sensor_smoke

integration open a IAS Zone 

When device get some smoke it start to ring, and IAS message is : 

`2019-03-27 12:01:56.615 (Zigate) Decode8401 - Reception Zone status change notification : 07010500028949000100ff0000
2019-03-27 12:01:56.615 (Zigate) Decode8401 - MsgSQN: 07 MsgSrcAddr: 8949 MsgEp:01 MsgClusterId: 0500 MsgZoneStatus: 0001 MsgExtStatus: 00 MsgZoneID: ff MsgDelay: 0000
2019-03-27 12:01:56.615 (Zigate) Decode8401 MsgZoneStatus: 01`

On Domoticz device isn't turned On, that mean that there is no notification or event.

Only event beome when pressing the only button on device to reset it : 

`2019-03-27 12:02:48.367 (Zigate) ReadCluster - 0000 / 0005 - Recepion Model: >lumi.sensor_smoke<
2019-03-27 12:02:48.568 (Zigate) Decode8401 - Reception Zone status change notification : 0a010500028949000100ff0000
2019-03-27 12:02:48.568 (Zigate) Decode8401 - MsgSQN: 0a MsgSrcAddr: 8949 MsgEp:01 MsgClusterId: 0500 MsgZoneStatus: 0001 MsgExtStatus: 00 MsgZoneID: ff MsgDelay: 0000
2019-03-27 12:02:48.568 (Zigate) Decode8401 MsgZoneStatus: 01
2019-03-27 12:02:48.568 Status: (Zigate) IAS Zone change for device:8949 - alarm1: 1, alaram2: 0, tamper: 0, battery: 0, Support Reporting: 0, restore Reporting: 0, trouble: 0, acmain: 0, test: 0, battdef: 0`

When this device is already turned ON, even if it get some smoke or button pressed : nothing append

pipiche38, 2019-03-27:
@deennoo , this looks strange and the messages that you put doesn't reflect what is in the code. Are you sure that you had copy-paste all ? Especially when the device get smoke ?

If I look to the raw data, the data are the same, so there is no reason that it doesn't work in one case, and it works in the other

Last , if the smoke stop (or if you release the test/reset button) the plugin today expect to receive a message from the device with a status change ( Alarm 0). It seems that it is not happening and that is why you still get the the Domoticz Widget stayed ON. On the other side, I think that in Domoticz, the smoke widget has a reset/release button, no ?

Here is what you got when detecting Smoke
```
2019-03-27 12:01:56.615 (Zigate) Decode8401 - Reception Zone status change notification : 07010500028949000100ff0000
```
Here is what you got when pressing the rset button
```
2019-03-27 12:02:48.568 (Zigate) Decode8401 - Reception Zone status change notification : 0a010500028949000100ff0000
```

Here after are the log reformatted:

```
2019-03-27 12:01:56.615 (Zigate) Decode8401 - Reception Zone status change notification : 07010500028949000100ff0000
2019-03-27 12:01:56.615 (Zigate) Decode8401 - MsgSQN: 07 MsgSrcAddr: 8949 MsgEp:01 MsgClusterId: 0500 MsgZoneStatus: 0001 MsgExtStatus: 00 MsgZoneID: ff MsgDelay: 0000
2019-03-27 12:01:56.615 (Zigate) Decode8401 MsgZoneStatus: 01
```

On Domoticz device isn't turned On, that mean that there is no notification or event.

Only event become when pressing the only button on device to reset it :

```
2019-03-27 12:02:48.367 (Zigate) ReadCluster - 0000 / 0005 - Recepion Model: >lumi.sensor_smoke<
2019-03-27 12:02:48.568 (Zigate) Decode8401 - Reception Zone status change notification : 0a010500028949000100ff0000
2019-03-27 12:02:48.568 (Zigate) Decode8401 - MsgSQN: 0a MsgSrcAddr: 8949 MsgEp:01 MsgClusterId: 0500 MsgZoneStatus: 0001 MsgExtStatus: 00 MsgZoneID: ff MsgDelay: 0000
2019-03-27 12:02:48.568 (Zigate) Decode8401 MsgZoneStatus: 01 2019-03-27 12:02:48.568 Status: (Zigate) IAS Zone change for device:8949 - alarm1: 1, alaram2: 0, tamper: 0, battery: 0, Support Reporting: 0, restore R
```

pipiche38, 2019-03-30:
When you want to ack a Smoke alert (true or test), you can use the Reset to remove the alert

[image]

pipiche38, 2019-05-25:
Just did a Test and it works has expected.

-> When smoke - the Widget goes ON
-> When no more smoke the Widget goes OFF

- When detecting smoke, there is an IAS message sent

```
May 25 19:59:49 rasp domoticz[24757]: (Zigate-DEV) Decode8401 - Reception Zone status change notification : 9001050002d123000100ff0000
May 25 19:59:49 rasp domoticz[24757]: (Zigate-DEV) Decode8401 - MsgSQN: 90 MsgSrcAddr: d123 MsgEp:01 MsgClusterId: 0500 MsgZoneStatus: 0001 MsgExtStatus: 00 MsgZoneID: ff MsgDelay: 0000
May 25 19:59:49 rasp domoticz[24757]: Status: (Zigate-DEV) IAS Zone change for device:d123  - alarm1: 1, alaram2: 0, tamper: 0, battery: 0, Support Reporting: 0, restore Reporting: 0, trouble: 0, acmain: 0, test: 0, battdef: 0
May 25 19:59:49 rasp domoticz[24757]: (Zigate-DEV) Decode8401 MsgZoneStatus: 01
May 25 19:59:49 rasp domoticz[24757]: (Zigate-DEV) UpdateDevice - (Zigate-DEV - lumi.sensor_smoke_Smoke-00158d0002b225cc-01) 1:On
May 25 19:59:49 rasp domoticz[24757]: (Zigate-DEV) ReadCluster - 0000 ff01 Saddr: d123 ClusterData: 0121990c0328210421a81305210b00062407000000000a210000082104106420ff96230002040200
May 25 19:59:49 rasp domoticz[24757]: (Zigate-DEV) ReadCluster - 0000/ff01 Saddr: d123 Battery: 100 OldValue: 98 Voltage: 3225
May 25 19:59:49 rasp domoticz[24757]: (Zigate-DEV) ReadCluster - 0000/ff01 Saddr: d123 stag0421: a813
May 25 19:59:49 rasp domoticz[24757]: (Zigate-DEV) ReadCluster - 0000/ff01 Saddr: d123 stag0521: 0b00
May 25 19:59:49 rasp domoticz[24757]: (Zigate-DEV) ReadCluster - 0000/ff01 Saddr: d123 stag0624: 0700
May 25 19:59:49 rasp domoticz[24757]: (Zigate-DEV) ReadCluster - 0000/ff01 Saddr: d123 stag0821: 0410
May 25 19:59:49 rasp domoticz[24757]: (Zigate-DEV) ReadCluster - 0000/ff01 Saddr: d123 stag0a21: 0000

when no more smoke, there is a new IAS message sent:
```

May 25 19:59:59 rasp domoticz[24757]: (Zigate-DEV) Decode8401 - Reception Zone status change notification : 9201050002d123000000ff0000
May 25 19:59:59 rasp domoticz[24757]: (Zigate-DEV) Decode8401 - MsgSQN: 92 MsgSrcAddr: d123 MsgEp:01 MsgClusterId: 0500 MsgZoneStatus: 0000 MsgExtStatus: 00 MsgZoneID: ff MsgDelay: 0000
May 25 19:59:59 rasp domoticz[24757]: Status: (Zigate-DEV) IAS Zone change for device:d123  - alarm1: 0, alaram2: 0, tamper: 0, battery: 0, Support Reporting: 0, restore Reporting: 0, trouble: 0, acmain: 0, test: 0, battdef: 0
May 25 19:59:59 rasp domoticz[24757]: (Zigate-DEV) Decode8401 MsgZoneStatus: 00
May 25 19:59:59 rasp domoticz[24757]: (Zigate-DEV) UpdateDevice - (Zigate-DEV - lumi.sensor_smoke_Smoke-00158d0002b225cc-01) 0:Off
May 25 19:59:59 rasp domoticz[24757]: (Zigate-DEV) ReadCluster - 0000 ff01 Saddr: d123 ClusterData: 0121990c0328210421a81305210b00062408000000000a2100000821041064200396230000000000
May 25 19:59:59 rasp domoticz[24757]: (Zigate-DEV) ReadCluster - 0000/ff01 Saddr: d123 Battery: 100 OldValue: 98 Voltage: 3225
May 25 19:59:59 rasp domoticz[24757]: (Zigate-DEV) ReadCluster - 0000/ff01 Saddr: d123 stag0421: a813
May 25 19:59:59 rasp domoticz[24757]: (Zigate-DEV) ReadCluster - 0000/ff01 Saddr: d123 stag0521: 0b00
May 25 19:59:59 rasp domoticz[24757]: (Zigate-DEV) ReadCluster - 0000/ff01 Saddr: d123 stag0624: 0800
May 25 19:59:59 rasp domoticz[24757]: (Zigate-DEV) ReadCluster - 0000/ff01 Saddr: d123 stag0821: 0410
May 25 19:59:59 rasp domoticz[24757]: (Zigate-DEV) ReadCluster - 0000/ff01 Saddr: d123 stag0a21: 0000
```
```

pipiche38, 2019-07-08:
I suggest to close that one. If you still have problem pleaqse re-open

closed 2019-07-08 by pipiche38