zigbeefordomoticz/Domoticz-Zigbee #421: New switch Xiaomi Aqara with gyroscope doesn't work with plugin

pipiche38, 2019-04-25:


cray19003, 2019-04-25:
Voici les résultats des commandes 

Click : 2019-04-25 10:03:25.680 (Zigate) Cluster0012 - unknown Model: lumi.sensor_switch.aq3 for this Attribute: 0055 value: 0001
Double Click : 2019-04-25 10:03:28.136 (Zigate) Cluster0012 - unknown Model: lumi.sensor_switch.aq3 for this Attribute: 0055 value: 0002
Long click : 2019-04-25 10:03:32.300 (Zigate) Cluster0012 - unknown Model: lumi.sensor_switch.aq3 for this Attribute: 0055 value: 0010
Click release : 2019-04-25 10:03:34.654 (Zigate) Cluster0012 - unknown Model: lumi.sensor_switch.aq3 for this Attribute: 0055 value: 0011
Shake : 2019-04-25 10:03:44.277 (Zigate) Cluster0012 - unknown Model: lumi.sensor_switch.aq3 for this Attribute: 0055 value: 0012

cray19003, 2019-04-25:
b5dd : {'NbEp': '1', 'RIA': '5', 'Type': {}, 'Status': 'UNKNOW', 'Model': 'lumi.sensor_switch.aq3', 'Ep': {'01': {'0000': {}}}, 'ProfileID': {}, 'App Version': '5', 'MacCapa': '80', 'IEEE': '00158d0002af8204', 'Heartbeat': '14', 'Battery': {}, 'SQN': '00', 'RSSI': 144, 'Stamp': {'MsgType': '8102', 'Time': '2019-04-25 09:54:45', 'LastSeen': 1556178820}, 'ZDeviceID': {}, 'Version': '3'}
9850 : {'NbEp': '1', 'RIA': '5', 'Type': {}, 'Status': 'UNKNOW', 'Model': 'lumi.sensor_switch.aq3', 'Ep': {'01': {'0000': {}}}, 'ProfileID': {}, 'App Version': '5', 'MacCapa': '80', 'IEEE': '00158d0002af8204', 'Heartbeat': '27', 'Battery': {}, 'SQN': '00', 'RSSI': 132, 'Stamp': {'MsgType': '8048', 'Time': '2019-04-25 09:53:38', 'LastSeen': 1556178771}, 'ZDeviceID': {}, 'Version': '3'}
0f58 : {'NbEp': '1', 'RIA': '5', 'Type': {}, 'Status': 'UNKNOW', 'Model': 'lumi.sensor_switch.aq3', 'Ep': {'01': {'0000': {}}}, 'ProfileID': {}, 'App Version': '5', 'MacCapa': '80', 'IEEE': '00158d0002af8204', 'Heartbeat': '32', 'Battery': {}, 'SQN': '01', 'RSSI': 111, 'Stamp': {'MsgType': '8048', 'Time': '2019-04-25 09:52:46', 'LastSeen': 1556178756}, 'ZDeviceID': {}, 'Version': '3'}
0000 : {'PowerSource': 'Main', 'Ep': {'01': {'0006': {}, '0004': {}, '0008': {}}}, 'IEEE': '00158d0001ed4716', 'version': '3'}
65d0 : {'ConfigureReporting': {'TimeStamps': {'01-0403': 1556178827, '01-0402': 1556178827, '01-0405': 1556178827, '01-0000': 1556178827, '01-0003': 0, '01-ffff': 0}, 'Ep': {'01': {'0000': {}, '0402': '', '0403': {}, '0003': {}, 'ffff': {}, '0405': {}}}}, 'RIA': '0', 'Type': 'Temp/Humi/Baro', 'Status': 'inDB', 'Model': 'lumi.weather', 'Ep': {'01': {'ClusterType': {'9': 'Temp+Hum', '8': 'Temp+Hum+Baro', '11': 'Humi', '10': 'Temp', '12': 'Baro'}, '0000': {}, 'Type': 'Temp/Humi/Baro', '0402': 25.7, '0403': 983.5, '0003': {}, 'ffff': {}, '0405': 43.4}}, 'Bind': {'0405': {'Phase': 'requested', 'Stamp': 1556178827, 'Status': ''}}, 'ConfigSource': 'DeviceConf', 'ProfileID': {}, 'App Version': '3', 'APS Failure': {'Status Code': 'f0', 'Status Msg': 'Pending transaction has expired and data discarded ', 'Stamp': 1556178965}, 'MacCapa': '80', 'IEEE': '00158d0003141148', 'Heartbeat': '29', 'Battery': {}, 'SQN': 'bf', 'ReadAttributes': {'TimeStamps': {'01-0403': 1556178827, '01-0402': 1556178827, '01-0405': 1556178827, '01-0000': 1556178827}, 'Ep': {'01': {'0402': {'0000': '00'}, '0403': {'0014': '00', '0010': '00', '0000': '00'}, '0000': {}, '0405': {'0000': '00'}}}}, 'RSSI': 120, 'Stamp': {'MsgType': '8702', 'Time': '2019-04-25 09:56:05', 'LastSeen': 1556178824}, 'ZDeviceID': {}, 'Version': '3'}

cray19003, 2019-04-25:
https://fr.aliexpress.com/item/Original-Xiaomi-Aqara-Smart-Wireless-Switch-Key-Built-In-Gyro-Multi-Functional-Intelligent-Remote-Control-ZigBee/32990494901.html?spm=a2g0w.10010108.1000016.1.4af8757bUpHzVO&isOrigTitle=true

C'est celui que j'ai acheté (plus cher car il était pas en promo).

pipiche38, 2019-04-25:
@cray19003 merci pour tout ca.
Il y a-t-il des events tel que Drop (tu laisses tomber le switch ) , Rotation , Slide (tu le deplaces sur un plan horizontal) ?

cray19003, 2019-04-25:
Non c'est les seules valeurs que j'ai pu obtenir.

pipiche38, 2019-04-25:
Voila, c'est poussé sur la branche beta

`git checkout beta
`

Tu relances le plugin et tu refais l'association.
Un Switch Selector devrait etre créé ...

J'ai pas pu tester donc surveilles les logs, si il y a un soucis. tu peux mettre les infos ici.

Je suis également joiniable en chat sur Keybase.io @pipiche

cray19003, 2019-04-25:
Super j'essaye ça tout de suite. Merci.

pipiche38, 2019-04-25:
https://zigate.fr/forum/topic/xiaomi-aqara-switch-with-gyro-lumi-sensor_switch-aq3/

cray19003, 2019-04-25:
Re,
J'ai des petites erreurs au démarrage (qui empêche de fonctionner).

 2019-04-25 11:21:54.810 (Zigate) GetType - Model lumi.sensor_switch.aq3 Profile ID : {} ZDeviceID : {}
2019-04-25 11:21:54.811 (Zigate) GetType - Model lumi.sensor_switch.aq3 Profile ID : {} ZDeviceID : {}
2019-04-25 11:21:54.810 Status: (Zigate) [0] NEW OBJECT: 700c Model Name: lumi.sensor_switch.aq3
2019-04-25 11:21:54.810 Status: (Zigate) [0] NEW OBJECT: 700c Model Name: lumi.sensor_switch.aq3
2019-04-25 11:21:54.811 Error: (Zigate) 'onHeartbeat' failed 'KeyError'.
2019-04-25 11:21:54.811 Error: (Zigate) ----> Line 657 in '/usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2019-04-25 11:21:54.811 Error: (Zigate) ----> Line 533 in '/usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2019-04-25 11:21:54.811 Error: (Zigate) ----> Line 408 in '/usr/local/domoticz/var/plugins/Domoticz-Zigate/Modules/heartbeat.py', function processListOfDevices
2019-04-25 11:21:54.811 Error: (Zigate) ----> Line 305 in '/usr/local/domoticz/var/plugins/Domoticz-Zigate/Modules/heartbeat.py', function processNotinDBDevices

pipiche38, 2019-04-25:
Peux-tu:
1- arreter le plugin 
2- regarder dans le fichier DeviceList-xx.txt qu'il n'y a pas de trace de ce device (tu peux chercher 00158d0002af8204 ). Supprime la ou les lignes en question

pipiche38, 2019-04-25:
PS/ A priori le device a été créé dans Domoticz, il faudrait que tu le supprimes également

cray19003, 2019-04-25:
J'avais tout effacé le dossier Domoticz-Zigate. J'ai également tout effacé dans Domoticz.

pipiche38, 2019-04-25:
ok

cray19003, 2019-04-25:
2019-04-25 11:44:09.177 (Zigate) Debug: 0
2019-04-25 11:44:09.193 (Zigate) Configure Permit To Join
2019-04-25 11:44:09.193 (Zigate) Request discovery mode for 254 seconds
2019-04-25 11:44:09.002 Status: (Zigate) Started.
2019-04-25 11:44:09.175 Status: (Zigate) Entering work loop.
2019-04-25 11:44:09.176 Status: (Zigate) Initialized version beta-4.2.0, author 'zaraki673 pipiche38'
2019-04-25 11:44:09.177 Status: (Zigate) Zigate plugin beta-4.2.0 started
2019-04-25 11:44:09.177 Status: (Zigate) Python Version - 3.5.1 (default, Jan 29 2018, 14:18:59)
2019-04-25 11:44:09.177 [GCC 4.9.3 20150311 (prerelease)]
2019-04-25 11:44:09.177 Status: (Zigate) Switching Hearbeat to 5 s interval
2019-04-25 11:44:09.177 Status: (Zigate) DomoticzVersion: 4.10604
2019-04-25 11:44:09.177 Status: (Zigate) DomoticzHash: 77a0b09-modified
2019-04-25 11:44:09.177 Status: (Zigate) DomoticzBuildTime: 2019-04-23 04:14:27
2019-04-25 11:44:09.181 Status: (Zigate) Opening DomoticzDB in raw
2019-04-25 11:44:09.182 Status: (Zigate) load PluginConf
2019-04-25 11:44:09.182 Status: (Zigate) PluginConf: /usr/local/domoticz/var/plugins/Domoticz-Zigate/Conf/PluginConf.txt
2019-04-25 11:44:09.182 Status: (Zigate) -vibrationAqarasensitivity: medium
2019-04-25 11:44:09.182 Status: (Zigate) -enablegroupmanagement: 0
2019-04-25 11:44:09.183 Status: (Zigate) -resetMotiondelay: 30
2019-04-25 11:44:09.183 Status: (Zigate) -logFORMAT: 0
2019-04-25 11:44:09.183 Status: (Zigate) -logLQI: 0
2019-04-25 11:44:09.183 Status: (Zigate) -channel: ['11', '15', '19', '20', '25', '26']
2019-04-25 11:44:09.185 Status: (Zigate) Plugin Database: DeviceList-3.txt
2019-04-25 11:44:09.191 Status: (Zigate) DeviceConf loaded
2019-04-25 11:44:09.191 Status: (Zigate) load ListOfDevice
2019-04-25 11:44:09.192 Status: (Zigate) Entries loaded from /usr/local/domoticz/var/plugins/Domoticz-Zigate/Data/DeviceList-3.txt : 0
2019-04-25 11:44:09.192 Status: (Zigate) Connection Name: Zigate, Transport: Serial, Address: /dev/ttyUSB0
2019-04-25 11:44:09.295 Status: (Zigate) Major Version Num: 0003
2019-04-25 11:44:09.295 Status: (Zigate) Installer Version Number: 030f
2019-04-25 11:44:09.496 Status: (Zigate) Zigate in Permit Join: On
2019-04-25 11:44:09.598 Status: (Zigate) Zigate addresses ieee: 00158d0001ed4716 , short addr: 0000
2019-04-25 11:44:09.599 Status: (Zigate) Network state UP, PANID: e20e extPANID: 0xe79fc4115bf4d0d5 Channel: 11
2019-04-25 11:44:09.700 Status: (Zigate) Number of devices recently active in Zigate = 4
2019-04-25 11:44:09.700 Status: (Zigate) [00] DevID = 00 Network addr = 24f9 IEEE = 00158d0002c3b294 LQI = 156 Power = 00 not found in ListOfDevices
2019-04-25 11:44:09.700 Status: (Zigate) [01] DevID = 01 Network addr = 2877 IEEE = 00158d0002af8204 LQI = 224 Power = 00 not found in ListOfDevices
2019-04-25 11:44:09.700 Status: (Zigate) [02] DevID = 02 Network addr = 65d0 IEEE = 00158d0003141148 LQI = 123 Power = 00 not found in ListOfDevices
2019-04-25 11:44:09.700 Status: (Zigate) [03] DevID = 03 Network addr = 9f9b IEEE = 00158d0002d6f722 LQI = 137 Power = 01 not found in ListOfDevices
2019-04-25 11:44:19.218 Status: (Zigate) Plugin with Zigate firmware 030f correctly initialized
2019-04-25 11:44:31.305 Error: (Zigate) Decode8102 - Receiving a message from unknown device : 24f9 with Data : 1124f90101010055002100020001
2019-04-25 11:45:00.999 Status: (Zigate) Leave indication from IEEE: 00158d0002af8204 , Status: 00
2019-04-25 11:45:02.453 Status: (Zigate) Device Annoucement ShortAddr: 8b1e, IEEE: 00158d0002af8204
2019-04-25 11:45:04.212 Status: (Zigate) [-] NEW OBJECT: 8b1e Request Simple Descriptor for Ep: 01
2019-04-25 11:45:07.418 (Zigate) Decode8043 - Overwrite ProfileID {} with 0104 from Ep: 01
2019-04-25 11:45:07.418 (Zigate) Decode8043 - Overwrite ZDeviceID {} with 0104 from Ep: 01
2019-04-25 11:45:07.418 Status: (Zigate) [-] NEW OBJECT: 8b1e Simple Descriptor EP 01
2019-04-25 11:45:07.418 Status: (Zigate) [-] NEW OBJECT: 8b1e ProfileID 0104
2019-04-25 11:45:07.418 Status: (Zigate) [-] NEW OBJECT: 8b1e ZDeviceID 5f01
2019-04-25 11:45:07.418 Status: (Zigate) [-] NEW OBJECT: 8b1e Cluster IN Count: 04
2019-04-25 11:45:07.418 Status: (Zigate) [-] NEW OBJECT: 8b1e Cluster In 1: 0000
2019-04-25 11:45:07.418 Status: (Zigate) [-] NEW OBJECT: 8b1e Cluster In 2: 0012
2019-04-25 11:45:07.418 Status: (Zigate) [-] NEW OBJECT: 8b1e Cluster In 3: 0006
2019-04-25 11:45:07.418 Status: (Zigate) [-] NEW OBJECT: 8b1e Cluster In 4: 0001
2019-04-25 11:45:07.418 Status: (Zigate) [-] NEW OBJECT: 8b1e Cluster OUT Count: 01
2019-04-25 11:45:07.418 Status: (Zigate) [-] NEW OBJECT: 8b1e Cluster Out 1: 0000
2019-04-25 11:45:09.221 (Zigate) ReadAttributeRequest_0000 - Key: 8b1e
2019-04-25 11:45:09.222 (Zigate) Too early, let's try to get the Model
2019-04-25 11:45:09.221 Status: (Zigate) [0] NEW OBJECT: 8b1e Request Model Name
2019-04-25 11:45:14.231 (Zigate) ReadAttributeRequest_0000 - Key: 8b1e
2019-04-25 11:45:14.232 (Zigate) Too early, let's try to get the Model
2019-04-25 11:45:14.231 Status: (Zigate) [1] NEW OBJECT: 8b1e Request Model Name
2019-04-25 11:45:19.193 (Zigate) ReadAttributeRequest_0000 - Key: 8b1e
2019-04-25 11:45:19.194 (Zigate) GetType - Model {} Profile ID : 0104 ZDeviceID : 5f01
2019-04-25 11:45:19.194 (Zigate) GetType - Model {} not found with Ep: 01 in DeviceConf. Continue with ClusterSearch
2019-04-25 11:45:19.194 (Zigate) GetType - ClusterSearch return : Voltage/Switch
[... cut ...]
2019-04-25 11:45:19.195 Error: (Zigate) ----> Line 613 in '/usr/local/domoticz/var/plugins/Domoticz-Zigate/Modules/domoticz.py', function CreateDomoDevice
2019-04-25 11:45:19.195 Error: (Zigate) ----> Line 36 in '/usr/local/domoticz/var/plugins/Domoticz-Zigate/Modules/domoticz.py', function deviceName
2019-04-25 11:45:22.257 (Zigate) ReadCluster - 0000 / 0005 - Recepion Model: >lumi.sensor_swit<
2019-04-25 11:45:28.577 Error: (Zigate) Decode8102 - Receiving a message from unknown device : 9f9b with Data : 2b9f9b010000ff01004200270328610521ea000727000000000000000008211701092100010a2100006420016520ff66219900

cray19003, 2019-04-25:
J'obtiens ça après avoir réinstaller un nouveau plugin Ziagte dans Domoticz et effacé tous les fichier Data/DevicesList.txt et que j'essaye de remettre l'interrupteur.

pipiche38, 2019-04-25:
désole. une erreur dans le fichier de Configuration.
Peux-tu :  

git pull pour recuperer la derniere version
si tu peux faire un RAZ c'est top

cray19003, 2019-04-25:
J'ai fait un reset et réinstaller tous mes objets. Ca marche, je n'ai plus d'erreur. Par contre le switch est toujours indiqué comme 'ON'.
Par contre bien que je vois des messages (ci-dessous) dans le log, je n'ai aucun effet sur l'icone dans Domoticz (mais c'est peut-être normal).

Si j'abuse pas trop, comment est-ce que je récuppère la valeur dans un script LUA ?
Merci.

cray19003, 2019-04-25:
2019-04-25 12:33:30.192 (Zigate) ReadCluster - ClusterId=0012 - Switch Aqara: EP: 01 Value: 1
2019-04-25 12:33:30.192 (Zigate) UpdateDevice - ( Switch) 1:00
2019-04-25 12:33:38.665 (Zigate) ReadCluster - ClusterId=0012 - Switch Aqara: EP: 01 Value: 16
2019-04-25 12:33:38.866 (Zigate) ReadCluster - ClusterId=0012 - Switch Aqara: EP: 01 Value: 17
2019-04-25 12:33:39.318 (Zigate) ReadCluster - ClusterId=0012 - Switch Aqara: EP: 01 Value: 2
2019-04-25 12:33:39.319 (Zigate) UpdateDevice - ( Switch) 2:10
2019-04-25 12:33:44.080 (Zigate) ReadCluster - ClusterId=0012 - Switch Aqara: EP: 01 Value: 2
2019-04-25 12:33:44.081 (Zigate) UpdateDevice - ( Switch) 2:10

cray19003, 2019-04-25:
J'ai essayé ça, mais sans succès. Il détecte bien le changemnt d'état (premier print) mais pas le deuxième en fonction de la valeur.
J'ai fait une erreur ?
_____________________________

commandArray = {}
if (devicechanged['Switch']) then
	print('click detected')
		if devicechanged['Switch_Value']=='2' then
			print (klik)	
		end
end
return commandArray

pipiche38, 2019-04-25:
Que veux-tu dire, il n'a pas créé un Switch Selector avec 5 elements ?
"Click|Double Click|Long Click|Release Click|Shake"

Peux-tu me donner les valeurs pour les actions en fonction de ce qui s'affiche dans les logs

cray19003, 2019-04-25:
[image]
Je pense que si.
J'obtiens ça dans le log (avec mon script).
D'ailleurs je ne détecte pas le 'shake' comme un changement d'état.
1: Click|2: Double Click|16: Long Click|17: Release Click|18: Shake

 2019-04-25 13:07:19.024 (Zigate) UpdateDevice - ( Switch) 1:00
2019-04-25 13:07:19.028 Status: LUA: click detected
2019-04-25 13:07:24.136 (Zigate) ReadCluster - ClusterId=0012 - Switch Aqara: EP: 01 Value: 2
2019-04-25 13:07:24.136 (Zigate) UpdateDevice - ( Switch) 2:10
2019-04-25 13:07:24.140 Status: LUA: click detected
2019-04-25 13:07:30.049 (Zigate) ReadCluster - ClusterId=0012 - Switch Aqara: EP: 01 Value: 2
2019-04-25 13:07:30.050 (Zigate) UpdateDevice - ( Switch) 2:10
2019-04-25 13:07:30.055 Status: LUA: click detected
2019-04-25 13:07:38.119 (Zigate) ReadCluster - ClusterId=0012 - Switch Aqara: EP: 01 Value: 18
2019-04-25 13:07:39.973 (Zigate) ReadCluster - ClusterId=0012 - Switch Aqara: EP: 01 Value: 16
2019-04-25 13:07:40.776 (Zigate) ReadCluster - ClusterId=0012 - Switch Aqara: EP: 01 Value: 17

pipiche38, 2019-04-25:
git pull
tu as juste à relancer le plugin , les mapping doivent etre correcte maintenant, y compris le shake

cray19003, 2019-04-25:
Le shake déclenche bien le script. Tout devrait marcher à présent.
Merci beaucoup.

pipiche38, 2019-04-25:
Je te laisse mettre un message sur le forum, pour dire que c'est tout bon ;-)

pipiche38, 2019-04-25:
merge in beta branch

closed 2019-04-25 by pipiche38