zigbeefordomoticz/Domoticz-Zigbee #429: Compatibilité Pluzzy

pipiche38, 2019-05-07:
When Pluzzy firmware will be available , make sure that the Pluzzy devices are correctly managed

pipiche38, 2019-05-11:
https://github.com/fairecasoimeme/ZiGate/issues/30

pipiche38, 2019-07-01:
https://github.com/fairecasoimeme/ZiGate_HA1.2

pipiche38, 2019-07-01:
For those whom want to test Pluzzy please use the referenced Firmware, but make sure to switch to beta version of the plugin for now.
If the integration work (which I don't think) becomes important I might create a dedicated branch for the work

cleminfo, 2019-07-01:
Il y a un tuto pour passer sur ce firmware ?

pipiche38, 2019-07-01:
I beleive that is the same as the standard Zigate firmware

cleminfo, 2019-07-08:
C'est pas très clair, quel est le fichier bin a utilisé pour la mise à jour du firmware ?

pipiche38, 2019-07-08:
Le firmware est disponible ici: https://github.com/fairecasoimeme/ZiGate_HA1.2/releases

pipiche38, 2019-07-13:
Open branch 'pluzzy' to work on it

pipiche38, 2019-07-17:
https://github.com/fairecasoimeme/ZiGate_HA1.2/issues/2

pipiche38, 2019-07-17:
https://github.com/fairecasoimeme/ZiGate_HA1.2/issues/1

pipiche38, 2019-07-18:
Une première version sur la branche pluzzy ( 77ec89865600bfa7c8d29f6ce0cf89dbe5ec2da0 ) fonctionne avec le capteur de Temp+Humi
Par contre le firmware n'est pas aligné avec le firmware classique de la Zigate et donc je patch le Payload

A first version is working on pluzzy branch ( 77ec89865600bfa7c8d29f6ce0cf89dbe5ec2da0 ). it works with Temp+Humi sensor
However the current zigate-pluzzy-firmware is not aligned with the classical zigate and I have to patch the payload in order to re-order some bytes (quiet nasty)

pipiche38, 2019-07-18:
Documentation on Temp+Humi sensor: https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&ved=2ahUKEwjuiY3ixb7jAhUaA2MBHc-hAy0QFjAAegQIARAC&url=http%3A%2F%2Fwww.netvox.com.tw%2Fusermanual%2FZ716A_UserManual_ENG_WEB.pdf&usg=AOvVaw2qaOK1dstzVKoVpEbFXU5c

pipiche38, 2019-07-18:
[image]

pipiche38, 2019-07-24:
Merged on beta eab8091
Temp+Humi sensor works.

closed 2020-12-01 by stale[bot]