zigbeefordomoticz/Domoticz-Zigbee #430: Plugin Web server idea

deennoo, 2019-05-11:
As v4.3 plugin  is base on plugin webserver interface, goal of this Issue is to list what is needed

BornTKill, 2019-05-11:
Admin/Configuration
Backup/Restore
Update FW Zigate/Devices
List/Manage Devices
Include/Exclude Devices
Network Map

pipiche38, 2019-05-12:
Would be good  to have your view on look and field

deennoo, 2019-05-12:
About fonctions : 
- groups management
- scènes management

About look : 
Keep it clear, easy to read.

pipiche38, 2019-05-12:
That is find, but please provide design in order to see from a User point of view what you expect

pipiche38, 2019-05-23:
Just to update the community on where we are going:

- From an architecture standpoint, we have open a REST api inside the plugin. In such the REST API can then expose a number of command to give access Read/Write of internal plugin values

For example: http://host:port/rest-zigate/1/zdevice will give you the list of Devices managed by the plugin

On the Frontend side, we are looking to provide a first release focusing on:

- Plugin settings (been able to access and update the Plugin parameters) instead of editing PluginConf
- Naming each of the Zigate devices. You will have the possibility to provide a name to each equipment.
- Network Topology. A Network Topology diagram will be provided for each report. The name of each device will be used.
- Group Management.  We will be able to Edit/Create/Delete groups
- Zigate live statistics will provide you a monitoring of the Transmission and Reception of Zigbee messages

In the first implementation, some of the features will required some plugin restart in order to take the changes in consideration. For example if you create/update a group, a restart of the plugin will be required

@ben33880

waltervl, 2019-05-28:
Function wish: Have plugin settings that are used to reset or erase something be single use settings. So that the next time you start the plugin for a normal reason the reset or erase action will not be performed again.

pipiche38, 2019-06-04:
> Function wish: Have plugin settings that are used to reset or erase something be single use settings. So that the next time you start the plugin for a normal reason the reset or erase action will not be performed again.

[image]

waltervl, 2019-06-04:
Looking good! What about resetConfigureReporting and resetReadAttributes?

waltervl, 2019-06-04:
And another function wish, but far less important: Firmware update of ZiGate device itself, eg update to 3.1a.
And a check if there is a new Firmware update available.

pipiche38, 2019-06-05:
> Looking good! What about resetConfigureReporting and resetReadAttributes?
Yep, that is the plan too. As well as Permit to Join and other Zigate function
[image]

pipiche38, 2019-06-05:
> And another function wish, but far less important: Firmware update of ZiGate device itself, eg update to 3.1a.
> And a check if there is a new Firmware update available.

Good point, and agree with you that it is not necessarily the top priority.
I think we could easily give an indication if there is a new Firmware (validated with the plugin)

Now as the Firmware update is concerned, I'm a bit reluctant to do it for now, as I think it is more the responsibility of the Zigate (and I'm not sure I want to take that load into the plugin).

In conclusion, lets make a first version of the WebUI featuring the most import needs ;-)

waltervl, 2019-06-05:
> Good point, and agree with you that it is not necessarily the top priority.
> I think we could easily give an indication if there is a new Firmware (validated with the plugin)
>
> Now as the Firmware [...]
Perhaps you can use the python flasher of Doudz: https://github.com/doudz/zigate
It would be nice as I (and I think almost all Domoticz Zigate users) use the Zigate only in Domoticz so to have one User Interface for controlling the ZiGate would be ideal. But indeed, perhaps the python flasher should be created and maintained by ZiGate instead of you or Doudz.

pipiche38, 2019-06-06:
> Perhaps you can use the python flasher of Doudz: https://github.com/doudz/zigate
> It would be nice as I (and I think almost all Domoticz Zigate users) use the Zigate only in Domoticz so to have one [...]

I understand and share somehow that needs. Need further investigations as compatibility with Domoticz Python Framework (which is not the case of Doudz library), compatibility with Windows, is sudo access right needed ....

As soon as the Python Framework will be fixed for the WebServer part, we will take the list of needed and probably priorities them.

Thanks anyway for your valuable input

pipiche38, 2019-07-23:
> 
> 
> Admin/Configuration
> Backup/Restore
> Update FW Zigate/Devices
> List/Manage Devices
> Include/Exclude Devices
> Network Map
> Logging

Could you clarify what you have in mind in Include/Exclude Devices ? 
Based on the latest development ( 4.5 Release candidate ). How do you feel ?

As of today, I do not consider integrating Update FW Zigate as this is quiet link to the Hardware it self and is very platform dependent ( Windows, Linux distribution)

pipiche38, 2019-07-31:
I suggest to close this as 4.5 is out. They are some remaining features like Scene, Bindings which we need to put in the Roadmap

closed 2019-07-31 by pipiche38