zigbeefordomoticz/Domoticz-Zigbee #458: Error when joining router

twoenter, 2019-07-11:
**Describe the bug**
When I try to join a CC2531 zigbee router I have error. With stable plugin no problems

Domoticz log: 
2019-07-11 19:17:13.035 Status: (Zigate) Started.
2019-07-11 19:17:13.402 Status: (Zigate) Entering work loop.
2019-07-11 19:17:13.403 Status: (Zigate) Initialized version beta-4.4, author 'zaraki673 pipiche38'
2019-07-11 19:17:13.410 Status: (Zigate) Zigate plugin beta-4.4.6 started
2019-07-11 19:17:13.411 Status: (Zigate) Python Version - 3.5.1 (default, Jan 29 2018, 14:16:30)
2019-07-11 19:17:13.411 [GCC 4.9.3 20150311 (prerelease)]
2019-07-11 19:17:13.411 Status: (Zigate) Switching Heartbeat to 5 s interval
2019-07-11 19:17:13.411 Status: (Zigate) DomoticzVersion: 4.10717
2019-07-11 19:17:13.411 Status: (Zigate) DomoticzHash: b38b49e-modified
2019-07-11 19:17:13.411 Status: (Zigate) DomoticzBuildTime: 2019-05-09 13:04:08
2019-07-11 19:17:13.411 Status: (Zigate) Opening DomoticzDB in raw
2019-07-11 19:17:13.412 Status: (Zigate) load PluginConf
2019-07-11 19:17:13.414 Status: (Zigate) filename set to /usr/local/domoticz/var/plugins/Zigate/Conf/PluginConf-27.json
2019-07-11 19:17:13.414 Status: (Zigate) pluginReports set to /usr/local/domoticz/var/plugins/Zigate/Reports/
2019-07-11 19:17:13.414 Status: (Zigate) pluginConfig set to /usr/local/domoticz/var/plugins/Zigate/Conf/
2019-07-11 19:17:13.414 Status: (Zigate) pluginOTAFirmware set to /usr/local/domoticz/var/plugins/Zigate/OTAFirmware/
2019-07-11 19:17:13.414 Status: (Zigate) homedirectory set to /usr/local/domoticz/var/plugins/Zigate/
2019-07-11 19:17:13.414 Status: (Zigate) pluginWWW set to /usr/local/domoticz/var/plugins/Zigate/www/
2019-07-11 19:17:13.414 Status: (Zigate) pluginData set to /usr/local/domoticz/var/plugins/Zigate/Data/
2019-07-11 19:17:13.414 Status: (Zigate) pluginHome set to /usr/local/domoticz/var/plugins/Zigate/
2019-07-11 19:17:13.415 Status: (Zigate) enableWebServer set to 1
2019-07-11 19:17:13.415 Status: (Zigate) enableWebServer set to 1
2019-07-11 19:17:13.421 Status: (Zigate) Plugin Database: DeviceList-27.txt
2019-07-11 19:17:13.440 Status: (Zigate) DeviceConf loaded
2019-07-11 19:17:13.440 Status: (Zigate) load ListOfDevice
2019-07-11 19:17:13.502 Status: (Zigate) Not Loading f45d as Status: UNKNOW
2019-07-11 19:17:13.515 Status: (Zigate) Not Loading 39dd as Status: UNKNOW
2019-07-11 19:17:13.533 Status: (Zigate) Entries loaded from /usr/local/domoticz/var/plugins/Zigate/Data/DeviceList-27.txt : 9
2019-07-11 19:17:13.538 Status: (Zigate) Transport mode: USB
2019-07-11 19:17:13.538 Status: (Zigate) Connection Name: Zigate, Transport: Serial, Address: /dev/ttyUSB2
2019-07-11 19:17:13.540 Status: (Zigate) Configure Permit To Join
2019-07-11 19:17:13.541 Status: (Zigate) Request Accepting new Hardware for 254 seconds
2019-07-11 19:17:13.541 Status: (Zigate) Trigger a Topology Scan
2019-07-11 19:17:13.541 Status: (Zigate) Trigger a Energy Level Scan
2019-07-11 19:17:13.644 Status: (Zigate) Major Version Num: 0003
2019-07-11 19:17:13.644 Status: (Zigate) Installer Version Number: 030f
2019-07-11 19:17:13.847 Status: (Zigate) Accepting new Hardware: On
2019-07-11 19:17:13.969 Status: (Zigate) Zigate addresses ieee: 00158d0001ecaee0 , short addr: 0000
2019-07-11 19:17:13.969 Status: (Zigate) Network state UP, PANID: 3b73 extPANID: 0xd56cfb146c18053c Channel: 11
2019-07-11 19:17:14.091 Status: (Zigate) Number of devices recently active in Zigate = 5 
 2019-07-11 19:17:23.409 Status: (Zigate) Zigate set to Certification : CE
2019-07-11 19:17:23.559 Status: (Zigate) Start Web Server connection
2019-07-11 19:17:23.561 Status: (Zigate) Web backend for Web User Interface started on port: 9440
2019-07-11 19:17:23.561 Status: (Zigate) Plugin with Zigate firmware 030f correctly initialized
2019-07-11 19:17:23.559 Error: (Zigate) ATTENTION: the WebServer part is not supported with this version of Domoticz. Please upgrade to a version greater than 4.10901 

2019-07-11 19:28:44.072 (Zigate) Decode004d - rebind clusters for 9909
2019-07-11 19:28:44.064 Status: (Zigate) Device Annoucement ShortAddr: 9909, IEEE: 00124b001936a406
2019-07-11 19:28:44.065 Status: (Zigate) NetworkID : 9909 is replacing e026 and is attached to IEEE : 00124b001936a406
2019-07-11 19:29:20.275 (Zigate) Decode004d - rebind clusters for fa87
2019-07-11 19:29:20.268 Status: (Zigate) Device Annoucement ShortAddr: fa87, IEEE: 00124b001936a406
2019-07-11 19:29:20.268 Status: (Zigate) NetworkID : fa87 is replacing 9909 and is attached to IEEE : 00124b001936a406
2019-07-11 19:29:22.439 Status: (Zigate) Network Energy Level Report: 0000
2019-07-11 19:29:22.439 Status: (Zigate) -----------------------------------------------
2019-07-11 19:29:22.439 Status: (Zigate) router <- nwkid Tx Failure 11 15 19 20 25 26
2019-07-11 19:29:22.439 Error: (Zigate) 'onHeartbeat' failed 'KeyError'.
2019-07-11 19:29:22.439 Error: (Zigate) ----> Line 803 in '/usr/local/domoticz/var/plugins/Zigate/plugin.py', function onHeartbeat
2019-07-11 19:29:22.439 Error: (Zigate) ----> Line 646 in '/usr/local/domoticz/var/plugins/Zigate/plugin.py', function onHeartbeat
2019-07-11 19:29:22.439 Error: (Zigate) ----> Line 529 in '/usr/local/domoticz/var/plugins/Zigate/Modules/heartbeat.py', function processListOfDevices
2019-07-11 19:29:22.439 Error: (Zigate) ----> Line 186 in '/usr/local/domoticz/var/plugins/Zigate/Classes/NetworkEnergy.py', function do_scan
2019-07-11 19:29:22.439 Error: (Zigate) ----> Line 230 in '/usr/local/domoticz/var/plugins/Zigate/Classes/NetworkEnergy.py', function _next_scan
2019-07-11 19:29:22.439 Error: (Zigate) ----> Line 251 in '/usr/local/domoticz/var/plugins/Zigate/Classes/NetworkEnergy.py', function finish_scan
2019-07-11 19:30:12.557 (Zigate) Permit Join status: 00
2019-07-11 19:30:12.557 (Zigate) ---> self.permitTojoin['Starttime']: 1562865957
2019-07-11 19:30:12.557 (Zigate) ---> Current time : 1562866212
2019-07-11 19:30:12.557 (Zigate) ---> self.Ping['Permit'] (prev) : On
2019-07-11 19:30:12.557 (Zigate) ---> self.Ping['Permit'] (new ) : Off 

**Expected behavior**

the plugin stop working. i have to restart and do not add router, the plugin works

**Desktop (please complete the following information):**
 - OS: Synology
 - Browser firefox
 - Version see log

**Plugin (please complete the following information):**
see log

**Additional context**
Add any other context about the problem here.

pipiche38, 2019-07-11:
Could you please attached the file Data/DeviceList-27.txt

Smanar, 2019-07-11:
Ha nice ^^, you are using a CC2531 as router with success ?
There is someone on forum that ask if it's something possible on zigate.

pipiche38, 2019-07-11:
The issue is not related to the beta, it is more related to the fact that you have a device which is doing a lot of device announced.
For instance the device with Mac Address 00124b001936a406

twoenter, 2019-07-12:
> 
> 
> Ha nice ^^, you are using a CC2531 as router with success (at least with stable) ?
> There is someone on forum that ask if it's something possible on zigate.

Yes, I read the post. Want to reply but I want to be sure it is working in future releases.

pipiche38, 2019-07-12:
Again, the issue is not linked to a compatibility topic.

twoenter, 2019-07-13:
This is my DeviceList-27.txt

```
42e9 : {'LogicalType': 'End Device', 'Status': 'inDB', 'ZDeviceID': {}, 'Ep': {'01': {'ClusterType': {'435': 'Temp+Hum+Baro', '437': 'Temp', '436': 'Temp+Hum', '439': 'Baro', '438': 'Humi'}, '0405': 66.5, 'Type': 'Temp/Humi/Baro', '0403': 1016.5, '0402': 21.9, 'ffff': {}, '0003': {}, '0001': '2975;2975;2975;2975', '0000': {}}}, 'Version': '3', 'Attributes List': {}, 'MacCapa': '80', 'Maufacturer Name': '', 'BatteryUpdateTime': 1563006513, 'DeviceType': 'RFD', 'RIA': '0', 'ProfileID': {}, 'Location': 'None', 'Manufacturer Name': 'LUMI', 'Bind': {'0000': {'Status': '', 'Phase': 'requested', 'Stamp': 1561140653}}, 'Heartbeat': '364701', 'ZCL Version': '1', 'RSSI': 75, 'Model': 'lumi.weather', 'NbEp': '', 'Manufacturer': '1037', 'SQN': 'b6', 'Type': 'Temp/Humi/Baro', 'SWBUILD_1': '20161129', 'App Version': '3', 'ReadAttributes': {'TimeStamps': {'01-0000': 1561140653}, 'Ep': {'01': {'0000': {'0007': '00', '0015': '86', '0005': '00', '0004': '00', '0010': '00', '0002': '00', '0003': '00', '0000': '00', 'ff02': '86', 'ff01': '00', '4000': '00', '0001': '00', '0006': '00', '000f': '86', '000a': '86'}}}}, 'Stamp': {'LastSeen': 1563006513, 'Time': '2019-07-13 10:28:33', 'MsgType': '8102'}, 'Battery': 56, 'ReceiveOnIdle': 'Off', 'PowerSource': 'Battery', 'Last Cmds': [], 'ZDeviceName': '', 'HW Version': '30', 'SWBUILD_3': '3000-0001', 'Health': 'Live', 'Stack Version': '2', 'IEEE': '00158d0002c8de2a'}
996a : {'LogicalType': '', 'Status': 'inDB', 'ZDeviceID': {}, 'Ep': {'01': {'ClusterType': {'440': 'Motion', '441': 'Lux'}, '0006': '00', '0400': '109', '0406': '01', '0001': '2995;2995;2995;2995', '0000': {}}}, 'Version': '3', 'Attributes List': {}, 'MacCapa': '80', 'Maufacturer Name': '', 'BatteryUpdateTime': 1563007562, 'DeviceType': '', 'RIA': '0', 'ProfileID': {}, 'Manufacturer Name': 'LUMI', 'Bind': {'0000': {'Status': '', 'Phase': 'requested', 'Stamp': 1561141108}}, 'Heartbeat': '364610', 'ZCL Version': '1', 'RSSI': 96, 'Model': 'lumi.sensor_motion.aq2', 'NbEp': '1', 'Manufacturer': '', 'SQN': '14', 'Type': 'Motion/Lux', 'SWBUILD_1': '20170627', 'App Version': '5', 'ReadAttributes': {'TimeStamps': {'01-0000': 1561141108}, 'Ep': {'01': {'0000': {'ff01': '00', '0005': '00', '0001': '00'}}}}, 'Stamp': {'LastSeen': 1563008349, 'Time': '2019-07-13 10:59:09', 'MsgType': '8102'}, 'Battery': 61, 'ReceiveOnIdle': '', 'PowerSource': '', 'Last Cmds': [], 'ZDeviceName': '', 'HW Version': '1', 'Health': 'Live', 'Stack Version': '2', 'IEEE': '00158d000257e9c6'}
535d : {'LogicalType': 'End Device', 'Status': 'inDB', 'ZDeviceID': {}, 'Ep': {'01': {'ClusterType': {'451': 'Door'}, 'ffff': {}, '0006': '01', 'Type': 'Door', '0003': {}, '0000': {}}}, 'Version': '3', 'Attributes List': {}, 'MacCapa': '80', 'Maufacturer Name': '', 'DeviceType': 'RFD', 'RIA': '0', 'ProfileID': {}, 'Manufacturer Name': 'LUMI', 'Bind': {'0000': {'Status': '', 'Phase': 'requested', 'Stamp': 1562500463}}, 'Heartbeat': '101502', 'ZCL Version': '1', 'RSSI': 36, 'Model': 'lumi.sensor_magnet.aq2', 'NbEp': '', 'Manufacturer': '1037', 'SQN': '28', 'Type': 'Door', 'SWBUILD_1': '20161128', 'App Version': '3', 'ReadAttributes': {'TimeStamps': {'01-0000': 1562500463}, 'Ep': {'01': {'0000': {'ff01': '00', '0006': '00', '0005': '00', '0004': '00', '0003': '00', '0002': '00', '0001': '00', '0000': '00'}}}}, 'Stamp': {'LastSeen': 1562614724, 'Time': '2019-07-08 21:38:44', 'MsgType': '8102'}, 'Battery': 76, 'ReceiveOnIdle': 'Off', 'PowerSource': 'Battery', 'Last Cmds': [], 'ZDeviceName': '', 'HW Version': '2', 'Health': '', 'Stack Version': '2', 'IEEE': '00158d000273bb11'}
0000 : {'version': '3', 'PowerSource': 'Main', 'Ep': {'01': {'0006': {}, '0004': {}, '0008': {}}}, 'IEEE': '00158d0001ecaee0'}
591b : {'LogicalType': 'End Device', 'Status': 'inDB', 'ZDeviceID': {}, 'Ep': {'01': {'ClusterType': {'449': 'Motion', '450': 'Lux'}, '0006': '00', '0400': '63', '0406': '01', '0001': '3035;3035;3035;3035', '0000': {}}}, 'Version': '3', 'Attributes List': {}, 'MacCapa': '80', 'Maufacturer Name': '', 'BatteryUpdateTime': 1563008640, 'DeviceType': 'RFD', 'RIA': '0', 'ProfileID': {}, 'Manufacturer Name': 'LUMI', 'Bind': {'0000': {'Status': '', 'Phase': 'requested', 'Stamp': 1561141785}}, 'Heartbeat': '64', 'ZCL Version': '1', 'RSSI': 153, 'Model': 'lumi.sensor_motion.aq2', 'NbEp': '', 'Manufacturer': '1037', 'SQN': '48', 'Type': 'Motion/Lux', 'SWBUILD_1': '20170627', 'App Version': '5', 'ReadAttributes': {'TimeStamps': {'01-0000': 1561141785}, 'Ep': {'01': {'0000': {'ff01': '00', '0006': '00', '0005': '00', '0004': '00', '0003': '00', '0002': '00', '0001': '00', '0000': '00'}}}}, 'Stamp': {'LastSeen': 1563008343, 'Time': '2019-07-13 11:04:00', 'MsgType': '8102'}, 'Battery': 71, 'ReceiveOnIdle': 'Off', 'PowerSource': 'Battery', 'Last Cmds': [(1562962324.3622081, '0092'), (1562966234.6217964, '0092'), (1562994027.4425535, '0092'), (1563008151.0924904, '0092'), (1563008343.960848, '0092')], 'ZDeviceName': '', 'HW Version': '1', 'Health': 'Live', 'Stack Version': '2', 'IEEE': '00158d0002b765aa'}
fa87 : {'LogicalType': 'Router', 'Status': 'inDB', 'ZDeviceID': '0100', 'Ep': {'08': {'0006': '00', 'ClusterType': {'452': 'Switch'}, '0000': {}}}, 'Version': '3', 'Attributes List': {'Ep': {'08': {'0006': {'0000': '10'}, '0000': {'0010': '42', '0007': '20', '0006': '42', '0005': '42', '0004': '42', '0003': '20', '0011': '20', '0012': '10'}}}}, 'MacCapa': '8e', 'Maufacturer Name': '', 'DeviceType': 'FFD', 'RIA': '3', 'ProfileID': '0104', 'Location': 'N [... cut ...] er': '1037', 'SQN': 'ff', 'Type': 'Temp/Humi/Baro', 'SWBUILD_1': '20161129', 'App Version': '3', 'ReadAttributes': {'TimeStamps': {'01-0000': 1561215454}, 'Ep': {'01': {'0000': {'ff01': '00'}}}}, 'Stamp': {'LastSeen': 1563007661, 'Time': '2019-07-13 10:47:41', 'MsgType': '8102'}, 'Battery': 59, 'ReceiveOnIdle': 'Off', 'PowerSource': 'Battery', 'Last Cmds': [], 'ZDeviceName': '', 'HW Version': '30', 'SWBUILD_3': '3000-0001', 'Health': 'Live', 'Stack Version': '2', 'IEEE': '00158d0002ca05a6'}

```

pipiche38, 2019-07-13:
This is not related to the router topic. it is more related to the fact that the device is not responding to EndPoint descriptions.
Most likely it did work once when you were on stable branch and now it doesn't.

closed 2019-07-13 by pipiche38