zigbeefordomoticz/Domoticz-Zigbee #459: Prepare Release 4.5

pipiche38, 2019-07-15:
- Web Admin Interface
- new certified devices
- compatible with Firmware 3.1a
- Online debuging mode
- Still backward compatibility with non-web Admin ( we keep key features in the plugin admin menu)

Based on beta 4.4

pipiche38, 2019-07-15:
#460

cdorbell2, 2019-07-20:
Hi
for testing your new candidate release we use your branch rc-4.5 ?
because i seen this error in my log : 

2019-07-20 19:48:08.031  Status: (zigate) DomoticzVersion: 4.11020
[...]
2019-07-20 19:48:18.059  Error: (zigate) Firmware 031a is not yet supported
2019-07-20 19:48:18.059  Status: (zigate) Plugin with Zigate firmware 031a correctly initialized

pipiche38, 2019-07-22:
Commit: 5a56696847ffce3a04690ce4a7b80f888e2cabb5

pipiche38, 2019-07-31:
cb8b61f9b3155bf2206055754d9c658894a9f0ac

closed 2019-07-31 by pipiche38