zigbeefordomoticz/Domoticz-Zigbee #467: ikea e1743 on off switch and long press

ganaux, 2019-08-21:
Hi,
I'd like to add a request for this switch (ikea tradfri e1743).
on and off command work well for this switch (domoticz 4.11.215 and zigate plugin 4.5.3).

The switch can send command for long press too but no command are send to domoticz.

in domoticz log with plugin debug i can see :

for up button long press and release :
2019-08-21 18:05:35.584 (zigate) Decode8085 - SQN: 15, Addr: 6fcc, Ep: 01, Cluster: 0008, Cmd: 05, Unknown: 02
2019-08-21 18:05:35.935 (zigate) Decode8085 - SQN: 16, Addr: 6fcc, Ep: 01, Cluster: 0008, Cmd: 07, Unknown: 02

for down button long press and release :
2019-08-21 18:07:43.418 (zigate) Decode8085 - SQN: 1b, Addr: 6fcc, Ep: 01, Cluster: 0008, Cmd: 01, Unknown: 02
2019-08-21 18:07:44.876 (zigate) Decode8085 - SQN: 1c, Addr: 6fcc, Ep: 01, Cluster: 0008, Cmd: 07, Unknown: 02

(the "release button" event is the same for up and down button).

So the plugin receive the long press from zigate but doesn't send any event to domoticz object.

Is there any way to send this events (long press) to domoticz object ?

pipiche38, 2019-08-22:
This could be possible.
It will required to use a Switch Selector and not a simple switch in Domoticz.

What do you suggest as the different Selectors ?

ganaux, 2019-08-22:
I suggest this : 
On
Off
Up_push
Down_push
UpOrDown_release (one selector exists for that ?)

I think we have to use "On" and "Off" for compatibility with simple light.
For long press "Up_push" and "Down_push" seems Ok but as there is one only command for release Up ou Down, i don't know what is the appropriate selector ( Off, ToggleOnOff).

pipiche38, 2019-08-22:
Ok. SwitchIKEA

00: Off
10: On
20:: Push Up
30: Push Down
40: Release

I'll make the change on beta branch. You'll have to remove the widget from Domoticz and repair the switch again to get the new behaviour.

Will let you know when the changes will be pushed on beta

pipiche38, 2019-08-22:
Integrated into beta branche 3c3b9ee

pipiche38, 2019-08-22:
The changes have been implemented and should work as expected.
It is backward compatible

Of course, you are aware the Long Press Up/Down are for Level Control.

ganaux, 2019-08-23:
I have tested. The plugin do exactly what i need now. I can use it for example to control blinds that are not zigbee (long press and release for partial opening, click for full opening).

I need to have domoticz object with the 5 actions. The name of each action could be changed for final version if needed for other application.

this remote is very low cost (5 euros more when bought with tradfri plug) but is very difficult to associate when the coordinator is zigate (touchlink needed to create group between remote and plug).

Thank you for this change, you're very reactive.

pipiche38, 2019-08-23:
With pleasure and I'm sure this will contribute to others, so feel free to advertise on the various Forum.

commit 3c3b9ee0 (2019-08-22): Fix #467
closed 2019-08-23 by pipiche38