zigbeefordomoticz/Domoticz-Zigbee #481: Tuya Zigbee Temp/Hum Sensor with Screen

deennoo, 2019-11-04:
Just get this device from Tuya : https://fr.aliexpress.com/item/4000277514396.html?spm=a2g0s.9042311.0.0.6e6f6c37U7yWC2

all is ok this PiZigate : 

`2019-11-04 17:16:12.590 Status: (Pi Zigate) Device Announced ShortAddr: 0039, IEEE: ec1bbdfffe783d07
2019-11-04 17:16:12.845 Status: (Pi Zigate) [-] NEW OBJECT: 0039 Request Simple Descriptor for Ep: 01
2019-11-04 17:16:13.000 (Pi Zigate) Decode8043 - Overwrite ProfileID {} with 0104 from Ep: 01
2019-11-04 17:16:13.000 (Pi Zigate) Decode8043 - Overwrite ZDeviceID {} with 0302 from Ep: 01
2019-11-04 17:16:13.000 Status: (Pi Zigate) [-] NEW OBJECT: 0039 Simple Descriptor EP 01
2019-11-04 17:16:13.000 Status: (Pi Zigate) [-] NEW OBJECT: 0039 ProfileID 0104
2019-11-04 17:16:13.000 Status: (Pi Zigate) [-] NEW OBJECT: 0039 ZDeviceID 0302
2019-11-04 17:16:13.001 Status: (Pi Zigate) [-] NEW OBJECT: 0039 Cluster IN Count: 04
2019-11-04 17:16:13.001 Status: (Pi Zigate) [-] NEW OBJECT: 0039 Cluster In 1: 0000 (Basic)
2019-11-04 17:16:13.001 Status: (Pi Zigate) [-] NEW OBJECT: 0039 Cluster In 2: 0001 (Power Configuration)
2019-11-04 17:16:13.001 Status: (Pi Zigate) [-] NEW OBJECT: 0039 Cluster In 3: 0402 (Temperature Measurement)
2019-11-04 17:16:13.001 Status: (Pi Zigate) [-] NEW OBJECT: 0039 Cluster In 4: 0405 (Relative Humidity Measurement)
2019-11-04 17:16:13.001 Status: (Pi Zigate) [-] NEW OBJECT: 0039 Cluster OUT Count: 01
2019-11-04 17:16:13.001 Status: (Pi Zigate) [-] NEW OBJECT: 0039 Cluster Out 1: 0019 (Over-the-Air Upgrade)
2019-11-04 17:16:20.068 Status: (Pi Zigate) [1] NEW OBJECT: 0039 Model Name: TS0201
2019-11-04 17:16:20.069 Status: (Pi Zigate) [1] NEW OBJECT: 0039 Model Name: TS0201
2019-11-04 17:16:25.030 (Pi Zigate) GetType - Model TS0201 Profile ID : 0104 ZDeviceID : 0302
2019-11-04 17:16:25.030 (Pi Zigate) GetType - Model TS0201 not found with Ep: 01 in DeviceConf. Continue with ClusterSearch
2019-11-04 17:16:25.031 (Pi Zigate) GetType - ClusterSearch return : Temp/Humi/Voltage
2019-11-04 17:16:25.029 Status: (Pi Zigate) [2] NEW OBJECT: 0039 Model Name: TS0201
2019-11-04 17:16:25.030 Status: (Pi Zigate) [2] NEW OBJECT: 0039 Model Name: TS0201
2019-11-04 17:16:59.689 Status: EventSystem: reset all device statuses...
2019-11-04 17:17:04.286 Status: EventSystem: reset all device statuses...
2019-11-04 17:17:07.025 Status: EventSystem: reset all device statuses...
2019-11-04 17:17:12.250 Status: EventSystem: reset all device statuses...
2019-11-04 17:17:17.711 (Pi Zigate) UpdateDevice - (Pi Zigate - TS0201_Temp+Hum-ec1bbdfffe783d07-01) 0:22.2;50;1
2019-11-04 17:17:17.746 (Pi Zigate) UpdateDevice - (Pi Zigate - TS0201_Temp-ec1bbdfffe783d07-01) 22.2:22.2
2019-11-04 17:17:18.153 (Pi Zigate) UpdateDevice - (Pi Zigate - TS0201_Temp+Hum-ec1bbdfffe783d07-01) 0:22.2;58.5;1
2019-11-04 17:17:18.168 (Pi Zigate) UpdateDevice - (Pi Zigate - TS0201_Humi-ec1bbdfffe783d07-01) 58.5:1
2019-11-04 17:17:37.703 (Pi Zigate) UpdateDevice - (Pi Zigate - TS0201_Temp+Hum-ec1bbdfffe783d07-01) 0:23.5;58.5;1
2019-11-04 17:17:37.734 (Pi Zigate) UpdateDevice - (Pi Zigate - TS0201_Temp-ec1bbdfffe783d07-01) 23.5:23.5
2019-11-04 17:17:38.237 (Pi Zigate) UpdateDevice - (Pi Zigate - TS0201_Temp+Hum-ec1bbdfffe783d07-01) 0:23.5;82.0;3
2019-11-04 17:17:38.250 (Pi Zigate) UpdateDevice - (Pi Zigate - TS0201_Humi-ec1bbdfffe783d07-01) 82.0:3`

Exept Voltage who give 0v

pipiche38, 2019-11-04:
Thanks, will fix it

deennoo, 2019-11-07:
ok just update and device creation give only 3 devices : temp+hum / temp / hum

perfect i close it

closed 2019-11-07 by deennoo