zigbeefordomoticz/Domoticz-Zigbee #482: Improve Group processing

pipiche38, 2019-11-06:
When scaning for Group, don't scan devices which are not in the ZigateGroupList

Gemies, 2019-11-06:
Maybe you can distinguish two seperate group scans. One who does the full group scan. And the other variant scanning  for new groups and adding only new groups too the devicelist without touching the existing groups. That would be perfect in the webui.

pipiche38, 2019-11-06:
@Gemies Thanks for the suggestion, here is where I am:

1. Group was never scan before ( eg there is no GroupList cache file), we do a full scan
2. Via the Web Interface, or manualy there is a new or updated Group (eg. ZigateGroupConfig has been updated), then we will only scan a limited number of devices.

In the case of 2. the easy/fast step is to only target the devices which are in the Group definition ( ZigateGroupConfig).
This is most-likely where I will got

Then , the next step, is to refactor the Group Creation/Update/Delete by allowing direct group creation/update/removal without this batch mode.

Gemies, 2019-11-06:
Next step sounds promising 👍

pipiche38, 2019-11-07:
Step1:  e615cb5

pipiche38, 2020-05-26:
#610

closed 2020-05-26 by pipiche38