zigbeefordomoticz/Domoticz-Zigbee #483: Tuya Zigbee Line Only (model : TS0011) reconize as Batterie Device

deennoo, 2019-12-14:
Running : 
Zigate fw : 031a
Plugin Beta : 4.6.036
Domoticz : Version: 4.11334

I get some Tuya 1 Gang Switch running Zigbee 3.0, they use Line Only they are reconize as Battery Device.

Surely they don't mesh our network, but surely they use Line to be powered.

GSzabados, 2020-01-07:
@deennoo a user on the Smartthings community posted a picture and ST recognized the device as SLEEPY_END_DEVICE. It is probably due to the missing Power source defined incorrectly in the basic cluster.
By the way, is the standby LED at the button is working on your device?

deennoo, 2020-01-08:
Clearly not

GSzabados, 2020-01-08:
@deennoo, thanks! It is might be due to the fact, that it is a Live only device. They can have a capacitor to hold the power when the light is off, or has a kind of parasitic power use. This way, they would not operate as a repeater. Maybe that is the whole reason behind it.
The user on the ST community forum was interested about the backlight of the button, as it is not on, when the light is turned off. He thought that should be on as a stand by light or as a night light. (That's how it was advertised?) Have you got that working on yours?

deennoo, 2020-01-18:
night light isn't working pair to a zigate.

I'll try to pair one to the tuya gateway

deennoo, 2020-01-19:
> Have you got that working on yours?

I Just test device call x713 (TS0013) : 3 Gangs line only : Night light doesn't work with Tuya Zigbee Gateway

deennoo, 2020-01-19:
>  Have you got that working on yours?

I have a test on x713 device 3 gang with line only, Reconize on Zigate as TS0013.
I Pair it on Tuya Zigbee Gateway : nightlight doesn't work.

deennoo, 2020-01-19:
@pipiche38 

there is my Export from Devices List RAW (json)

Here we are talking about device reconize as : 
 - TS0011 = 1 Gang Line Only
 - TS0012  = 2 Gang Line Only
 - 
[export.zip](https://github.com/pipiche38/Domoticz-Zigate/files/4082948/export.zip)
TS0013  = 3 Gang Line Only

deennoo, 2020-01-19:
on pairing i'm facing : 

2020-01-19 17:57:29.282 (PiZigate Wifi) ReadCluster - 0000 d985/01 attribute Zemismart - ffe2: 0x13 19
2020-01-19 17:57:29.283 (PiZigate Wifi) readCluster - 0000 - d985/01 unknown attribute: ffe1 48 0000
2020-01-19 17:57:29.334 (PiZigate Wifi) readCluster - 0000 - d985/01 unknown attribute: fffe 30 0001 00
2020-01-19 17:57:29.334 (PiZigate Wifi) readCluster - 0000 - d985/01 unknown attribute: ffe0 48 0000

pipiche38, 2020-01-19:
6ec0a76

closed 2020-01-28 by pipiche38