zigbeefordomoticz/Domoticz-Zigbee #486: Loosing devices in plugin database

ZeGuigui, 2020-01-02:
After domoticz restart, I loose some (or all) of my paired devices and I have to associate them again (deleting domoticz devices before hand).

In the logs here is what I can find:

```
2020-01-02 11:54:30.764 Status: (Zigate) [01] DevID = 01 Network addr = 48d0 IEEE = 00158d0002c14c86 LQI = 057 Power = 00 not found in ListOfDevices
2020-01-02 11:54:30.764 Status: (Zigate) [02] DevID = 02 Network addr = 9332 IEEE = 00158d0002c1472d LQI = 140 Power = 00 not found in ListOfDevices
 2020-01-02 11:54:30.010 Error: (Zigate) checkListOfDevice2Devices - Salle à manger Temp+Hum - 00158d0002c1472d not found in Plugin Database
 2020-01-02 11:55:44.790 Error: (Zigate) Decode8102 - Receiving a message from unknown device : 9332 with Data : bb93320104020000002900020716 
```

Any idea how to backup / check if database is correctly writen? I tryed to adjust unix permissions on files but I have not idea if I changed the right files!

I am using domoticz beta on a synology NAS:
```
Version: 4.11563
Build Hash: d5e8a14bf-modified
Compile Date: 2019-12-09 20:21:24
dzVents Version: 2.5.2
```

Plugin is up-to-date with github

pipiche38, 2020-01-02:
Better to use the forum. The GitHub issue is for bug reporting

ZeGuigui, 2020-01-02:
Well it looked like a bug to me! I will post in the forum

pipiche38, 2020-01-02:
Thanks for posting it to the forum , which is the best place.

closed 2020-01-02 by pipiche38