zigbeefordomoticz/Domoticz-Zigbee #561: Aqara switch not working with 4.8.18

egrueda, 2020-05-04:
Using Domoticz 2020.2 (build 12017) and Aqara Single Wall Switch
Was workinf fine during months
But since last pugin update, it stopped working

Other aqara devices work, only affects to the single wall switch

Failing scenario:
Domoticz 2020.2
Plugin: 4.8.018
Firmware: 031c
Device found and connected
Not receiving any communication

Back to stable:
Domoticz 2020.2
Plugin version: 4.7.010
Firmware: 031c
Device found and connected
Communication received by domoticz

tl;dr: with plugin 4.8.018 I get no communicatons from aqara single wall switch but other zigbee devices do keep working. When I downgrade to stable, without any other change, ir works again and get dada in the logs.

(Not a problem for me, just to let you know)

pipiche38, 2020-05-05:
Please use the forum and get to the latest version

closed 2020-05-05 by pipiche38

egrueda, 2020-05-05:
I already got it working with the latest version.
Just wanted to advise you there is a BUG with version 4.8.018
Regards

pipiche38, 2020-05-05:
4.8  is beta and is in continuous development.

egrueda, 2020-05-05:
I know, I just thought you would like to receive some feedback on the beta version.
Never mind, bet regards

pipiche38, 2020-05-05:
Feedbacks are good. Thanks for that. Just use the forum (and usually I'm active on it), it is easier and you get a broader attention ;-)

egrueda, 2020-05-05:
Ok! I will report in the forum. Thanks!