zigbeefordomoticz/Domoticz-Zigbee #683: Error on Windows Version 4.9.004

geertvercamer, 2020-06-29:
2020-06-29 17:16:55.191 Error: (Zigate) 'onMessage' failed 'PermissionError'.
2020-06-29 17:16:55.192 Error: (Zigate) ----> Line 984 in 'C:\Program Files (x86)\Domoticz\plugins\Domoticz-Zigate\plugin.py'
2020-06-29 17:16:55.192 Error: (Zigate) ----> Line 519 in 'C:\Program Files (x86)\Domoticz\plugins\Domoticz-Zigate\plugin.py'
2020-06-29 17:16:55.192 Error: (Zigate) ----> Line 325 in 'C:\Program Files (x86)\Domoticz\plugins\Domoticz-Zigate\Classes\Transport.py'
2020-06-29 17:16:55.192 Error: (Zigate) ----> Line 542 in 'C:\Program Files (x86)\Domoticz\plugins\Domoticz-Zigate\Classes\Transport.py'
2020-06-29 17:16:55.192 Error: (Zigate) ----> Line 523 in 'C:\Program Files (x86)\Domoticz\plugins\Domoticz-Zigate\plugin.py'
2020-06-29 17:16:55.192 Error: (Zigate) ----> Line 122 in 'C:\Program Files (x86)\Domoticz\plugins\Domoticz-Zigate\Modules\input.py'
2020-06-29 17:16:55.192 Error: (Zigate) ----> Line 547 in 'C:\Program Files (x86)\Domoticz\plugins\Domoticz-Zigate\Modules\input.py'3

Tried to change files location to D:\DomoticzExtra\ZiGate\.......

no change.

**Desktop (please complete the following information):**
 - OS: W10
 - Chrome

**Plugin (please complete the following information):**
 - OS: W10
 - Plugin version 4.9.004
 - Plugin branch: master

pipiche38, 2020-06-29:
Please do use the forums for that.
Looks like you have Access Right issues

Forum en Français https://easydomoticz.com/forum/viewforum.php?f=28
Forum in French https://www.domoticz.com/forum/viewforum.php?f=68

pipiche38, 2020-06-29:
The error is happening while trying to create/update a file named Conf/PluginConf-xx.json where xx is the hardware Id of the plugin instance in Domoticz

The plugin by default create files under its directory tree. Basically

Domoticz-Zigate is the home directory of the plugin
Domoticz-Zigate/Conf  is where configuration file are store and can be updated via the Web Interface
Domoticz-Zigate/Data is where plugin Database is store

pipiche38, 2020-06-29:
Please do start Domoticz with Administrator mode as shown in this picture

[image]

geertvercamer, 2020-06-29:
It is running as a service...

However, I've changed the config file to
{
  "channel": "11",
  "filename": "D:\\DomoticzExtra\\Zigate\\Conf\\PluginConf-19.json",
  "homedirectory": "C:\\Program Files
(x86)\\Domoticz\\plugins\\Domoticz-Zigate\\",
  "pluginConfig": "D:\\DomoticzExtra\\Zigate\\Conf\\",
  "pluginData": "D:\\DomoticzExtra\\Zigate\\Data\\",
  "pluginHome": "C:\\Program Files
(x86)\\Domoticz\\plugins\\Domoticz-Zigate\\",
  "pluginLogs": "D:\\DomoticzExtra\\Zigate\\Logs\\",
  "pluginOTAFirmware": "D:\\DomoticzExtra\\Zigate\\OTAFirmware\\",
  "pluginReports": "D:\\DomoticzExtra\\Zigate\\Reports\\",
  "pluginWWW": "C:\\Program Files
(x86)\\Domoticz\\plugins\\Domoticz-Zigate\\www\\"
}

and it looks ok. No more errors as far as I can tell.

Maybe something to mention somewhere in the documentation?

Thanks for your help!

Met vriendelijke groet,

Geert

pipiche38, 2020-06-30:
Thanks, for sharing this tips. 
But I guess you have also move the Domotciz DB outside of the standard path.
So installing plugins on a non-system-admin folder could be a better way.

closed 2020-06-30 by pipiche38

geertvercamer, 2020-06-30:
no, I haven't moved the Domoticz DB. I wasn't aware of that possibility.
Installing in C:\Domoticz instead of C:\Program Files\Domoticz is probably
a better option.

Thanks for all the help anyway.
Met vriendelijke groet,

Geert

pipiche38, 2020-06-30:
@geertvercamer  would appreciate your help in documenting what you did for the Windows setup. I do agree that it will help Windows users to handle those access rights issues
https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Plugin-windows.md

geertvercamer, 2020-07-01:
@Pipiche38, I guess your page is rather obvious. (there is a 'n' too much in the title though)
 
I made a new folder, outside of the 'Program Files' structure. I copied all the folders from the original location. Then edited the configuration file in order to change the path. The new settings were saved both in the original location and in the new folder. (won't work if you don't save it in the new location, it seems very strange to have a config file in the original location pointing to the file that will be used in the new location).
No idea if the homedirectory and/or the pluginHome path should change too, up until now it works fine for me without any error message