zigbeefordomoticz/Domoticz-Zigbee #703: Failed to load plugin.py at Synology NAS

twoenter, 2020-10-01:
After upgrade firmware from 30f to 31c and a git pull command for the plugin I have this error

2020-10-01 10:55:45.780 Error: (Zigate) failed to load 'plugin.py', Python Path used was '/usr/local/domoticz/var/plugins/Zigate/:/volume1/@appstore/py3k/usr/local/lib/python35.zip:/volume1/@appstore/py3k/usr/local/lib/python3.5/:/volume1/@appstore/py3k/usr/local/lib/python3.5/plat-linux:/volume1/@appstore/py3k/usr/local/lib/python3.5/lib-dynload:/volume1/@appstore/py3k/usr/local/lib/python3.5/site-packages'. 

2020-10-01 10:55:45.780 Error: (Zigate) Module Import failed, exception: 'SyntaxError' 

2020-10-01 10:55:45.780 Error: (Zigate) Import detail: File: /usr/local/domoticz/var/plugins/Zigate/Modules/logging.py, Line: 130, offset: 42 

2020-10-01 10:55:45.780 Error: (Zigate) Error Line ' self.LogErrorHistory[module]['4']: loggingBuildContext(self, module, message, nwkid, context)

I switched to stable and plugin is working with 31c.

From all routers i get NACK errors.

twoenter, 2020-10-01:
NACK errors are solved by turning on and off one of the ikea bulbs. Cut power off.

pipiche38, 2020-10-05:
Not sure what it is, but please use the Forums first to qualify if the issue is relevant to a bug in the plugin or not.
Thanks

closed 2020-10-05 by pipiche38