zigbeefordomoticz/Domoticz-Zigbee #716: Web UI is not started if the ZiGate is not running !

pipiche38, 2020-10-12:
The web UI is not started when the ZiGate has issues.

waltervl, 2020-12-02:
I had this issue too! At least some months ago, now it is working fine. But for the future is it possible to use Monit (https://mmonit.com/monit/) to monitor port 9440. If no reaction then reboot Domoticz server.
But how to monitor webadmin?
Domotic itself is monitored with:
```
`url http://127.0.0.1:8080/json.htm?type=command&param=getversion
         and content = '"status" : "OK"`
```
See alse https://www.domoticz.com/wiki/Monitoring_domoticz

Is there a common way to do this for the ZiGate WebAdmin?
I see for example http://127.0.0.1:9440/manifest.json
That will show the json with content "name": "ZigatePlugin" that can be checked.
But is this manifest.json always available in the next releases of the WebAdmin?

pipiche38, 2020-12-02:
Very good suggestion. Will propose a get version API

pipiche38, 2020-12-13:
> /manifest.json

@waltervl the /manifest.json is related to AngularJS that we use. I don't know if this will stay for ever.
What you can do is to select one of the plugin REST API which are described here

https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Technical/REST-API.md

waltervl, 2020-12-13:
Thank you.
I have setup a monitoring now with the following lines. It will check the plugin url and when it fails 2 times in a row it will restart domoticz.

```
  if failed
     url http://127.0.0.1:9440/rest-zigate/1/plugin
         and content = '"Name": "ZiGate"'
     for 2 cycles
     then restart

```

pipiche38, 2020-12-14:
Just make sure that when you start Domoticz, the Web UI is not there immediately

FWES70, 2020-12-19:
I have the same issue here. runnig Domoticz (v2020.2.12731-8 from Jadahl) on Synology (DS214SE). as soon as I make the hardware, Domoticz webUI blocks.  By removing the plugin it runs fine again.
I also don't know how to enable the permit to join, should this be available in Domoticz when creating the hardware?

PS: I also have an RFX transceiver in the other USB port. could that be the problem?

pipiche38, 2020-12-19:
There is a known issue on Syno. Can you try to edit the file Domoticz-Zigate/Conf/PluginConf-xx.json
and add the following line

`"internetAccess": 0,`

the best is to add it in the middle of the file, so you don't break the all file syntax

FWES70, 2020-12-19:
thanks a lot, that did it. I'll use the forum next time - sorry.

pipiche38, 2021-08-17:
31e8fd56b3250b2ec39c39047839dc3a1ad2da0c

closed 2021-08-17 by pipiche38