zigbeefordomoticz/Domoticz-Zigbee #724: Support Samotech Zigbee Interrupteurs et variateurs

pipiche38, 2020-10-21:
Based on the donation received I have ordered 1 simple item
Tuya ZigBee 3.0 Smart Light Switch Module

https://www.moeshouse.com/search?type=product%2Carticle%2Cpage&q=Tuya+ZigBee+3.0+Smart+Light+Switch+Module

pipiche38, 2020-10-21:
Wiating for delivery

jfcjfc, 2020-10-21:
pipiche38 : As dry contact, L1 and L-out are the relay,  L and N are the 220v power  ??? Test after the holidays, it's the heating, I want to be there.

pipiche38, 2020-10-31:
Device received.
Tuya Brand
Model Name TS0121 (as a Smart Plug from the same brand)
There are Cluster 0b04 and 0702 but seems not to be used !

closed 2020-11-03 by pipiche38

pdelrot-samea, 2020-12-22:
Bonjour / Hello. I have the same module here. Can't réinitialize it too. Following the user manual, on/off sequence 5 times. I tried short (1s or less) and long (5s) sequences. Device doesn't beep as the first power on, and not detected.
Did you managed to reinitialize the network and how ?
Thanks, Pascal

pipiche38, 2020-12-22:
@pdelrot-samea 

> Bonjour / Hello. I have the same module here. Can't réinitialize it too. Following the user manual, on/off sequence 5 times. I tried short (1s or less) and long (5s) sequences. Device doesn't beep [...]

Please use the forum to get support, not the github which is dedicated to Bug !
Here is the link I get from the vendor: https://we.tl/t-uUgiOAObOo