zigbeefordomoticz/Domoticz-Zigbee #728: Error: (Zigate) 'onMessage' failed 'error':'argument out of range'.

Miloune, 2020-10-24:
Hello,

Since lot of time, I have some errors from Zigate plugin in my domoticz.
```
2020-10-24 13:53:13.316 Error: (Zigate) 'onMessage' failed 'error':'argument out of range'.
2020-10-24 13:53:13.316 Error: (Zigate) ----> Line 1007 in '/home/pi/domoticz/plugins/domoticz-zigate/plugin.py', function onMessage
2020-10-24 13:53:13.316 Error: (Zigate) ----> Line 517 in '/home/pi/domoticz/plugins/domoticz-zigate/plugin.py', function onMessage
2020-10-24 13:53:13.317 Error: (Zigate) ----> Line 548 in '/home/pi/domoticz/plugins/domoticz-zigate/Classes/Transport.py', function on_message
2020-10-24 13:53:13.317 Error: (Zigate) ----> Line 1163 in '/home/pi/domoticz/plugins/domoticz-zigate/Classes/Transport.py', function process_frame
2020-10-24 13:53:13.317 Error: (Zigate) ----> Line 523 in '/home/pi/domoticz/plugins/domoticz-zigate/plugin.py', function processFrame
2020-10-24 13:53:13.317 Error: (Zigate) ----> Line 229 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/input.py', function ZigateRead
2020-10-24 13:53:13.317 Error: (Zigate) ----> Line 2818 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/input.py', function Decode8102
2020-10-24 13:53:13.317 Error: (Zigate) ----> Line 2925 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/input.py', function read_report_attributes
2020-10-24 13:53:13.317 Error: (Zigate) ----> Line 217 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/readClusters.py', function ReadCluster
2020-10-24 13:53:13.317 Error: (Zigate) ----> Line 511 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/readClusters.py', function Cluster0000
2020-10-24 13:53:13.317 Error: (Zigate) ----> Line 462 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/lumi.py', function readXiaomiCluster
2020-10-24 14:51:55.826 Error: (Zigate) 'onMessage' failed 'error':'argument out of range'.
2020-10-24 14:51:55.826 Error: (Zigate) ----> Line 1007 in '/home/pi/domoticz/plugins/domoticz-zigate/plugin.py', function onMessage
2020-10-24 14:51:55.826 Error: (Zigate) ----> Line 517 in '/home/pi/domoticz/plugins/domoticz-zigate/plugin.py', function onMessage
2020-10-24 14:51:55.826 Error: (Zigate) ----> Line 548 in '/home/pi/domoticz/plugins/domoticz-zigate/Classes/Transport.py', function on_message
2020-10-24 14:51:55.827 Error: (Zigate) ----> Line 1163 in '/home/pi/domoticz/plugins/domoticz-zigate/Classes/Transport.py', function process_frame
2020-10-24 14:51:55.827 Error: (Zigate) ----> Line 523 in '/home/pi/domoticz/plugins/domoticz-zigate/plugin.py', function processFrame
2020-10-24 14:51:55.827 Error: (Zigate) ----> Line 229 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/input.py', function ZigateRead
2020-10-24 14:51:55.827 Error: (Zigate) ----> Line 2818 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/input.py', function Decode8102
2020-10-24 14:51:55.827 Error: (Zigate) ----> Line 2925 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/input.py', function read_report_attributes
2020-10-24 14:51:55.827 Error: (Zigate) ----> Line 217 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/readClusters.py', function ReadCluster
2020-10-24 14:51:55.827 Error: (Zigate) ----> Line 511 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/readClusters.py', function Cluster0000
2020-10-24 14:51:55.827 Error: (Zigate) ----> Line 462 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/lumi.py', function readXiaomiCluster
2020-10-24 15:42:37.033 Error: (Zigate) 'onMessage' failed 'error':'argument out of range'.
2020-10-24 15:42:37.033 Error: (Zigate) ----> Line 1007 in '/home/pi/domoticz/plugins/domoticz-zigate/plugin.py', function onMessage
2020-10-24 15:42:37.033 Error: (Zigate) ----> Line 517 in '/home/pi/domoticz/plugins/domoticz-zigate/plugin.py', function onMessage
2020-10-24 15:42:37.033 Error: (Zigate) ----> Line 548 in '/home/pi/domoticz/plugins/domoticz-zigate/Classes/Transport.py', function on_message
2020-10-24 15:42:37.033 Error: (Zigate) ----> Line 1163 in '/home/pi/domoticz/plugins/domoticz-zigate/Classes/Transport.py', function process_frame
2020-10-24 15:42:37.033 Error: (Zigate) ----> Line 523 in '/home/pi/domoticz/plugins/domoticz-zigate/plugin.py', function processFrame
2020-10-24 15:42:37.033 Error: (Zigate) ----> Line 229 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/input.py', function ZigateRead
2020-10-24 15:42:37.033 Error: (Zigate) ----> Line 2818 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/input.py', function Decode8102
2020-10-24 15:42:37.033 Error: (Zigate) ----> Line 2925 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/input.py', function read_report_attributes
2020-10-24 15:42:37.033 Error: (Zigate) ----> Line 217 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/readClusters.py', function ReadCluster
2020-10-24 15:42:37.033 Error: (Zigate) ----> Line 511 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/readClusters.py', function Cluster0000
2020-10-24 15:42:37.033 Error: (Zigate) ----> Line 462 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/lumi.py', function readXiaomiCluster
2020-10-24 16:36:49.003 Error: (Zigate) 'onMessage' failed 'error':'argument out of range'.
2020-10-24 16:36:49.003 Error: (Zigate) ----> Line 1007 in '/home/pi/domoticz/plugins/domoticz-zigate/plugin.py', function onMessage
2020-10-24 16:36:49.003 Error: (Zigate) ----> Line 517 in '/home/pi/domoticz/plugins/domoticz-zigate/plugin.py', function onMessage
2020-10-24 16:36:49.003 Error: (Zigate) ----> Line 548 in '/home/pi/domoticz/plugins/domoticz-zigate/Classes/Transport.py', function on_message
2020-10-24 16:36:49.003 Error: (Zigate) ----> Line 1163 in '/home/pi/domoticz/plugins/domoticz-zigate/Classes/Transport.py', function process_frame
2020-10-24 16:36:49.003 Error: (Zigate) ----> Line 523 in '/home/pi/domoticz/plugins/domoticz-zigate/plugin.py', function processFrame
2020-10-24 16:36:49.003 Error: (Zigate) ----> Line 229 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/input.py', function ZigateRead
2020-10-24 16:36:49.003 Error: (Zigate) ----> Line 2818 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/input.py', function Decode8102
2020-10-24 16:36:49.003 Error: (Zigate) ----> Line 2925 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/input.py', function read_report_attributes
2020-10-24 16:36:49.003 Error: (Zigate) ----> Line 217 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/readClusters.py', function ReadCluster
2020-10-24 16:36:49.003 Error: (Zigate) ----> Line 511 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/readClusters.py', function Cluster0000
2020-10-24 16:36:49.003 Error: (Zigate) ----> Line 462 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/lumi.py', function readXiaomiCluster
```

Domoticz info:
Version: 2020.2
Build Hash: b63341bc0
Compile Date: 2020-04-26 13:47:55
dzVents Version: 3.0.2
Python Version: 3.7.3 (default, Jul 25 2020, 13:03:44) [GCC 8.3.0]

I think i'm using ## 29 Sept. 2020  - 4.11.2 version if I can refer to the release note. Anyway, i'm on the latest commit of the stable branch.

Where the error come from? And what should I do to fix that?

pipiche38, 2020-10-24:
Do you have Xiaomi Weather Sensor ?
This could be one of them creating this issue .
Check if there battery level are correct ?

Miloune, 2020-10-25:
Which one Xiaomi Weather Sensor device is? All of my sensor are from Xiaomi but I don't think I got this one.
I got one lumi sensor with less than 10% of battery. Just switched it, but I have a question : Should I pair again the device to the Zigate?

Switched to stable-fix and restarted Domoticz, I will let you know if the error continue to occur.

pipiche38, 2020-10-25:
Basically the issue happened on decoding a Baro Pressure value, this is why I suspect a Weather sensor. 

I put a trap to capture new decoding error. Please send it back to me if it happens

Miloune, 2020-10-25:
New error message:
```
2020-10-25 18:05:24.464 Error: (Zigate) 'onMessage' failed 'TypeError'.
2020-10-25 18:05:24.464 Error: (Zigate) ----> Line 1007 in '/home/pi/domoticz/plugins/domoticz-zigate/plugin.py', function onMessage
2020-10-25 18:05:24.464 Error: (Zigate) ----> Line 517 in '/home/pi/domoticz/plugins/domoticz-zigate/plugin.py', function onMessage
2020-10-25 18:05:24.464 Error: (Zigate) ----> Line 548 in '/home/pi/domoticz/plugins/domoticz-zigate/Classes/Transport.py', function on_message
2020-10-25 18:05:24.465 Error: (Zigate) ----> Line 1163 in '/home/pi/domoticz/plugins/domoticz-zigate/Classes/Transport.py', function process_frame
2020-10-25 18:05:24.465 Error: (Zigate) ----> Line 523 in '/home/pi/domoticz/plugins/domoticz-zigate/plugin.py', function processFrame
2020-10-25 18:05:24.465 Error: (Zigate) ----> Line 229 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/input.py', function ZigateRead
2020-10-25 18:05:24.465 Error: (Zigate) ----> Line 2818 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/input.py', function Decode8102
2020-10-25 18:05:24.465 Error: (Zigate) ----> Line 2925 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/input.py', function read_report_attributes
2020-10-25 18:05:24.465 Error: (Zigate) ----> Line 221 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/readClusters.py', function ReadCluster
2020-10-25 18:05:24.465 Error: (Zigate) ----> Line 515 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/readClusters.py', function Cluster0000
2020-10-25 18:05:24.465 Error: (Zigate) ----> Line 467 in '/home/pi/domoticz/plugins/domoticz-zigate/Modules/lumi.py', function readXiaomiCluster
```

pipiche38, 2020-10-25:
could you try again 
update the stable-fix branch and restart plugin

Miloune, 2020-10-25:
Did it, I will let you know.
I used a new battery in this sensor but the plugin still say the current value is 1%. Do you think it's normal?

Miloune, 2020-10-27:
No more errors in my Domoticz logs 😊 thank you

closed 2020-10-27 by Miloune