zigbeefordomoticz/Domoticz-Zigbee #730: OpenWRT python3.9

Alex18081, 2020-11-01:
Hey.
On the latest version of Python3.9, the zigate plugin does not start, it stops in the Domoticz log on the first lines.
Help solve the problem
Other Python plugins work
Thanks!

pipiche38, 2020-11-01:
Logs please

Alex18081, 2020-11-01:
in the log of domotics I see only this

2020-11-01 09:53:26.933 Sunrise: 07:21:00 SunSet: 17:06:00
2020-11-01 09:53:26.934 Day length: 09:45:00 Sun at south: 12:13:00
2020-11-01 09:53:26.936 Civil twilight start: 06:47:00 Civil twilight end: 17:40:00
2020-11-01 09:53:26.938 Nautical twilight start: 06:08:00 Nautical twilight end: 18:18:00
2020-11-01 09:53:26.938 Astronomical twilight start: 05:31:00 Astronomical twilight end: 18:56:00
2020-11-01 09:53:27.908 Active notification Subsystems: (0/13)
2020-11-01 09:53:28.106 Starting shared server on: 0.0.0.0:6144
2020-11-01 09:53:33.528 (PPmanager) Debug logging mask set to: NONE
2020-11-01 09:53:33.530 (PPmanager) Domoticz Node Name is:OpenWrtGibrid
2020-11-01 09:53:33.531 (PPmanager) Domoticz Platform System is:Linux
2020-11-01 09:53:33.532 (PPmanager) Default Python Version is:3.9.0.
2020-11-01 09:53:33.534 (PPmanager) Collecting Updates for All Plugins!!!
2020-11-01 09:53:33.536 (PPmanager) Plugin Idle 
2020-11-01 10:26:27.684 Status: (Zigate) Entering work loop.
2020-11-01 10:26:27.683 Status: (Zigate) Initialized version 4.9, author 'zaraki673 pipiche38' 

is it possible to see a more detailed log?

pipiche38, 2020-11-01:
Is that a standard domoticz distribution or did you compile domoticz yourself ?

you might use the Debug Option when launching the plugin in Domoticz ? You can enable debug there

Alex18081, 2020-11-01:
Yes, this is not an assembly from a distribution, since the domoticz reset in the snapshot version of the opnvrt was broken and the domotics was removed from the snapshot branch.
devby solved the problem of compiling domotics in python3.9 and put together the latest distribution.
https://github.com/openwrt/packages/pull/13762#issuecomment-719752053
There was one problem with the launch of the zigat plugin
I already have loglevel 3 in domotics

pipiche38, 2020-11-01:
I'm not sure the compilation is solved, as I'm not able to do so whatever I applied his patch.

```
/usr/bin/ld: CMakeFiles/domoticz.dir/main/EventsPythonModule.cpp.o: in function `py3__Py_DECREF(char const*, int, _object*)':
EventsPythonModule.cpp:(.text+0xc0): undefined reference to `_Py_Dealloc'
/usr/bin/ld: CMakeFiles/domoticz.dir/main/EventsPythonDevice.cpp.o: in function `py3__Py_DECREF(char const*, int, _object*)':
EventsPythonDevice.cpp:(.text+0x84): undefined reference to `_Py_Dealloc'
/usr/bin/ld: CMakeFiles/domoticz.dir/hardware/plugins/Plugins.cpp.o: in function `py3__Py_DECREF(char const*, int, _object*)':
Plugins.cpp:(.text+0x10c): undefined reference to `_Py_Dealloc'
/usr/bin/ld: CMakeFiles/domoticz.dir/hardware/plugins/PluginProtocols.cpp.o: in function `py3__Py_DECREF(char const*, int, _object*)':
PluginProtocols.cpp:(.text+0x10c): undefined reference to `_Py_Dealloc'
/usr/bin/ld: CMakeFiles/domoticz.dir/hardware/plugins/PluginTransports.cpp.o: in function `py3__Py_DECREF(char const*, int, _object*)':
PluginTransports.cpp:(.text+0xc0): undefined reference to `_Py_Dealloc'
/usr/bin/ld: CMakeFiles/domoticz.dir/hardware/plugins/PythonObjects.cpp.o:PythonObjects.cpp:(.text+0x10c): more undefined references to `_Py_Dealloc' follow
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/domoticz.dir/build.make:4302: domoticz] Error 1
make[1]: *** [CMakeFiles/Makefile2:355: CMakeFiles/domoticz.dir/all] Error 2
```

pipiche38, 2020-11-01:
As suggested, please enabled the Full Debug when starting the plugin .

You can also edit the file Conf/PluginConf-xx.json and add (following the json syntax) : 

- "debugPlugin": 1,

pipiche38, 2020-11-03:
Any update ?

Any insight on how I could fix the compilation issues I have ? So I could eventually troubleshoot myself .

Alex18081, 2020-11-04:
unfortunately I could not get a more extensive log.
Have you tried applying the patch from devbis?

pipiche38, 2020-11-04:
This is what I did . And got the Link errors !

Alex18081, 2020-11-04:
have you applied all the previous patches as well?
https://github.com/openwrt/packages/pull/13762/commits/9a56ca92ded3c0a59dd1fafe44bc27c6614360ab#diff-6cc72eec635ebca5e5a339b7c71e81467084a2f5ce980740781852157007a2b6R78

closed 2021-01-15 by stale[bot]