zigbeefordomoticz/Domoticz-Zigbee #741: Is it possible support 1 000 000 baud speed?

itProfi, 2020-11-12:
Good day! Is it possible to integrate 1 000 000 baud rate to serial speed to support zigate firmware with 1000000 baud rate?,

pipiche38, 2020-11-12:
Why ? The ZiGate firmware are only on 115200 which is more than enought.

itProfi, 2020-11-12:
Not only 115200, and user use 1000000 baud rate also..
[image]

pipiche38, 2020-11-12:
Yes, I know , but why are you building the firmware and not using the standard one ?
The recommendation is to use 115200 and going to  1M is not providing any benefits

closed 2020-11-12 by itProfi

pipiche38, 2020-11-30:
If you really want to use 100000 Bauds and you are able to build your own firmware with such speed. I suggest that you change the BAUDS parameter in the source code ( Classes/Transport.py )