zigbeefordomoticz/Domoticz-Zigbee #768: give more explicit messages about current and supported firmware

dpocock, 2020-12-30:
My Zigate stick was running 3.1a with 4.4.9 (the Debian package)

I decided to try the latest beta code, 4.12.087

In the log, it says:

```
Status: (zigate0) Plugin with Zigate firmware 031a correctly initialized
(zigate0) checkFirmwareUpdate - Firmware update available
Status: (zigate0) There is a newer Zigate Firmware version available
```

In the file ReleaseNotes.md it says

```
## 5 June 2020 - 4.9.1 (Last release supporting Firmware below 3.1c)
```

Is the 031a firmware still supported with 4.12.x?

The message ```Firmware update available``` may need to be stronger, e.g. ```You must use at least firmware version 3.1c and at most 3.1d```

If the detected version is not within the range then it would be useful for that log entry to have the highest severity, e.g. CRIT or ERROR

pipiche38, 2020-12-31:
There is no change for now and stable 4.11.xxx still support 31a firmware

pipiche38, 2020-12-31:
Will be addressed when needed

closed 2020-12-31 by pipiche38