zigbeefordomoticz/Domoticz-Zigbee #770: fails to write log files

dpocock, 2020-12-30:
I'm using beta v4.12.087

Whenever it tries to write to the log, it fails:

```
Error: (zigate0) ----> Line 129 in '/usr/lib/domoticz/plugins/Domoticz-Zigate/Classes/LoggingManagement.py', function logging
Error: (zigate0) ----> Line 160 in '/usr/lib/domoticz/plugins/Domoticz-Zigate/Classes/LoggingManagement.py', function loggingError
Error: (zigate0) ----> Line 222 in '/usr/lib/domoticz/plugins/Domoticz-Zigate/Classes/LoggingManagement.py', function loggingWriteErrorHistory
```

These are the settings revealed in the log

```
 Status: (zigate0) filename set to /etc/domoticz/plugins/Domoticz-Zigate/PluginConf-02.json
Status: (zigate0) pluginHome set to /usr/lib/domoticz/plugins/Domoticz-Zigate/
Status: (zigate0) homedirectory set to /usr/lib/domoticz/plugins/Domoticz-Zigate/
Status: (zigate0) pluginData set to /var/lib/domoticz/plugins/Domoticz-Zigate/Data/
Status: (zigate0) pluginConfig set to /etc/domoticz/plugins/Domoticz-Zigate/
Status: (zigate0) pluginOTAFirmware set to /usr/lib/domoticz/plugins/Domoticz-Zigate/OTAFirmware/
Status: (zigate0) pluginReports set to /var/lib/domoticz/plugins/Domoticz-Zigate/Reports/
Status: (zigate0) pluginWWW set to /usr/lib/domoticz/plugins/Domoticz-Zigate/www/
Status: (zigate0) pluginLogs set to /usr/lib/domoticz/plugins/Domoticz-Zigate/Logs/
Status: (zigate0) Log history not found, no error logged 
```

and the permissions on the directory:

```
$ ls -la /usr/lib/domoticz/plugins/Domoticz-Zigate/Logs/
total 0
drwxr-xr-x 1 domoticz domoticz  0 Dec 30 20:45 .
drwxr-xr-x 1 root     root     30 Dec 30 21:17 ..
```

pipiche38, 2020-12-31:
Without the full log I cannot really do anything, so please provide the logs around the error it self.

pipiche38, 2021-01-20:
@deufo can you kindly have a look to that one.

I think there is indeed a potential issue is this code:

In case self.PluginHealth is not yet set, than the if `'Txt' not in self.PluginHealth` will crash 

```
        if not self.PluginHealth:
            _txt = 'Not Started'
        if 'Txt' not in self.PluginHealth:
            _txt = 'Not Started'
        else:
            _txt = self.PluginHealth['Txt']
        _context = {
                    'Time': int(time.time()),
                    'Module': module,
                    'nwkid': nwkid,
                    'PluginHealth': _txt,
                    'message': message,
                    'PermitToJoin': self.permitTojoin,
                    'Thread': threading.current_thread().name
                }
```

SylvainPer, 2021-01-21:
@pipiche38 on the v4.12.087, the line 222 is this one:

`        with open( jsonLogHistory, "w", encoding='utf-8') as json_file:`

It seems that the Open fonction is not launched before the write. I need to modify the option to 'w+'.

SylvainPer, 2021-01-21:
But it shouldn't happen, this modification will avoid the error but the log will be cleared if the file was existing.

SylvainPer, 2021-01-21:
@dpocock can you give us more details ?
I checked the plugin init and I don't understand why the file is not created.
You should have first an error with the openLogFile function. There, the option request the creation file.

closed 2021-06-09 by stale[bot]