zigbeefordomoticz/Domoticz-Zigbee #771: using Ikea remote/dimmer buttons in a Domoticz group

dpocock, 2020-12-30:
I read the instructions in the wiki pages about setting Zigbee groups involving Ikea bulbs with Ikea remote/dimmer buttons
https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/5c1e0778eebebf20928fb22e9b8ec4b213300f6e/en-eng/IKEA-Tradfri-corner.md
https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Ikea-Remote.md

At the moment, I don't pair remotes directly to bulbs

I configure groups in Domoticz, on the Scenes tab

I can join the Ikea remotes with the Zigate and I can see them in the Devices tab and in the Zigate plugin web page on port 9440.

If I go to the Scenes page, select a group and click Edit, I try to add the Ikea remote.  I click "Add Device" and it asks me to press the button on the remote.  I try all the buttons but it is not recognized.  Then the "Timeout" error appears in Domoticz.

The same setup procedure is working with the Aqare and Philips Hue dimmers but not the Ikea remote/dimmers.

deennoo, 2020-12-30:
Have You test some other "python plugin" device ?

Aqara and hue are native on domoticz code.

On my side i use ikea switch (small on/off), i have 6 of them, to drive domoticz group, i use lua event

`if (devicechanged['Inter Chevet Nico'] =="On" or  devicechanged['Inter Chevet Mel'] =="On")  then 
     commandArray['Group:nomdugroupe']='On''

Can be reproduce for off, and can be use just a push button : 

`if (devicechanged['Inter Chevet Nico'] =="On" or  devicechanged['Inter Chevet Mel'] =="On") and (otherdevices_scenesgroups['nomdugroupel'] == 'Off') then 
     commandArray['Group:nomdugroupe']='On'
elseif (devicechanged['Inter Chevet Nico'] =="On" or  devicechanged['Inter Chevet Mel'] =="On") and (otherdevices_scenesgroups['nomdugroupel'] ~= 'Off') then 
     commandArray['Group:nomdugroupe']='Off''

on the last exemple : you can check the groupe status (on/off/mixte) and any press (on or off) on the switch will revert the groupe state

pipiche38, 2020-12-31:
@ dpocock  Please use the forum instead of github until we confirm it is an issue.

ZiGate and the plugin support native group and will give you much better User Experience than the Domoticz group.

closed 2020-12-31 by pipiche38