zigbeefordomoticz/Domoticz-Zigbee #775: Problème de démarrage du plugin avec Python 3.9

schnokbul, 2021-01-09:
**Describe the bug**
Bonjour,
Après mise à jour de mon système (Linux atalante 4.19.0-8-686-pae #1 SMP Debian 4.19.98-1+deb10u1 (2020-04-27) i686 GNU/Linux) sur python 3.9, j'ai du recompiler domoticz. J'ai utilisé le patch fourni sur le forum.
Jusque là pas de soucis, Domoticz tourne sans problèmes.
Par contre, le plugin refuse de se lancer et dans le log Domoticz j'ai ça :
2021-01-09 02:13:22.754 Error: (Zigate) 'onStart' failed 'TypeError'.
2021-01-09 02:13:22.754 Error: (Zigate) ----> Line 985 in '/home/schnokbul/domoticz/plugins/Domoticz-Zigate/plugin.py', function onStart
2021-01-09 02:13:22.754 Error: (Zigate) ----> Line 222 in '/home/schnokbul/domoticz/plugins/Domoticz-Zigate/plugin.py', function onStart
2021-01-09 02:13:22.754 Error: (Zigate) ----> Line 293 in '/usr/lib/python3.9/json/__init__.py', function load
2021-01-09 02:13:22.755 Error: (Zigate) ----> Line 359 in '/usr/lib/python3.9/json/__init__.py', function loads

J'ai activé le debug du plugin via "debugPlugin":1, dans le PluginConf-15.json mais rien ne sort de plus dans aucun log...

Domoticz version 2020.2(build 12834)
plugin 4.11.013 stable

J'ai cherché sur les forums, mais il semble que personne n'ai encore eu le problème.

Si quelqu'un a une idée...

Merci beaucoup

schnokbul, 2021-01-09:
Bon après avoir consulté un ami expert en python, le problème venait bien de la fonction json.load.
Il suffit d'enlever l'argument encoding=dict dans tous les endroits où il est utilisé. 
C'est à dire dans :
- Domoticz-Zigate/plugin.py :222
- Domoticz-Zigate/Classes/PluginConf.py :380
- Domoticz-Zigate/Classes/LoggingManagement.py :63

Et là plus aucune erreur, tout se lance sans soucis !

Je laisse le soin à qui de droit de prévoir un patch pour régler ça dans les futures versions.

pipiche38, 2021-01-09:
Thanks to use English to report bugs/issues.

Please provide a PR fixing those issues and we will be pleased to integrated

closed 2021-01-09 by pipiche38