zigbeefordomoticz/Domoticz-Zigbee #777: Tuya Dimmer TS0601 / Manufacturer Name : "_TZE200_dfxkcots"

deennoo, 2021-01-12:
Starting Plugin 4.7, there are some tools which allow to share all information collected during the pairing/discovery process.

[image]

1. Got to the Tools Menu
1. Select Plugin Raw Devices (json)
1. Export

This will create an export.json file that you can save.
   
Finally you create an issue on GitHub with the following details

* Product Name : TS0601 - TZ200 dimmer
* Brand : Earda
* 
* URL on a web site : https://www.aliexpress.com/item/Tuya-Zigbee-Smart-Dimmer-Switch-Remote-Control-Touch-Dimmer-LED-Voice-Control-Compatiable-With-Alexa-Google/1005001505069156.html?spm=a2g0s.9042311.0.0.26906c37v3W5Q3
* 
* What do you expect : on/off/dim up / dim down
* 
* The logs from the pairing process
```
2021-01-12 18:01:22.428 Status: (Zigate RDC) [-] NEW OBJECT: d41e Active Endpoint Response Ep: 01 LQI: 129
2021-01-12 18:01:22.429 Status: (Zigate RDC) [-] NEW OBJECT: d41e Request Simple Descriptor for Ep: 01
2021-01-12 18:01:22.786 Status: (Zigate RDC) [-] NEW OBJECT: d41e Simple Descriptor Response EP: 0x01 LQI: 129
2021-01-12 18:01:22.786 Status: (Zigate RDC) [-] NEW OBJECT: d41e ProfileID 0104
2021-01-12 18:01:22.786 Status: (Zigate RDC) [-] NEW OBJECT: d41e ZDeviceID 0051
2021-01-12 18:01:22.786 Status: (Zigate RDC) [-] NEW OBJECT: d41e Application Version 0001
2021-01-12 18:01:22.786 Status: (Zigate RDC) [-] NEW OBJECT: d41e Cluster IN Count: 04
2021-01-12 18:01:22.786 Status: (Zigate RDC) [-] NEW OBJECT: d41e Cluster In 1: 0000 (Basic)
2021-01-12 18:01:22.786 Status: (Zigate RDC) [-] NEW OBJECT: d41e Cluster In 2: 0004 (Groups)
2021-01-12 18:01:22.786 Status: (Zigate RDC) [-] NEW OBJECT: d41e Cluster In 3: 0005 (Scenes)
2021-01-12 18:01:22.786 Status: (Zigate RDC) [-] NEW OBJECT: d41e Cluster In 4: ef00
2021-01-12 18:01:22.786 Status: (Zigate RDC) [-] NEW OBJECT: d41e Cluster OUT Count: 02
2021-01-12 18:01:22.786 Status: (Zigate RDC) [-] NEW OBJECT: d41e Cluster Out 1: 0019 (Over-the-Air Upgrade)
2021-01-12 18:01:22.786 Status: (Zigate RDC) [-] NEW OBJECT: d41e Cluster Out 2: 000a (Time)
2021-01-12 18:01:24.049 (Zigate RDC) Creation Group
2021-01-12 18:01:24.423 (RFXTRXE) Lighting 4 (Motion Bureau2)
2021-01-12 18:01:24.042 Status: (Zigate RDC) [0] NEW OBJECT: d41e Model Name: TS0601
```

* export.json file as described here above

full device capture Using Ziagte fw 3.1e + plugin v 4.12.093
Off
On with mini dim
On with max dim
Dim up
Dim Down
[ts0601 dimmer.zip](https://github.com/pipiche38/Domoticz-Zigate/files/5803911/ts0601.dimmer.zip)

pipiche38, 2021-01-18:
Integrated with 6d6decd

pipiche38, 2021-01-18:
Duplicate #778

closed 2021-01-18 by pipiche38