zigbeefordomoticz/Domoticz-Zigbee #779: Improving eTRV Tuya integration

pipiche38, 2021-01-12:
https://www.domoticz.com/forum/viewtopic.php?f=68&t=35161

pipiche38, 2021-01-12:
2021-01-11 21:55:07.932 Status: (ZiGate) [ZiGateForwarder_8] ZigateTransport: thread_processing_and_sending Thread start.
2021-01-11 21:55:17.854 Status: (ZiGate) [ZiGateForwarder_8] ZiGate Classic
2021-01-11 21:55:17.854 Status: (ZiGate) [ZiGateForwarder_8] Installer Version Number: 031d
2021-01-11 21:55:26.914 (ZiGate) [ZiGateForwarder_8] setZigateIEEE - Set Zigate IEEE: 00158d0001ed4521
2021-01-11 21:55:26.914 Status: (ZiGate) [ZiGateForwarder_8] Zigate addresses ieee: 00158d0001ed4521 , short addr: 0000
2021-01-11 21:55:26.914 Status: (ZiGate) [ZiGateForwarder_8] Network state UP, PANID: 2108 extPANID: 0x4e8fd40d0afd69a1 Channel: 11
2021-01-11 21:55:32.781 Status: (ZiGate) [ZiGateForwarder_8] Accepting new Hardware: Disable
2021-01-11 21:55:33.086 Status: (ZiGate) [ZiGateForwarder_8] Number of devices recently active in Zigate = 15
2021-01-11 21:55:33.086 Status: (ZiGate) [ZiGateForwarder_8] [00] DevID = 00 Network addr = 9419 IEEE = 00158d0002c9cbc3 LQI = 117 Power = 00 Model = lumi.weather ( Temp Slaapkamer )
2021-01-11 21:55:33.087 Status: (ZiGate) [ZiGateForwarder_8] [01] DevID = 01 Network addr = d9bb IEEE = 00158d0002b81047 LQI = 179 Power = 00 Model = lumi.remote.b186acn01 ( Schakelaar Eetkamer )
2021-01-11 21:55:33.087 Status: (ZiGate) [ZiGateForwarder_8] [02] DevID = 02 Network addr = 87bf IEEE = 90fd9ffffef166a6 LQI = 227 Power = 00 Model = TRADFRI remote control ( Ikea Remote device )
2021-01-11 21:55:33.088 Status: (ZiGate) [ZiGateForwarder_8] [03] DevID = 03 Network addr = a73b IEEE = 00158d0003058fc2 LQI = 136 Power = 00 Model = lumi.sensor_motion.aq2 ( bewegin computerkast )
2021-01-11 21:55:33.088 Status: (ZiGate) [ZiGateForwarder_8] [04] DevID = 04 Network addr = 5fdb IEEE = 00158d0004521e0d LQI = 130 Power = 00 Model = lumi.sensor_motion.aq2 ( Beweging Woonkamer )
2021-01-11 21:55:33.089 Status: (ZiGate) [ZiGateForwarder_8] [05] DevID = 05 Network addr = 2543 IEEE = 00158d00028afbfe LQI = 238 Power = 00 Model = lumi.sensor_cube.aqgl01 ( Cube schakelaar )
2021-01-11 21:55:33.089 Status: (ZiGate) [ZiGateForwarder_8] [06] DevID = 06 Network addr = a34b IEEE = 00158d0002ec3353 LQI = 165 Power = 00 Model = lumi.sensor_motion.aq2 ( Beweging Eetkamer )
2021-01-11 21:55:33.090 Status: (ZiGate) [ZiGateForwarder_8] [07] DevID = 08 Network addr = 59b2 IEEE = 00158d0002c441e1 LQI = 216 Power = 00 Model = lumi.sensor_switch ( Ronde Schakelaar )
2021-01-11 21:55:33.090 Status: (ZiGate) [ZiGateForwarder_8] [08] DevID = 28 Network addr = 4727 IEEE = d0cf5efffe79fab1 LQI = 118 Power = 01 Model = TRADFRI bulb E27 W opal 1000lm ( Hanglamp Eetkamer )
2021-01-11 21:55:33.091 Status: (ZiGate) [ZiGateForwarder_8] [09] DevID = 29 Network addr = ba6a IEEE = 000d6ffffe268b00 LQI = 072 Power = 01 Model = TRADFRI bulb E27 CWS opal 600lm ( Schemerlamp Dressoir )
2021-01-11 21:55:33.091 Status: (ZiGate) [ZiGateForwarder_8] [11] DevID = 2b Network addr = e90e IEEE = 000b57fffeef8f35 LQI = 063 Power = 01 Model = TRADFRI control outlet ( Maxcube Reset switch )
2021-01-11 21:55:33.092 Status: (ZiGate) [ZiGateForwarder_8] [12] DevID = 2d Network addr = 9a93 IEEE = 00158d00032d3827 LQI = 192 Power = 01 Model = lumi.plug ( lamp computer )
2021-01-11 21:55:33.092 Status: (ZiGate) [ZiGateForwarder_8] [14] DevID = 2f Network addr = 63d2 IEEE = d0cf5efffec6771b LQI = 067 Power = 01 Model = TRADFRI transformer 10W ( Led keuken )
2021-01-11 21:55:40.295 Status: (ZiGate) [ZiGateForwarder_8] TxPower Attenuation : 0 dBm
2021-01-11 21:56:05.092 (ZiGate) [ZiGateForwarder_8] Receive NACK from Tuinlamp (03d2) clusterId: 0000 Status: a7
2021-01-11 21:56:15.090 (ZiGate) [ZiGateForwarder_8] Receive NACK from Schemerlamp Dressoir (ba6a) clusterId: 0000 Status: a7
2021-01-11 21:56:21.589 (ZiGate) [ZiGateForwarder_8] Receive NACK from Signal Repeater waskamer (ce5d) clusterId: 0000 Status: a7
2021-01-11 21:56:30.089 (ZiGate) [ZiGateForwarder_8] Receive NACK from aqara schakelaar woonkamer (6f39) clusterId: 0000 Status: a7
2021-01-11 21:56:36.531 (ZiGate) [ZiGateForwarder_8] Receive NACK from Led keuken (63d2) clusterId: 0000 Status: a7
2021-01-11 21:57:02.032 (ZiGate) [ZiGateForwarder_8] Receive NACK from Schemerlamp Dressoir (ba6a) clusterId: 0300 Status: a7
2021-01-11 21:57:08.491 (ZiGate) [ZiGateForwarder_8] Receive NACK from Led keuken (63d2) clusterId: 0006 Status: a7
2021-01-11 21:57:10.214 (ZiGate) [ZiGateForwarder_8] Receive NACK from Led keuken (63d2) clusterId: 0008 Status: a7
2021-01-11 22:00:11.301 Status: (ZiGate) [ZiGateForwarder_8] ZigateTransport: thread_processing_and_sending Thread stop.
2021-01-11 22:00:12.029 Status: (ZiGate) [ZiGateForwarder_8] ZigateTransport: thread_processing_and_sending Thread start.
2021-01-11 22:00:21.902 Status: (ZiGate) [ZiGateForwarder_8] ZiGate Classic
2021-01-11 22:00:21.902 Status: (ZiGate) [ZiGateForwarder_8] Installer Version Number: 031d
2021-01-11 22:00:30.989 (ZiGate) [ZiGateForwarder_8] setZigateIEEE - Set Zigate IEEE: 00158d0001ed4521
2021-01-11 22:00:30.989 Status: (ZiGate) [ZiGateForwarder_8] Zigate addresses ieee: 00158d0001ed4521 , short addr: 0000
2021-01-11 22:00:30.989 Status: (ZiGate) [ZiGateForwarder_8] Network state UP, PANID: 2108 extPANID: 0x4e8fd40d0afd69a1 Channel: 11
2021-01-11 22:00:36.812 Status: (ZiGate) [ZiGateForwarder_8] Accepting new Hardware: Disable
2021-01-11 22:00:37.114 Status: (ZiGate) [ZiGateForwarder_8] Number of devices recently active in Zigate = 12
[... cut ...]
2021-01-11 23:39:28.498 (ZiGate) [ZiGateForwarder_8] tuyaReadRawAPS - Unknown command 24 MsgPayload 1907240009
2021-01-11 23:45:49.920 (ZiGate) [ZiGateForwarder_8] Decode8002 - NwkId: 8d4d Ep: 01 Cluster: ef00 GlobalCommand: False Command: 24 Data: 0031
2021-01-11 23:45:49.921 (ZiGate) [ZiGateForwarder_8] tuyaReadRawAPS - Nwkid: 8d4d Ep: 01, Cluster: ef00, dstNwkid: 0000, dstEp: 01, Payload: 1959240031
2021-01-11 23:45:49.921 (ZiGate) [ZiGateForwarder_8] tuyaReadRawAPS - Unknown command 24 MsgPayload 1959240031

pipiche38, 2021-01-12:
please do use the SireneTuyaTS0601 branch

`git checkout SireneTuyaTS0601`

I have push cc31c3df which start integrating the Valve widget . To be tested. For now there is no way to control the valve directly. The only possibility to action the valve is via SetPoint

To do the test, remove all widgets, and redo the full pairing 

please keep debugTuya on and capture the log when exercising the setpoint ...

waltervl, 2021-01-12:
[tuyats0601_20210112.log](https://github.com/pipiche38/Domoticz-Zigate/files/5804489/tuyats0601_20210112.log)
Attached the log (approx 1 hour)
My observations: 
Window Lock: At approx 20.51 I activated the Window lock action by tacking it outside (0 degr). I had to set it manually on the eTRV. The switch was created but no action on switching and no logging when activated.
What is the usage of the Switch Widget? Reporting if the window lock has been activated by the the device or setting the Window lock option On/Off?
Setpoint: When changed from Domoticz there seems to be no reaction at device. When changing it on the eTRV it changed in Domoticz.
Valve: Dimmer style switch created. Not neccessary as it is no setting. It is a Sensor reading. So a percentage sensor widget is enough. On the eTRV I can se the values changing when setting the setpoints.
Battery: no battery feedback

waltervl, 2021-01-12:
Additional, I left the other eTRV paired and there the Setpoint action in Domoticz is read by the eTRV.

waltervl, 2021-01-12:
Just checked the log and see that room temp is now 30 degr which is not true. I saw this also in the first hour that the room temp suddenly rised to 30 degree.

pipiche38, 2021-01-13:
I'm afraid that we won't succeed like that. I'm reading from your comments that some stuff which were working don't anymore which sound bizarre as I didn't touch for instance Temperature nor Setpoint
`
I'll order one eTRV and when receive will work on the integration.

waltervl, 2021-01-13:
Thanks, I bought the 2 from aliexpress, send from Tsjech, took 2 weeks (during Christmas time).

€ 16,18  40%OFF | Tuya ZIGBEE3.0 Smart Radiator Actuator Programmeerbare Thermostaatkraan Temperatuur Controller Voice Control Via Alexa
https://a.aliexpress.com/_BPQ00f

pipiche38, 2021-01-13:
Bought that one. I guess they will be similar

https://fr.aliexpress.com/item/1005002004862997.html?spm=a2g0s.9042311.0.0.43c76c37ezErvE

pipiche38, 2021-01-18:
@waltervl  

> percentage sensor widget is enough

I'm not aware of such Widget in Domoticz . if you are I would really like to see how does it looks

waltervl, 2021-01-18:
There are see screenshot and see for example the maxcube plugin "GitHub - mvzut/maxcube-Domoticz-plugin: ELV/EQ-3 MAX! Python plugin for Domoticz" https://github.com/mvzut/maxcube-Domoticz-plugin

[image]

pipiche38, 2021-01-18:
indeed much better !

pipiche38, 2021-01-18:
Valve reporting PI Heating demand in % is now in place. It has been validated with Eurotronics equipment.
Just waiting for receiving the TRV

pipiche38, 2021-01-21:
eTRV on my desk .... As soon as I have some time I'll start decoding it.

The good news is with the request from @waltervl I have improved the behaviour of the Danfoss and Eurotronics TRV with the Valve % which is a great feature.

pipiche38, 2021-01-24:
https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Technical/Tuya-0xEF00.md

pipiche38, 2021-01-24:
450e909d

Something starting to work.

- Setpoint Ok
- Local Temperature Ok
- Calibration Ok

Need to work on Program Auto (by default, versus Manual)

For now
No battery
No Valve %
No Window detection

pipiche38, 2021-01-26:
So I have made some good progress and get some better understanding.
It looks like they are as many diffent Tuya eTRV as you can purchase.

The test I 'm doing with one. Do not report instant Valve %, nor battery level.

Would appreciate @waltervl if you could give me an export.json ( Plugin Device List RAW (json) ) from the system where you have already paired your device. This is to see if that is the same as mine or one of the many other version (with différent flavour)

waltervl, 2021-01-26:
Sure, no problem. Attached the export.json
[export20210126.json.txt](https://github.com/pipiche38/Domoticz-Zigate/files/5876430/export20210126.json.txt)

pipiche38, 2021-01-27:
@waltervl thanks for that.

if you want to test where am I this would be helpfull as I don't have the same eTRV as you .

to do so:

```
git pull
git checkout SireneTuyaTS0601
git pull
```

restart the plugin
remove the existing widget in order to redo the pairing
make sure that tuyaDebug is enabled 

any logs would be helpfull in addition to the export.json

if you need to fall back to beta

`git checkout beta`

waltervl, 2021-01-28:
[Domoticztuyalog.txt](https://github.com/pipiche38/Domoticz-Zigate/files/5889803/Domoticztuyalog.txt)
[export(2).json.txt](https://github.com/pipiche38/Domoticz-Zigate/files/5889804/export.2.json.txt)
See logs.
My observations: 
1. Valve widget created and working as  expected.
2. Temperature widget created and working as expected.
3. Mode selector switch created but no response from eTRV when switching. I did not change modes from the eTRV. I will try this later.
4. Setpoint widget created. Reading changes from eTRV works as expected (but a 3 degrees change is reported to domoticz in steps of 0.5 degrees). Changing the setpoint in Domoticz is not received by the eTRV, no change visible on the eTRV. Domoticz shows the new setpoint but after 5-10 minutes it is set back to the original setpoint (as reported by the eTRV)

pipiche38, 2021-01-28:
Sound good. we are progressing.
Many things for the logs and the export.

1/ The SetPoint issue should be solved in the latest version.
2/ For the Mode I hope to have fix it on the latest version pushed now.

As the Setpoint going slowly to the target by step of .5 I do confirm this is what the Device is sending

Commit of the day: 0eafbe39

pipiche38, 2021-01-29:
I have done some changes to make the code more maintainable based so far with 3 differents devices
( c13e0e35 )

please do a git pull to get the latest version of the branch, then. you need to remove the existing widgets and redo the pairing

waltervl, 2021-01-30:
Did a git pull, including latest commit of today. Widgets are created but nu feedback on widgets, no temp, valve etc.
See attached log
[domoticzLog30-1.txt](https://github.com/pipiche38/Domoticz-Zigate/files/5898395/domoticzLog30-1.txt)

pipiche38, 2021-01-30:
Please try once more

waltervl, 2021-01-31:
Yes, much better now ;-) I will leave it running for a day of so.

waltervl, 2021-01-31:
By the way, mine looks like this:
[image]

pipiche38, 2021-01-31:
would appreciate if you could drop me an export.json after it has run a while ;-)

waltervl, 2021-01-31:
Here is my export. It is not completely there yet.
Update of temperature and valve seems only to work when setpoint is set. 
Device is not listening to Manual/Automatic mode change from Domoticz.
[export31-1.json.txt](https://github.com/pipiche38/Domoticz-Zigate/files/5900331/export31-1.json.txt)

waltervl, 2021-02-01:
I have 2 the same eTRV's. One is acting normal (Nwkid f4e3) the other one (a5a2) is sending data but less active then Nwkid f4e3 to Domoticz. For example: could be hours silence and then 4 degres rise in temp. Also switching of mode from Domoticz is not working on this a5a2 device. Switching mode on the device is working and reported back to Domoticz. Setpoint from Domoticz is working on both devices.

You see me switching mode and five minutes later the mode is switched back to the original state as reported by the device.

[domoticzlog_tuya1-2.txt](https://github.com/pipiche38/Domoticz-Zigate/files/5906000/domoticzlog_tuya1-2.txt)
[export1-2.json.txt](https://github.com/pipiche38/Domoticz-Zigate/files/5906003/export1-2.json.txt)

pipiche38, 2021-02-01:
LQI for the a5a2 seems quiet low. Can I suggest that you swap the 2 , and you check how it works ?

waltervl, 2021-02-01:
I have restarted the eTRV (batteries out/in) and now it is connected to a router (Ikea bulb) within 50 cm instead of a router (ikea bulb) 6 meter away.
It is now listening. Let's see how it evolves....

waltervl, 2021-02-04:
I have had them close together for the last 2 days. Almost the same high LQI.
One is updating regurarly, the other still is slow in updating temperatures.
I will update to beta as I see the integration is merged into beta.

pipiche38, 2021-02-05:
May be you should ask to get the eTRV changed

pipiche38, 2021-02-06:
Closing as the work as been merged in beta.

closed 2021-02-06 by pipiche38

waltervl, 2021-02-06:
Thank you again for your work.